mongoose-os-apps/shelly-homekit #809: Shelly 1 relay stops working & I have to reset it

raffi961, 2021-10-16:
Hello everyone
I am having a weird problem with my shelly 1 which is installed in my room.

just for context i have many shelly 1s with mongoose installed in the house and all work perfectly except this 1 shelly which is in my room.

every couple of months the relay inside of the shelly 1 stops responding to the actual button/switch on the wall.
So what happens is that i have a motion sensor which triggers the shelly to turn on the light.
What happens is that it turn it on and then i cannot turn it off wither with the manual switch or in the app.
When in the home app, when i press the manual switch on the wall, i can see that in the home app, the switch goes from on to off and vis versa but the light stays on. The shelly doesnt do the usual click sound which means the relay is not working.
I tried cutting off the power to the switch by turning off the electrical jumper and still nothing.
The only way to get it working again is to actually restore the shelly to its original firmware and then reinstalling mongoose.
What i have tried to do is to install a completely nee shelly and after couple of months same thing happened.
So now i always have a backup shelly at hand with mongoose firmware on it and when this happens and i replace the one that stopped working with the backup one and then i reset the one i removed from the wall to the original shelly firmware and then re-install mongoose and keep it as a backup for next time this happens.
What could be the reason ?

rojer, 2021-10-16:
hm, that's really strange. so it reacts to input and changes state in homekit but doesn't change the output? what if you change the state in homekit, does that actuate the output?
maybe something happens to the config, restoring stock wipes homekit config keys.
anyway, next time this happens please collect the log (debug log link)
and also contents of `/rpc/Config.Get` (make sure to screen out wifi credentials).

raffi961, 2021-10-16:
Indeed What happens is when the Light stays on and doesn't react to the physical Switch on the wall, when I press the physical button, the state of the switch in the Home app changes from On to Off. If i press the button on the Home app, the state in the home app changes to ON and when i press the Physical Button on the wall, the state changes in the Home app to OFF. BUT the actual output (Light) stays on. When this happens again, I will save and share the Log output

morrisk6, 2021-11-05:
I am experiencing what seems like the same issues that are described.with 2 Shelly 1's. That is the Home app will show a different state to the Shelly app. Further on one Shelly I can turn the switch off from Home but not on and from the other Shelly it is the opposite - I can turn the with on from Home but not off.
Have you made any progress on this issue after looking at the log outputs?

closed 2021-12-14 by github-actions[bot]