mongoose-os-apps/shelly-homekit #829: Double click and long press detached

bmpalmeida, 2021-11-02:
Hello,

I’m using the detached mode to have access in HomeKit to double click and long press events. But the single click gets really slow. Because it needs to be processed by the HomeKit. 

Any change to have the single click “standard”. E.g. turn on and off the lights directly in the switch. Without rely on HomeKit. But expose the double and long press to the HomeKit?

lucaspinelli85, 2022-01-03:
Very interesting, it would also be useful for me.
It would have many advantages.
- in the absence of wifi, the simple pressure would work the same.
-the single press would be much faster and more direct without going through homekit
-does not need to configure the single press in homekit
-long and double press would be used to make functionality in addition to the standard single press.
All this without putting the single pressure in detached mode

andyblac, 2022-02-19:
double of https://github.com/mongoose-os-apps/shelly-homekit/issues/9, and that was merged with https://github.com/mongoose-os-apps/shelly-homekit/issues/147

timoschilling, 2022-02-19:
This is something else then #9

andyblac, 2022-02-19:
look at #147 @rojer wanted ALL inputs modes in that issue.

andyblac, 2022-02-19:
also read through the #9 issue it does mention this issue.

andyblac, 2022-02-19:
see https://github.com/mongoose-os-apps/shelly-homekit/issues/320

closed 2022-02-19 by andyblac