mongoose-os-apps/shelly-homekit #830: Homekit firmware problem for RGBW2

sg-generix, 2021-11-03:
Hello,
I flashed 2 devices - Shelly RGBW2 with the last firmware, following the standard procedure, described on git hub page. The devices was installed in Shelly app, and after that I updated to homekit using IP adrress and URL http://A.B.C.D/ota?url=http://rojer.me/files/shelly/shelly-homekit-ShellyRGBW2.zip.

The devices lost WIFI connection after update and they switch to AP mode. When I connect with AP and open it from browser, I  try to add WIFI connection they restart again in AP mode. It is interesting that the STATUS button is switching non-stop for every 3-4 seconds and when I check log the device non stop turn on and turn off.  

Any help please? I flashed some other type of devices like Shelly 1 and Shelly Plug S and no problem. Everything works perfect.

timoschilling, 2021-11-03:
Can you post a draw or a picture of your wiring?

sg-generix, 2021-11-04:
Hello, we connected according the schema rgbw_led_strip_shelly.png. I attached some pictures to see how we made connection. Thank you for helping.
[image]
[image]

timoschilling, 2021-11-04:
Remove the 240V to 12V Switch converter, you don't need it in your case and we don't support it at the moment.
[image]

sg-generix, 2021-11-04:
Thank you very much. I will test it later and I will report the result.

sg-generix, 2021-11-05:
Thank you @timoschilling 
Everything works perfect. I removed the switch convector.

closed 2021-11-06 by timoschilling

kellery82, 2022-08-19:
> Remove the 240V to 12V Switch converter, you don't need it in your case and we don't support it at the moment. [image]

Hi timoschilling, apologies for resurrecting a closed post, so if I wanted to wire this up like sg-generix has, I should use my 230v to 12v transformer still and remove the black part provided with the RGBW2 package? I won't be using an LED strip rather some LED bulbs as per the wiring diagram on Shelly's website.

timoschilling, 2022-08-19:
Using 12 or 24V bulbs is the same as using a strip.