mongoose-os-apps/shelly-homekit #833: Shelly2.5 can’t calibrate

lucaspinelli85, 2021-11-03:
Someone can tell me why i can't calibrate this roller shutter?

lucaspinelli85, 2021-11-03:
8596286040 shelly_debug.cpp:231    No log file, sending new entries
8596880815 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.0.199:49181
8600588334 shelly_main.cpp:513     Up 8600.57, HAP 0/2/12 ns 2, RAM: 26652/42160; st 38; 4.1: c:1 mp:0.00 mt_ms:1088 cp:44.58 tp:44.58 md:0 lemd:0 lhmd:0
8601149540 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.0.199:49181
8602138173 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.0.199:49181
8603161570 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.0.199:49181
8604142988 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.0.199:49181
8605147140 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.0.199:49181
8606160004 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.0.199:49181
8606758092 mg_rpc.c:306            Shelly.SetState via WS_in 192.168.0.199:49181
8606771679 shelly_hap_window_c:327 WC 1: State: idle -> precal0 (0 -> 10)
8606784285 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.0.199:49181
8606861710 shelly_hap_window_c:449 Begin calibration
8606891017 mgos_sys_config.c:232   Loading conf2.json
8607103338 mgos_sys_config.c:174   Saved to conf9.json
8607110152 shelly_output.cpp:63    Output 1: off -> on (precal0)
8607116589 shelly_hap_window_c:327 WC 1: State: precal0 -> cal0 (10 -> 11)
8607129881 shelly_hap_window_c:465 WC 1: P0 = 0.000
8607143164 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.0.199:49181
8607930169 shelly_hap_window_c:465 WC 1: P0 = 0.000
8608142201 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.0.199:49181
8608588762 shelly_main.cpp:513     Up 8608.58, HAP 0/2/12 ns 2, RAM: 26652/42160; st 38; 4.1: c:0 mp:0.00 mt_ms:1088 cp:44.58 tp:44.58 md:0 lemd:0 lhmd:0
8608730356 shelly_hap_window_c:465 WC 1: P0 = 0.000
8609155617 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.0.199:49181
8609529970 shelly_hap_window_c:465 WC 1: P0 = 0.000
8610151196 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.0.199:49181
8610329846 shelly_hap_window_c:465 WC 1: P0 = 0.000
8611130328 shelly_hap_window_c:465 WC 1: P0 = 0.000
8611155428 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.0.199:49181
8611930131 shelly_hap_window_c:465 WC 1: P0 = 0.000
8612130965 shelly_output.cpp:63    Output 1: on -> off (cal0)
8612137428 shelly_hap_window_c:327 WC 1: State: cal0 -> postcal0 (11 -> 12)
8612156451 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.0.199:49181
8612231435 shelly_hap_window_c:327 WC 1: State: postcal0 -> precal1 (12 -> 13)
8612330421 shelly_output.cpp:63    Output 2: off -> on (precal1)
8612336788 shelly_hap_window_c:327 WC 1: State: precal1 -> cal1 (13 -> 14)
8612430161 shelly_hap_window_c:496 WC 1: P1 = 0.000
8613153397 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.0.199:49181
8613230145 shelly_hap_window_c:496 WC 1: P1 = 0.000
8613430779 shelly_output.cpp:63    Output 2: on -> off (cal1)
8613438815 shelly_hap_window_c:500 WC 1: calibration done, move_time 1088, move_power 0.000
8613445070 shelly_hap_window_c:327 WC 1: State: cal1 -> postcal1 (14 -> 15)
8613531422 shelly_hap_window_c:336 WC 1: Cur pos 44.58 -> 0.00, P = -1.00
8613560662 mgos_sys_config.c:232   Loading conf2.json
8613694503 mgos_sys_config.c:174   Saved to conf9.json
8613703027 shelly_hap_window_c:347 WC 1: Tgt pos 44.58 -> 50.00 (postcal1)
8613709186 shelly_hap_window_c:327 WC 1: State: postcal1 -> idle (15 -> 0)
8613723204 shelly_hap_window_c:327 WC 1: State: idle -> move (0 -> 20)
8613729889 shelly_output.cpp:63    Output 1: off -> on (move)
8613735899 shelly_hap_window_c:327 WC 1: State: move -> rampup (20 -> 22)
8613830030 shelly_hap_window_c:549 P = 0.00 -> 0.00
8613836512 shelly_hap_window_c:327 WC 1: State: rampup -> moving (22 -> 23)
8614153670 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.0.199:49181
8614430901 shelly_output.cpp:63    Output 1: on -> off (moving)
8614437263 shelly_hap_window_c:327 WC 1: State: moving -> stop (23 -> 24)
8614554410 mgos_sys_config.c:232   Loading conf2.json
8614688952 mgos_sys_config.c:174   Saved to conf9.json
8614697033 shelly_hap_window_c:327 WC 1: State: stop -> stopping (24 -> 25)
8614713784 shelly_hap_window_c:327 WC 1: State: stopping -> idle (25 -> 0)
8614730307 shelly_hap_window_c:327 WC 1: State: idle -> move (0 -> 20)
8614830156 shelly_output.cpp:63    Output 2: off -> on (move)
8614836593 shelly_hap_window_c:327 WC 1: State: move -> rampup (20 -> 22)
8614930395 shelly_hap_window_c:549 P = 0.00 -> 0.00
8614936874 shelly_hap_window_c:327 WC 1: State: rampup -> moving (22 -> 23)
8615031062 shelly_output.cpp:63    Output 2: on -> off (moving)
8615037397 shelly_hap_window_c:327 WC 1: State: moving -> stop (23 -> 24)
8615153980 mgos_sys_config.c:232   Loading conf2.json
8615288606 mgos_sys_config.c:174   Saved to conf9.json
8615296688 shelly_hap_window_c:327 WC 1: State: stop -> stopping (24 -> 25)
8615310911 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.0.199:49181
[... 873 lines cut ...]
8665073858 mgos_sys_config.c:232   Loading conf2.json
8665255422 mgos_sys_config.c:174   Saved to conf9.json
8665263504 shelly_hap_window_c:327 WC 1: State: stop -> stopping (24 -> 25)
8665277322 shelly_hap_window_c:327 WC 1: State: stopping -> idle (25 -> 0)
8665377284 shelly_hap_window_c:327 WC 1: State: idle -> move (0 -> 20)
8665474952 shelly_output.cpp:63    Output 1: off -> on (move)
[... cut ...]
8666074700 shelly_output.cpp:63    Output 2: off -> on (move)
8666081121 shelly_hap_window_c:327 WC 1: State: move -> rampup (20 -> 22)
8666174983 shelly_hap_window_c:549 P = 0.00 -> 0.00
8666181465 shelly_hap_window_c:327 WC 1: State: rampup -> moving (22 -> 23)
8666276047 shelly_output.cpp:63    Output 2: on -> off (moving)
8666282382 shelly_hap_window_c:327 WC 1: State: moving -> stop (23 -> 24)
8666398840 mgos_sys_config.c:232   Loading conf2.json
8666533743 mgos_sys_config.c:174   Saved to conf9.json
8666541824 shelly_

tnt-1991, 2021-11-04:
I can confirm the problem on 2.10.1, rollback to 2.10 solves the issue

Logs:
shelly_hap_window_c:549 P = 0.00 -> 166.93
286538297 shelly_hap_window_c:549 P = 0.00 -> 166.93
286543082 shelly_hap_window_c:556 Failed to start moving
286549112 shelly_hap_window_c:327 WC 1: State: rampup -> stop (22 -> 24)
286637535 shelly_output.cpp:63    Output 1: on -> off (stop)
286662075 mgos_sys_config.c:232   Loading conf2.json
286835997 mgos_sys_config.c:174   Saved to conf9.json
286844043 shelly_hap_window_c:327 WC 1: State: stop -> stopping (24 -> 25)
286857703 shelly_hap_window_c:327 WC 1: State: stopping -> idle (25 -> 0)
286955927 shelly_hap_window_c:327 WC 1: State: idle -> error (0 -> 100)
287056955 shelly_hap_window_c:347 WC 1: Tgt pos 100.00 -> 49.64 (error)
287063294 shelly_hap_window_c:327 WC 1: State: error -> idle (100 -> 0)
288060992 mongoose.c:12126        Failed to resolve 'time.google.com', server 192.168.0.1
291597311 shelly_main.cpp:513     Up 291.58, HAP 0/1/12 ns 1, RAM: 27416/42160; st 40; 4.1: c:1 mp:166.93 mt_ms:27397 cp:49.64 tp:49.64 md:0 lemd:1 lhmd:0
299597334 shelly_main.cpp:513     Up 299.58, HAP 0/1/12 ns 1, RAM: 27416/42160; st 40; 4.1: c:1 mp:166.93 mt_ms:27397 cp:49.64 tp:49.64 md:0 lemd:1 lhmd:0
307597182 shelly_main.cpp:513     Up 307.58, HAP 0/1/12 ns 1, RAM: 27416/42160; st 40; 4.1: c:1 mp:166.93 mt_ms:27397 cp:49.64 tp:49.64 md:0 lemd:1 lhmd:0

closed 2021-11-05 by lucaspinelli85