mongoose-os-apps/shelly-homekit #834: changeover circuit causes flashing

paulhfischer, 2021-11-04:
i'm using shelly-homekit on my shelly 1 and have set it up as a [changeover circuit](https://forum.creationx.de/index.php?attachment/2472-shelly1-wechselschaltung-png/). It works fine with the stock firmware, but as soon as I install the homekit-patch and actuate a switch, the connected lamp starts to flicker (instantly turning off and on again).

timoschilling, 2021-11-05:
Can you post a screenshot from the switch settings please.

paulhfischer, 2021-11-05:
these are my current settings, i also tried with input mode `edge`:

[image]
[image]

timoschilling, 2021-11-05:
Try "Input Mode: Toggle"

paulhfischer, 2021-11-06:
okay i'm sorry for the issue - seems like a hardware issue. thought it was related to the firmware as it appeared after i flashed the firmware, but it had the same behaviour with the stock firmware i reinstalled later. i installed a new shelly today and have no issue with the homekit firmware so far with this wiring.

thank you anyway for your help :)

closed 2021-11-06 by paulhfischer