mongoose-os-apps/shelly-homekit #842: Shelly25 after Update to 2.10.1 restarts very often and is not available

SAPSLL, 2021-11-14:
Hello,

since the update to 2.10.1, the Shelly 2.5 is regularly inaccessible and keeps restarting.
It also keeps switching itself off, but I'm not sure if this is simply due to the restart that it performs every few minutes.

I am attaching the log and hope you can help me?

Greetings SAPSLL

162323279 shelly_debug.cpp:231    No log file, sending new entries
168606095 shelly_main.cpp:513     Up 168.59, HAP 0/2/12 ns 2, RAM: 25536/42160; st 58; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
176477710 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.178.123:53290
176606534 shelly_main.cpp:513     Up 176.59, HAP 0/2/12 ns 2, RAM: 25536/42160; st 58; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
177452726 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.178.123:53290
177479999 mgos_mongoose.c:66      New heap free LWM: 19976
178445252 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.178.123:53290
184606014 shelly_main.cpp:513     Up 184.59, HAP 0/2/12 ns 2, RAM: 26132/42160; st 58; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
192606132 shelly_main.cpp:513     Up 192.59, HAP 0/2/12 ns 2, RAM: 26132/42160; st 58; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
200606099 shelly_main.cpp:513     Up 200.59, HAP 0/2/12 ns 2, RAM: 26132/42160; st 58; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
208605913 shelly_main.cpp:513     Up 208.59, HAP 0/2/12 ns 2, RAM: 26132/42160; st 58; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
216605949 shelly_main.cpp:513     Up 216.59, HAP 0/2/12 ns 2, RAM: 26132/42160; st 58; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
224605836 shelly_main.cpp:513     Up 224.59, HAP 0/2/12 ns 2, RAM: 26132/42160; st 58; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
232605906 shelly_main.cpp:513     Up 232.59, HAP 0/2/12 ns 2, RAM: 26132/42160; st 58; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
240606200 shelly_main.cpp:513     Up 240.59, HAP 0/2/12 ns 2, RAM: 26132/42160; st 58; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
248606106 shelly_main.cpp:513     Up 248.59, HAP 0/2/12 ns 2, RAM: 26132/42160; st 58; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
256605937 shelly_main.cpp:513     Up 256.59, HAP 0/2/12 ns 2, RAM: 26132/42160; st 58; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
264605893 shelly_main.cpp:513     Up 264.59, HAP 0/2/12 ns 2, RAM: 26132/42160; st 58; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
272605738 shelly_main.cpp:513     Up 272.59, HAP 0/2/12 ns 2, RAM: 26132/42160; st 58; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
280605810 shelly_main.cpp:513     Up 280.59, HAP 0/2/12 ns 2, RAM: 26132/42160; st 58; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
288605979 shelly_main.cpp:513     Up 288.59, HAP 0/2/12 ns 2, RAM: 26132/42160; st 58; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
296605983 shelly_main.cpp:513     Up 296.59, HAP 0/2/12 ns 2, RAM: 26132/42160; st 58; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
304605698 shelly_main.cpp:513     Up 304.59, HAP 0/2/12 ns 2, RAM: 26132/42160; st 58; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
312605882 shelly_main.cpp:513     Up 312.59, HAP 0/2/12 ns 2, RAM: 26132/42160; st 58; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
320605826 shelly_main.cpp:513     Up 320.59, HAP 0/2/12 ns 2, RAM: 26132/42160; st 58; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
328605904 shelly_main.cpp:513     Up 328.59, HAP 0/2/12 ns 2, RAM: 26132/42160; st 58; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
336605970 shelly_main.cpp:513     Up 336.59, HAP 0/2/12 ns 2, RAM: 26132/42160; st 58; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
344605987 shelly_main.cpp:513     Up 344.59, HAP 0/2/12 ns 2, RAM: 26132/42160; st 58; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
352605810 shelly_main.cpp:513     Up 352.59, HAP 0/2/12 ns 2, RAM: 26132/42160; st 58; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
360606024 shelly_main.cpp:513     Up 360.59, HAP 0/2/12 ns 2, RAM: 26132/42160; st 58; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
368606072 shelly_main.cpp:513     Up 368.59, HAP 0/2/12 ns 2, RAM: 26132/42160; st 58; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
376605914 shelly_main.cpp:513     Up 376.59, HAP 0/2/12 ns 2, RAM: 26132/42160; st 58; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
384606246 shelly_main.cpp:513     Up 384.59, HAP 0/2/12 ns 2, RAM: 26132/42160; st 59; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
392606236 shelly_main.cpp:513     Up 392.59, HAP 0/2/12 ns 2, RAM: 26132/42160; st 58; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
400605926 shelly_main.cpp:513     Up 400.59, HAP 0/2/12 ns 2, RAM: 26132/42160; st 58; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
408605870 shelly_main.cpp:513     Up 408.59, HAP 0/2/12 ns 2, RAM: 26132/42160; st 59; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
416605870 shelly_main.cpp:513     Up 416.59, HAP 0/2/12 ns 2, RAM: 26132/42160; st 58; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
424606047 shelly_main.cpp:513     Up 424.59, HAP 0/2/12 ns 2, RAM: 25960/42160; st 59; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
431354656 mgos_http_server.c:180  0x3fff5a64 HTTP connection from 192.168.178.123:53363
431381897 mgos_http_server.c:180  0x3fff5d7c HTTP connection from 192.168.178.123:53364
431388576 mgos_mongoose.c:66      New heap free LWM: 19456
431417274 mgos_mongoose.c:66      New heap free LWM: 18280
431582115 mgos_http_server.c:180  0x3fff5b1c HTTP connection from 192.168.178.123:53365
431613677 mg_rpc.c:306            Shelly.GetInfo via WS_in 192.168.178.123:53365
431638755 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.178.123:53365
432606148 shelly_main.cpp:513     Up 432.59, HAP 0/2/12 ns 2, RAM: 24740/42160; st 58; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
432618598 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.178.123:53365
433580632 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.178.123:53365
434582194 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.178.123:53365
435581704 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.178.123:53365
436582885 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.178.123:53365
440606046 shelly_main.cpp:513     Up 440.59, HAP 0/2/12 ns 2, RAM: 25328/42160; st 59; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
448605760 shelly_main.cpp:513     Up 448.59, HAP 0/2/12 ns 2, RAM: 25696/42160; st 58; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
456605901 shelly_main.cpp:513     Up 456.59, HAP 0/2/12 ns 2, RAM: 25696/42160; st 58; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
464605893 shelly_main.cpp:513     Up 464.59, HAP 0/2/12 ns 2, RAM: 25696/42160; st 59; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
465518139 mgos_http_server.c:180  0x3fff5cc4 HTTP connection from 192.168.178.123:53374
465529589 mgos_http_server.c:180  0x3fff586c HTTP connection from 192.168.178.123:53375
466577023 mgos_http_server.c:180  0x3fff586c HTTP connection from 192.168.178.123:53376
466605032 mg_rpc.c:306            Shelly.GetInfo via WS_in 192.168.178.123:53376
466642247 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.178.123:53376
468007636 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.178.123:53376
[... 30 lines cut ...]
616605828 shelly_main.cpp:513     Up 616.59, HAP 0/2/12 ns 2, RAM: 25096/42160; st 59; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
624606499 shelly_main.cpp:513     Up 624.59, HAP 0/2/12 ns 2, RAM: 25096/42160; st 59; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
632606335 shelly_main.cpp:513     Up 632.59, HAP 0/2/12 ns 2, RAM: 25096/42160; st 58; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
[... cut ...]
784605940 shelly_main.cpp:513     Up 784.59, HAP 0/2/12 ns 2, RAM: 25096/42160; st 59; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
792606154 shelly_main.cpp:513     Up 792.59, HAP 0/2/12 ns 2, RAM: 25096/42160; st 59; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
800605852 shelly_main.cpp:513     Up 800.59, HAP 0/2/12 ns 2, RAM: 25096/42160; st 59; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0
808605804 shelly_main.cpp:513     Up 808.59, HAP 0/2/12 ns 2, RAM: 25096/42160; st 59; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:0 in_st:0 inm:1 ininv:0

timoschilling, 2021-11-14:
@rojer ?

Stundenplan7b, 2021-11-14:
Same Problem - one of ten Shelly 2.5

rojer, 2021-11-14:
@timoschilling i was meaning to do library changes bisection between 2.10.0 and 2.10.1 but haven't had the time yet.

SAPSLL, 2021-11-24:
Hello together,

do you have any news or Work-A-Rounds for me?
We've been sitting in the dark at dinner for a week now and slowly my girlfriend is losing patience. I'm afraid I'll have to remove the shelly otherwise :(

many greetings

romanbalsiger, 2021-11-24:
the one that made the same problems in my house, actually died a few days later...

SAPSLL, 2021-11-24:
are sure that's died? because my one is also completely down, until I take the power fuse and restart it. Than is running for a few hours again

romanbalsiger, 2021-11-24:
> are sure that's died? because my one is also completely down, until I take the power fuse and restart it. Than is running for a few hours again

Yes, at least it's  not usable anymore now. the problem started with the update to 2.10.1 from 2.9 - constant reboot. On 2.9 it didn't have any problems at all. So i downgraded from 2.10.1 to 2.9 and first it worked again like before.
then it started to behave the same way under 2.9 and i noticed a hissing noise coming from the shelly when connected to mains. And at last it crashed and didn't reboot anymore. it's still connected to mains and now and then it appears in homekit and the web frontend ist reachable, but most of the time it's dead.

SAPSLL, 2021-11-24:
How can I return to the previous version?

romanbalsiger, 2021-11-24:
you can use the shell script:

[https://github.com/mongoose-os-apps/shelly-homekit/wiki/Flashing#Script](https://github.com/mongoose-os-apps/shelly-homekit/wiki/Flashing#Script)

romanbalsiger, 2021-11-24:
i used this command in the terminal
python3 flash-shelly.py shelly-OG-1.local -V 2.9.0

timoschilling, 2021-11-24:
@SAPSLL  you can download older releases here https://github.com/mongoose-os-apps/shelly-homekit/releases and upload them via the web UI

SAPSLL, 2021-11-24:
Thank you very much! Now I was able to return to the stable version and the Shelly works again 😄

SAPSLL, 2021-11-24:
> @SAPSLL you can download older releases here https://github.com/mongoose-os-apps/shelly-homekit/releases and upload them via the web UI

Good to know 😂 so I first had to install python and to make it run. But after a few minutes i got it.

Ioan-Popovici, 2022-01-20:
Same problem, after updating to the last version my shelly 25 switches are stuck in a reboot loop. I can't access them in order to downgrade or revert to stock.
Tried reseting them with no luck.

Can we maybe add a feature to trigger revert to stock if pressing the reset button 10 times or something?

timoschilling, 2022-01-20:
@rojer do you any idea?

tdexwjgb, 2022-02-06:
> then it started to behave the same way under 2.9 and i noticed a hissing noise coming from the shelly when connected to mains. And at last it crashed and didn't reboot anymore. it's still connected [...]

I think this problem is somehow related to this: https://github.com/mongoose-os-apps/shelly-homekit/issues/844

See my latest reply there: I also had the reboot loop with FW 2.10.1, but luckily was able to revert to 2.10.0. Mine makes a hissing noise too, no matter on which FW. Removed it from the roller shutter for the moment just to be sure not to damage the motor when it finally dies...

closed 2023-03-15 by github-actions[bot]