mongoose-os-apps/shelly-homekit #843: Possible reason for Shelly 2.5 repeatedly reset/reboot itself---hardware failure

kerhbal, 2021-11-14:
Hi all, sorry if I put the issue in wrong place, since I find similar issues here. Please let me know if somewhere else is better for me to discuss this. 

Possible related issues:
https://github.com/mongoose-os-apps/shelly-homekit/issues/795
https://github.com/mongoose-os-apps/shelly-homekit/issues/252

In my case the repeatedly reset is more specific:
Action to trigger device reset: 
- Try to open the url of the device, this action will have a 70~80% chance to trigger a device reset
- Or just wait for some time, a device reset will be seen. Though this way the time can vary from minutes to hours

And since I have 6 Shelly 2.5, was wondering If something can be found on hardware.

Prepare: Latest firmware(v1.11.7), shelly 2.5 is connected to 2 independent switches/lights at 110v, turn on lights on both channels, select client mode to connect to my wifi. for multiple shelly 2.5, they have exact same setting (button type, wifi network, access point etc) and exact same physical location.

**Result**
I tried to load the device url 20 times for each device, then remove it and install next one.
3 of shelly2.5 is totally fine, 3 of them has the problem. And the problematic one all come with a prefix "98F4ABF2" in device name(or device version?) as a part of default wifi name.

> And for logs, first of all I don't see anything in `.../debug/core`, and somehow it will not tail the log in `.../debug/log`, I need to manually refresh it, so after the reset I still get nothing in `.../debug/log`.

Hope this is helpful regarding the similar issue ppl are experiencing.

timoschilling, 2021-11-15:
Thanks for finding and posting that detail! Do you know at wich time you bayed the problematic once?

kerhbal, 2021-11-15:
@timoschilling from Aug to Oct 2021, since all the devices looks identical, once they are unpacked, I've no idea which are from Aug and which are from Oct... Though I bought from a third-party vendor, so probably not the case if you buy from Shelly directly.

closed 2021-12-23 by github-actions[bot]