mongoose-os-apps/shelly-homekit #844: shelly 2.5 no action, need to recalibrate since 2.10.1

MagOO33, 2021-11-15:
Hello,
my problem #820 is still present on one of my shellies2.5 since I flashed it with 2.10.1.
It happened 2 more time since last refresh.
Here is the log (no action, recalibrate and finally OK):

37602069834 shelly_debug.cpp:231    No log file, sending new entries
37604550355 shelly_main.cpp:513     Up 37604.54, HAP 0/7/12 ns 7, RAM: 23944/42160; st 41; 4.1: c:1 mp:105.13 mt_ms:21888 cp:0.00 tp:0.00 md:0 lemd:0 lhmd:0
37606841486 HAPPlatformTCPStrea:121 0x3fff6154 192.168.1.81:59740 Incoming HAP connection, ns 0/7/12
37606851485 HAPPlatformTCPStrea:276 0x3fff6154 192.168.1.81:59740 Accepted HAP connection, ns 0/8/12 ts 0x3fff6a9c
37608534751 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
37608554437 mgos_mongoose.c:66      New heap free LWM: 18672
37609515880 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
37609535625 mgos_mongoose.c:66      New heap free LWM: 18336
37610516308 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
37611516349 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
37612517549 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
37612555045 shelly_main.cpp:513     Up 37612.54, HAP 0/8/12 ns 8, RAM: 22072/42160; st 41; 4.1: c:1 mp:105.13 mt_ms:21888 cp:0.00 tp:0.00 md:0 lemd:0 lhmd:0
37613517838 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
37614518559 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
37615519379 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
37616520504 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
37617521706 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
37618522565 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
37619524101 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
37620531493 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
37620563271 shelly_main.cpp:513     Up 37620.55, HAP 0/8/12 ns 8, RAM: 22072/42160; st 41; 4.1: c:1 mp:105.13 mt_ms:21888 cp:0.00 tp:0.00 md:0 lemd:0 lhmd:0
37621525497 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
37622526649 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
37623526831 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
37624527905 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
37625529232 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
37626530197 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
37627530944 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
37628532927 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
37628564570 shelly_main.cpp:513     Up 37628.55, HAP 0/8/12 ns 8, RAM: 21952/42160; st 41; 4.1: c:1 mp:105.13 mt_ms:21888 cp:0.00 tp:0.00 md:0 lemd:0 lhmd:0
37629533184 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
37630534004 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
37631537373 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
37632536239 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
37633541813 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
37634538105 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
37635539000 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
37636539498 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
37636571585 shelly_main.cpp:513     Up 37636.56, HAP 0/8/12 ns 8, RAM: 22072/42160; st 41; 4.1: c:1 mp:105.13 mt_ms:21888 cp:0.00 tp:0.00 md:0 lemd:0 lhmd:0
37637540678 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
37638540636 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
37639541526 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
37640542471 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
37641544195 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
37642543747 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
37643546028 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
37644550209 shelly_main.cpp:513     Up 37644.54, HAP 0/8/12 ns 8, RAM: 22988/42160; st 41; 4.1: c:1 mp:105.13 mt_ms:21888 cp:0.00 tp:0.00 md:0 lemd:0 lhmd:0
37644598121 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
37645546267 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
37646547104 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
37647549236 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
37648549219 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
37649569297 HAPPlatformTCPStrea:105 0x3fff5c3c 192.168.1.100:50814 HAP connection closed, ns 0/8/12 i 2
37649577342 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff44b4 nc 0x0 0/7/12
37649592123 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
37649612852 mgos_mongoose.c:66      New heap free LWM: 18184
37650550533 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
37651552784 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
37652550219 shelly_main.cpp:513     Up 37652.54, HAP 0/7/12 ns 7, RAM: 23380/42160; st 41; 4.1: c:1 mp:105.13 mt_ms:21888 cp:0.00 tp:0.00 md:0 lemd:0 lhmd:0
37652562306 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
37653552833 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
37654554069 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
37655554460 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
37656556141 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
37657556446 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.2:51230
[... 269 lines cut ...]
37743512898 shelly_hap_window_c:336 WC 1: Cur pos 39.70 -> 39.25, P = 109.01
37743611445 shelly_hap_window_c:599 Moving to 0, p 108.95
37744312968 shelly_hap_window_c:336 WC 1: Cur pos 36.05 -> 35.59, P = 108.66
37744411575 shelly_hap_window_c:599 Moving to 0, p 108.84
37745113232 shelly_hap_window_c:336 WC 1: Cur pos 32.39 -> 31.94, P = 108.91
37745211443 shelly_hap_window_c:599 Moving to 0, p 108.74
37745913008 shelly_hap_window_c:336 WC 1: Cur pos 28.74 -> 28.28, P = 108.67
37746011744 shelly_hap_window_c:599 Moving to 0, p 108.65
37746713121 shelly_hap_window_c:336 WC 1: Cur pos 25.07 -> 24.63, P = 108.45
37746811801 shelly_hap_window_c:599 Moving to 0, p 108.45
37747513139 shelly_hap_window_c:336 WC 1: Cur pos 21.43 -> 20.97, P = 109.03
37747611326 shelly_hap_window_c:599 Moving to 0, p 109.27
37748313009 shelly_hap_window_c:336 WC 1: Cur pos 17.77 -> 17.32, P = 109.60
37748412519 shelly_hap_window_c:599 Moving to 0, p 109.48
37748550426 shelly_main.cpp:513     Up 37748.54, HAP 0/8/12 ns 8, RAM: 23620/42160; st 42; 4.1: c:1 mp:109.40 mt_ms:21888 cp:16.40 tp:0.00 md:2 lemd:0 lhmd:0
37749113124 shelly_hap_window_c:336 WC 1: Cur pos 14.11 -> 13.66, P = 109.77
37749211554 shelly_hap_window_c:599 Moving to 0, p 109.66
37750076887 shelly_hap_window_c:336 WC 1: Cur pos 9.71 -> 9.26, P = 110.27
37750175046 shelly_hap_window_c:599 Moving to 0, p 110.91
37750375278 shelly_output.cpp:63    Output 2: on -> off (moving)
37750382106 shelly_hap_window_c:327 WC 1: State: moving -> stop (23 -> 24)
37750496918 mgos_sys_config.c:232   Loading conf2.json
37750634748 mgos_sys_config.c:174   Saved to conf9.json
37750642815 shelly_hap_window_c:327 WC 1: State: stop -> stopping (24 -> 25)
37750658214 shelly_hap_window_c:327 WC 1: State: stopping -> idle (25 -> 0)

MagOO33, 2021-12-12:
I finally bought a new shelly 2.5 to replace the faulty one since 2.10.1.
Seems working now.
But, today, another shelly 2.5 that was working before is falling with the same problem.
Roller shutter are opening normally at morning but one shows a closed HAP status.
No way to shut it, only calibrate does fix it for a while.
Is it possible to have a new 2.10 based firmware with only the temperature fix for shelly 2.5 ?

Xaruqui, 2022-01-11:
Hi. I have exactly the same problem. At first I also thought it was a faulty Shelly and kept calibrating it every once in a while but now it has happened to another one (out of x12 2.5). After calibrating it it does NOT save the calibration since the calibration button is still there after doing it. Going back to stock firmware fixes the issue so the real problem is in 2.10.1 and not in the hardware.

I'm open to test, guys. Just throw some questions and I'll try to assess.

MagOO33, 2022-01-11:
Hi,
For me, the problem is that the shelly is rebooting while it closes the relay to open or close the roller shutter.
After that, it shows the wrong state and I need to recalibrate.
I think, the 2.10.1 firmware reduced the power consumption of the shelly so that it is more sensible to inductive voltage spikes.
For me, changing the worst 2.5 is enough. I have no problem now.
But, look at this : [https://shop.shelly.cloud/rc-snubber-wifi-smart-home-automation](url)
They admit there may be inductive glitchs in some conditions.
I bought some to modify my wirings if needed, but no need for the moment.

xtr-ss, 2022-01-14:
Hi, having the very same problem since a month ago or so (i think i updated all my Shellys as i bought a new one to control lights). Need to Recalibrate one of my 3 Shelly 2.5 almost every week (stops working until recalibration). Its not the one further from the AP, and it has never happened on the other two connected to rolling shutters.

The only strange symptom i see is that i order "roll down to 30%" almost every night and in the morning i find the shutter completely closed and does not respond nor to homekit nor button. Saw some post saying that the shelly reboots while rolling down? Is that a confirmed situation i should look upon?

If there are some suggestion to how to log this behavior i'll try.

MagOO33, 2022-01-14:
Hi,
if you have a fixed time to roll down to 30%, you can compare it to the last reboot time in web UI.
For me, it was the same, so that I could suspect an inductive glitch at the closing of the internal relay.
If so, I think you should try the capacitors (link above)

timoschilling, 2022-01-14:
@rojer can maybe the threshold change which is in the 2.11 fix this problem?

xtr-ss, 2022-01-14:
> Hi, if you have a fixed time to roll down to 30%, you can compare it to the last reboot time in web UI. For me, it was the same, so that I could suspect an inductive glitch at the closing of the internal relay. If so, I think you should try the capacitors (link above)

I'll try to get the RC Snubbers and get them at least on the Rolling shutter that is failing, (assuming that at any other given time any of the others could begin doing the same problem) (the 3 shutters have the same motor and same weight/length, only one is showing this behavior).

Thanks for the tip.

Xaruqui, 2022-01-14:
> @rojer can maybe the threshold change which is in the 2.11 fix this problem?

I updated one of my two faulty shelly's 2.5 to the last 2.11.0-alpha8 yesterday in order to check, but it might take some days/weeks to come to a conclusion.

rojer, 2022-01-14:
@timoschilling no, this seems unrelated.

so, reading the log i see that we turn on the output and... nothing happens. eventually, we give up and go into the error state:
```
37687077885 shelly_hap_window_c:347 WC 1: Tgt pos 0.00 -> 100.00 (RPC)
...
37687139187 shelly_hap_window_c:327 WC 1: State: idle -> move (0 -> 20)
37687209175 shelly_output.cpp:63 Output 1: off -> on (move)
37687215988 shelly_hap_window_c:327 WC 1: State: move -> rampup (20 -> 22)
37687310000 shelly_hap_window_c:549 P = 0.00 -> 105.13
...
37692309987 shelly_hap_window_c:549 P = 0.00 -> 105.13
37692314822 shelly_hap_window_c:556 Failed to start moving
37692320863 shelly_hap_window_c:327 WC 1: State: rampup -> stop (22 -> 24)
...
37692582521 shelly_hap_window_c:327 WC 1: State: stop -> stopping (24 -> 25)
37692615180 shelly_hap_window_c:327 WC 1: State: stopping -> idle (25 -> 0)
37692713566 shelly_hap_window_c:327 WC 1: State: idle -> error (0 -> 100)
```

i just need to verify two things:
 * the curtain was fully closed at the time
 * output 1 of the shelly opens the curtain
 
is that correct?

Xaruqui, 2022-01-14:
Hi master. I'll reply from my side, I guess that @timoschilling will do also from his.

So, in my case:

- Yes, initial state is fully closed when this happens (it closes by the time we go to bed but it won't open anymore in the morning)
- No, output 1 is not opening the roller shutter since it behaves as if it wasn't calibrated, as soon as it's re-calibrated it starts working again. In the morning HAP shows that the roller shutter is open, nonetheless it's fully closed.

rojer, 2022-01-14:
@Xaruqui i am not sure your issue is the same as the one @MagOO33 is having, in this case i was looking at his log and need confirmation from him.

but in general, if reboot / failure happens at the time when motor is turned on or off, that is a clue to the inductive load spike issue and that you need to install a capacitor.

Xaruqui, 2022-01-14:
@rojer Understood. This has something to do with 2.10.1 though, I've been using my shelly's for a very long time and this happened to two of them repeatedly since the last update (same connected hardware, nothing changed).

I'm getting those RC snubbers anyway ;)

rojer, 2022-01-14:
@Xaruqui ok, let's focus on that. if you go back to 2.10.0, does it still happen? you can find the release history [here](https://github.com/mongoose-os-apps/shelly-homekit/releases), if you can pinpoint exactly the release where it starts happening in your setup, that will be very helpful.

MagOO33, 2022-01-14:
Xaruqui case looks like mine.
As device rebooted, no log could bring a clue for me but I saw that the time of the last reboot in UI was the time of the last automation.
I found that the device lost is current state when I wanted to use it hours after the reboot which is the real culprit.
But I though it was because of the new firmware as it happened only with 2.10.1.
This is strange, my guess is that it's all about power consumption and voltage levels.
2.10.1 lowers the temperature of my 2.5 devices about 8°C ...
For me, it's definitively an inductive problem that occurs with some touchy devices, in some case (temperature of the motor ...)
Perhaps it could be solved be firmware, playing with latency, voltages ?
... but the original firmware seems to have those kind of problems as Shelly is selling capacitors now.

rojer, 2022-01-14:
yes, the entire point of 2.10.1 was to solve the issue of incorrectly configured pin causing higher power consumption and as a result - increased temperature. it may result in different response to power spikes, i don't really know...
we can't really influence much in terms of hardware. one thing we can potentially do on a 2.5 is time relay state changes to occur during voltage zero crossing to reduce current spikes

rojer, 2022-01-14:
@MagOO33 meanwhile, can you also confirm that this change happens between 2.10.0 and 2.10.1?

MagOO33, 2022-01-14:
I confirm

xtr-ss, 2022-01-15:
Asked the Rolling shutter to go 60%, and went fully open...
Device ID: shellyswitch25-Dorm1
Uptime: 0:00:00:35
Temperature: 36°C

Indeed, rebooted during this process. will order a batch of snubbers and see...

rojer, 2022-01-15:
did it leave a core dump behind? check `http://a.b.c.d/debug/core`. when firmware crashes, it records a dump. if no dump was left, then it was a reset due to power spike.

xtr-ss, 2022-01-15:
> did it leave a core dump behind? check `http://a.b.c.d/debug/core`. when firmware crashes, it records a dump. if no dump was left, then it was most likely a reset due to power spike.

So if the first line is:
"12561989 shelly_debug.cpp:231    No log file, sending new entries" 

my guess this is a "no-dump scenario" ?

Xaruqui, 2022-01-17:
@rojer I installed the v2.11.0-alpha8 in one of my faulty Shellys 2.5 three days ago, only to check if the version solved the issue. It didn't, the same happened as with v2.10.1. Here below the core dump:

--- BEGIN CORE DUMP ---
{"app": "switch25", "arch": "esp8266", "version": "2.11.0-alpha8", "build_id": "20220109-231733/2.10.1-42-gb712fa5-master", "build_ts": "2022-01-09T23:17:33Z",
"build_image": "docker.io/mgos/esp8266-build:3.0.5_b57142b8-r1", "uptime": 53136811394,
"REGS": {"addr": 1073740176, "data": "
8p4pQAD7/z8AAAAAAAAAAEAAAAAAAAAAQAAAAAQAAAC4JP8/kAEAAAkAAAAgAAAAAAAAANnr/j/Z6/4/AAAAALLhAEAEAAAAAAAAAAAAAAAAAAAAMAAAAA==", "crc32": 1757806051},
"DRAM": {"addr": 1073643520, "data": "
GckgQAiA/j8CAAAA6CP/P1Ak/z+4JP8/AAAAAAAAAAABAAAAAAAAAB3EIUDEJ/8/MC0iQAAAAAAAAAAAAAAAAAAAAABsWP8/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAADcI/8/AAAAAAAAAAAAAAAA
AAAAAAAAAAA0AAAAzvv//8sDAAABAAAAAAAAADUAAADO+///ywMAAAEAAAAAAAAATV8hQE1fIUAkFv8/AQAAAAAAAABKAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAASgAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAEFTQ0lJAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAksohQDvEIUA7xCFAO8QhQDvEIUA7xCFAO8QhQDvEIUA7xCFAO8QhQP//////////////////AAABAAAAUPEhQHzxIUAAAAAAAAAAAAEAAAD9AQAA
/////wEBAAAAAAAAAAAAAAEAAAAAAAAAAAAAAAAAAACCAAAAAwAAAIQAAAACAAAAiwAAAAEAAACWAAAAAAAAADYAAAAAAAAAMAAAAAEAAAAkAAAAAgAAABgAAAADAAAAEgAAAAQAAAAMAAAABQAAAAkAAAAGAAAA
BgAAAAcAAACLAAAACAAAAIQAAAAJAAAAggAAAAoAAAAAAAAAAAAAAEgAAAABAAAAQQAAAAIAAAA5AAAAAwAAACsAAAAEAAAAHAAAAAUAAAAVAAAABgAAAA4AAAAHAAAABwAAAAgAAACLAAAACQAAAIQAAAAKAAAA
ggAAAAsAAAAAAAAAAAAAAIIAAAAAAAAAhAAAAAEAAACLAAAAAgAAAJYAAAADAAAAMAAAAAgAAAAYAAAACQAAAAwAAAAKAAAABgAAAAsAAAA2AAAADAAAACQAAAANAAAAEgAAAA4AAAAJAAAADwAAAAcAAAAQAAAA
DgAAABEAAAAVAAAAEgAAABwAAAATAAAAKwAAABQAAAA5AAAAFQAAAEEAAAAWAAAASAAAABcAAAABAAAAAAAAAAAAAAAAAAAAAQIDBAUGBwgJCgsMDQ4AAAQAAAAAAAAAfNQjQHACI0B81CNAfNQjQOgBI0DcVCNA
dDYjQAIAAAACAAAAwMYtAGQAogDAJwkA8AAAALio/j8BAQAAsNz/PwAAAAAAAAAAOgECAd8A1QA6AaIAfwB1ACwALAAwADwALAAsADAAPAA8ADAAMAAsACwALAAsACwAPAAwADAALAAsACwALAAsAAsCCwILAgAF
CwAAAAAAAAAAAwADAAoACgALAAsABwAHAAgACAAAAAAAAAAAAAAAAB8CFgILAgAFC3IAABcCFQILAgAFC3IBABYCFAILAgAFC3ICABUCEwILAgAFC2ADABQCEgILAgAFC1UEABMCEgILAgAFC0sFABICEQIQAgsF
C0AGABECEAILAgsFCzsHABACCwILAgsFCzIIAAAAAAAfAhYCCwIABQtyAAAXAhUCCwIABQtyAQAWAhQCCwIABQtyAgAVAhMCCwIABQtgAwAUAhICCwIABQtVBAATAhICCwIABQtLBQASAhECCwIABQtABgARAgsC
BQIABQs7BwAQAgsCBQIABQsyCAAGAgUCAAIABQs5CQAFAgACAAIABQs5CgAAAgACAAIABQsACwAMAggCCwIABQtyAAAIAg0CCwIABQtlAQANAgkCCwIABQtcAgAJAg4CCwIABQtYAwAOAgoCCwIABQtSBAAKAg8C
CwILBQtLBQAPAgsCCwILBQs4BgALAgsCCwILBQtABwAMAggCCwIABQtyAAAIAg0CCwIABQtlAQENAgkCCwIABQtcAgAJAg4CCwIABQtYAwAOAgoCCwIABQtSBAAKAgsCBQIABQtLBQAPAgsCBQIABQs4BgALAgUC
AAIABQtABwAGAgUCAAIABQs5CAAFAgACAAIABQs5CQAAAgACAAIABQsACgAAAAAAAAAAAAAAAAAHAgYCBQIABQtAAAAGAgUCAAIABQs5AQAFAgACAAIABQs5AgAAAgACAAIABQsAAwAAsf4/AJABAFkA8AsEdyRA
DIskQGh1JEAs4SRA7GAAQNBgAEAoqSRAJOEkQE5ONABBAAAAAgAAAKgAAAABAAAAsFchQP////8BAAAA6FsQQAAAAABkELcdyCBuO6ww2SaQQdx29FFra1hhsk08cQVQIIO47USTD/Doo9bWjLNhy7DCZJvU0tOG
eOIKoBzyvb0AAAAAKcEhQDLBIUA5wSFAQ8EhQE7BIUBgwSFAqKwgQHDBIUB5wSFAhsEhQJDBIUCdwSFAqcEhQLTBIUDCwSFAz8EhQEPBIUDWwSFA3sEhQPDBIUAFwiFAEsIhQLjx/j8MAAAAtOv+PwAGAAABAAAA
kOEAQP//////////ytI6qSDRgkuOEP3g/FvbQzGr5XSgT+II4kbZusCICXdsXWF6VxfhuwwgexcYKx9SZGrIPnMCdthkCIrZBvov8Wvu0hom0uPOnWElSuCUjB7XMwnbjK71q8fk4aaFD5ezfQwGXVdx6ooK79tY
BIX77GS6HN+rIVWoM3pQBA0XM60txKqKWo5yFRAF+pgYJtIV5WqV6nxJlTkYF1iV9ssr3slSTG/wXcW1j6IH7KKDJ5sDhg4YLHee4zvONi5GXpAyfCEYyghsdPEEmLxKTjUMZ22WlnAHKdWeu1KFIFbzYhyWraPc
I11lg1/PJP2oPxZpmtNVHDZI2pgFv2OhuHwAwj1b5OxRZihJ5h9LfBEkn66ln4la+2s47u23BvS2XP8La+03pulCTPTGfl5idrWF5EXCUW1tNeFPNxRf8m0KKzAbQzrNsxmV790ENI55CEpRIpsTO6a+CwJ0zGeK
CE4CKdEc3ICLYsbENMJoIaLaD8n//////////0Ki39FXqJGrSSf26YXEFOPTj/5IlXIO8O/5nNIpNqgG9r4kx7ee0VkVz34aFpJDXCAPu/xiTi6gloMdpDXWjy1eBzjaZVIoM4r5SnzhyCkTUUbWoXocdgUAhtki
/ugoAAanOzNCfi6EHW/xUI5I/BEg8Lwo1jSFB9dwCBpcJs25wIZvHW9H16b2P73DeQuM8erwK1FMxyNoagML6mWOcCaKUguG5dV8XCepvb/02TSYZuqW9noWFe28U1Q3WLmhXSkjAACPrCBATcwhQFjMIUAgwSFA
ecEhQEy2IEC64zYBAAQAAHgAAAD/AAAAqNQhQLbUIUDL1CFA9dQhQALVIUAQiv4/cBz/P////wASOiZAAAAAAAAAAADQFP8/jwAAAAAAAADAqAQB////AMCoBAGsEAHe////AKwQAQFKP9qN1ahIP9qN1agAABkA
8BL/P4/8vt2EHgAAVN8qQAAAAAD8AQAA+wEAAAEAAAAAEAgAANAfAAYAAAA8vSFAAQAAADwo/z/cHP8/AAAAAAAAAABQiv4/NKj+P/w3JEAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAABkIitAAAAAAAh9JkAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
CAAAADQd/z8AAAAAxCD/PwMAAAB0Tv8/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAABAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
[... cut ...]
0P7/P+AJJ0DUZv8/eDklQNRm/z+kZP8/1Gb/P1xn/z9Mj/8/AAAgACoAAAC4MyVAKEPhYaRk/z/UZv8/XGf/P9Rm/z+kZP8/1Gb/P91LJUAqAAAAAAAgACD//z8AAAAAEP//PwEAAABQG/8/XGf/P9Rm/z8AAAAA
AAAAAEhxJUCAAAAAAGf/P9w9/z9QJCVAaM8uylB42EEA8P8/XGf/P9Rm/z9QeNhBaM8uyv13JUAEXP8/aCP/PwAAAAC4MyVA8Nr/PwAAAAAsI/8/AAAAAPiuGAABAAAAUBv/P9iCJUCYeiZAhCn/PzgR/z/UEv8/
8Nr/PwAAAAAAAAAAEdYlQI5/I0AAAAAAcGAmQJRgJkCoiv4/AAAAAHBgJkDmbCZASQ8AQEkPAECw2v8/SQ8AQBkOAEB4GABABAAAAAAAAAAAEQBABAAAAAAAAAAAEQBARwoQQAAAAAAAAAAAABEAQIUCEEAEAAAA
eBgAQNn8EEBMBABAAAAAAAAAAAAB0s/X", "crc32": 588011221}}
---- END CORE DUMP ----

rojer, 2022-01-17:
@Xaruqui thank you! quick look shows that the device ran out of memory, but i will take a closer look to see why.

rojer, 2022-01-17:
the memory is chock full of network traffic from 172.16.1.16 (i think, though i didn't quite figure out where the headers start).
it appears to be low level data, before TCP or UDP processing and appears to be orphaned, i.e. not referenced from anywhere.
i think it is leaked by the Espressif SDK and i found one way where it can be leaked: receive queue can only hold 10 of these packets and if there are at any point more, they cannot be added and are just dropped on the floor, not properly cleaned up.
at least that's how i read the assembly code (it's in the closed source part).
i will reach out to Espressif about this, see what they say.

Xaruqui, 2022-01-18:
Oups! So it seems like this'll take some serious digging then.

From my side, I think that 172.16.1.16 was my computer's IP (can't confirm since it's not set up as static and I rebooted). What I can confirm is that I had the shelly page opened in my browser for a couple of days (idle, but opened). Could it also be that there were too many comms back and forth while not using the web interface? Dunno, just trying to put some light in the dark.

Thanks so far for everything you are doing for this project. I'm Paypaling again, mate.

rojer, 2022-01-20:
@Xaruqui Espressif got back to me and sent libraries with a fix. can you please put [this version](http://rojer.me/files/shelly/2.11.0-alpha8-fix/) on your shelly 2.5 that has this problem and let me know if it's any better? if it crashes, please send core dump to rojer@rojer.me

Xaruqui, 2022-01-20:
@rojer Got it, boss!

Xaruqui, 2022-01-22:
@rojer It happened again today with 2.11.0-alpha8-fix but no core dump this time. I had to calibrate and it's up and running again. I guess the only thing left to check is the RC snubber...

rojer, 2022-01-22:
hm, ok. yes, reboots that don't leave a core dump are most likely due to power disturbance.

rojer, 2022-01-24:
@Xaruqui can you check if you got new core dumps? i know it didn't fix your original problem but i need to get back to espressif on the packet leak fix.

Xaruqui, 2022-01-24:
Hi @rojer. All good, no core dumps so far in any of the two anarchy-loving devices.

Thanks.

rojer, 2022-01-24:
very good, i passed it on to espressif.

w-marco, 2022-02-05:
I can confirm that in 2.11 Alpha 9 this issue seems to still exist. One of my Shellies just had the Bug again where it said pos0 but was actually opened 100%.

tdexwjgb, 2022-02-06:
I have a similar problem. Out of eight Shelly 2.5's (all on FW 2.10.0) used for roller shutters in my appartment, one has started to become occasionally totally unresponsive (both in HomeKit and via Web UI) about a week ago. I always must power-cycle it by the fuse to make it alive again.

Yesterday, I checked the uptime counter and this is the only one which reboots about every 30 minutes. Just for curiosity, I updated it to 2.10.01 and after that, it got locked in a reboot loop, where I only could access the WebUI for ~50 seconds, before it reboots again.

I removed it from the roller shutter and wired it to mains just for testing purposes. I somehow managed to downgrade to 2.10.0 in an alive phase. So far, it is up and running for more than five minutes. Will update on how it bevahes now.

Any recommendations?

Xaruqui, 2022-03-09:
Hi all, just to say that until last week I couldn't install my RC Snubbers so It's still too early to say if they worked. I handmade them (R=100 ohms;C=0'1uF) but I had to wait until I got the material at home. It's never passed a month without an issue in at least one of my two faulty blinds so I guess that in a month I'd be able to say if they are working fine or not.

Fingers crossed.

Xaruqui, 2022-04-09:
Hi all. After a month with my Snubbers installed I can now say that they worked. Everything is working fine now and no re-calibrations are needed anymore.

closed 2022-05-19 by github-actions[bot]

gmuth, 2024-10-07:
> I confirm

Does that imply a different behavior or sensitivity concerning power usage and potential inductive spikes?

MagOO33, 2024-10-07:
> > I confirm
> 
> Does that imply a different behavior or sensitivity concerning power usage and potential inductive spikes?

Yes, I assume that more consumption make more margin on spikes. Solution for me was to add a capacitor (shelly is selling those capacitors).