mongoose-os-apps/shelly-homekit #846: Issues Flashing Shelly 2 Dimmer

neekul, 2021-11-17:
I have looked at the wiki and tried both the webURL flash and the script via terminal on my mac.

When using the URL method, I get the following:

`{"status":"updating","has_update":false,"new_version":"20211109-131033/v1.11.7-g682a0db","old_version":"20211109-131033/v1.11.7-g682a0db"}`

When loading the Shelly device URL. Nothing has changed. Still has Shelly FW and works as expected

When using the Script method I get the following:

```
Host: http://shellydimmer2-C45BBE56DE28.local
Device Name: Kitchen Light
Model: ShellyDimmer2
Firmware: Stock 1.11.7 ➜ HomeKit Not supported
```

Devices found: 1 Upgradeable: 0 Flashed: 0

My firmware as found on Shelly `Current version: 20211109-131033/v1.11.7-g682a0db`

I tried downloading the ZIP and ran a local file update and got the below:

```
Looking for Shelly devices...
Starting local webserver on port 8381

Host: http://shellydimmer2-C45BBE56DE28.local
Device Name: Kitchen Light
Model: ShellyDimmer2
Firmware: Stock 1.11.7 ➜ Stock Invalid file
Version 1.11.7 is not available...

Devices found: 1 Upgradeable: 0 Flashed: 0
```

So pulling my hair out trying to figure out what's up.
Could it be I have a firmware version (stock) that can't be flashed?

Any help is super appreciated!

timoschilling, 2021-11-17:
> `Firmware: Stock 1.11.7 ➜ HomeKit Not supported`

We do not support the Shelly Dimmer 2, we only support the Shelly 2 (a ancestor of the 2.5).

closed 2021-11-17 by timoschilling

neekul, 2021-11-17:
@timoschilling Ah then that makes sense! Doh! Thanks for the quick response and clarification! 👍  :)