mongoose-os-apps/shelly-homekit #847: Firmware 2.10.1 - loses connection/not responsive after trigger

mrkwagga, 2021-11-18:
Hey Guys,

So recently installed another Shelly 1 for my collection, I set it up as a momentary switch, with auto off.

For some reason 3 out of 4 triggers via Homekit or the button, the devices stops working, and I have to power cycle in order for it to work again.

For reference, no other Shelly device is doing this, but that said... This is the only device running the latest firmware.
[image]

WiFi signal is fine.

What could be going on?

Current Build: 20211016-122946/2.10.1-gc8923b3-master

timoschilling, 2021-11-18:
If you are using a fan on a Shelly you maybe need one of this https://shop.shelly.cloud/rc-snubber-wifi-smart-home-automation

Please revert to stock firmware and test again.

bmpalmeida, 2021-12-04:
Hello, I’m having similar problem. On a brand new 1PM. After I switch the relay off, It always crashes and I have to power cycle to get up again. I tested with stock firmware and it works correctly. Maybe coincidence or not, It is controlling a fan, In this case a bathroom exhaust fan. All my other Shellie are running the last version as well and never had any issue like this.

rojer, 2021-12-04:
1PM, inductive load (fan)... this sounds awfully like the issue where you need to add a capacitor.
in fact, i had to do it myself as well.
i have no idea why stock is fine in this case, maybe they do fancy zero crossing switch or something... not us though, so you need a capacitor in parallel with the fan. they even sell them, iirc.

bmpalmeida, 2021-12-05:
Really interesting. I will put the capacitor. It’s also curious I have another fan, different brand, and and was lucky this didn’t happen. Thanks for the support one more time

rojer, 2021-12-05:
it's a confirmed defect in 1PM's internal layout, current spikes on inductive loads cause interference and processor resets.
really interesting why it doesn't affect stock, maybe zero crossing workaround, i don't know...
but i personally had to add a capacitor to my 1PM that controls a fan.

> I have another fan, different brand, and and was lucky this didn’t happen
all depends on the motor, i guess.

closed 2021-12-17 by rojer

mpartoglou, 2021-12-26:
Sorry to post on a closed thread, I have the exact same issue (have reached out to the aussie retailed where i purchased my Shelly devices to ask if they have something similiar to the RC Snubber linked above).

I wanted to confirm if this issue is present in both the Shelly 1 and the Shelly 1PM (I've experienced an identical issue to the one posted in this thread on both type when triggering an exhaust fan).