mongoose-os-apps/shelly-homekit #848: No response

Undrandi, 2021-11-18:
Hi
I have a Shelly 1 and two Shelly 2.5 that have been running without any problems for a few months. I have been using a ASUS AC68 router with two Lyra minis as a mesh in order to get a good wifi throughout the home. Recently I switched to a setup with a TP link X20 mesh system because my old ASUS started to fail on me.  So after that the network has become even better then before - except for my Shellys.
They now become unresponsive all the time and are useless to me.  I did a firmware update and the problem persists.  
I have read through the other issues here and I don't really think this is a problem with the Shellys but I don't know how to proceed so I thought I might try to ask here.  I apologize if this is not in order.  
To get a good connection in the whole house I run the main router X20 through a switch to the other X20 routers.  It's just an unmanaged TP link switch so that could be the problem but I don't really know.  So now I don't know if I should buy a managed switch (and learn how to set one up) or just throw out the system and get another ASUS.  I have a log if that would be any help.

timoschilling, 2021-11-18:
Yes here isn’t the right place. Maybe try in a Facebook group or in a Shelly forum.

closed 2021-11-18 by timoschilling

Undrandi, 2021-11-18:
Actually I bypassed the switch and I get the same problems.  So the switch is not the problem.  Since everything else on my network is functioning flawlessly the problem, most likely, is the Shelly.