mongoose-os-apps/shelly-homekit #849: Use Shelly1 (or any other HK compatible hardware) as a doorbell input

morettoni, 2021-11-19:
Hello it is doable to add to the firmware an "option" to configure Shelly1 (or any other device) as a doorbell input? I mean: I'd like to configure one Shelly to detect when someone rings at my external doorbell and trigger some actions, like:

- ring a chime;
- turn on a light;
- and of course receive a notification.

Some times ago I saw something similar in one implementation, but then looks like it is disappeared (correct me if I am wrong).
In any case any comment and/or any other solution is always welcome!

timoschilling, 2021-11-19:
Please read https://github.com/mongoose-os-apps/shelly-homekit/wiki/Input-Switch-Settings

closed 2021-11-19 by timoschilling

morettoni, 2021-12-19:
Hello, sorry if I am going to open again this issue, but I changed the input type to “doorbell” but into HomeKit is reported as “unsupported”. Maybe is a compatibility issue with iOS 15.2?

In attach the Shelly1 settings and Home app screenshot.

Thanks for the help!

[image]
[image]

timoschilling, 2021-12-19:
That's not a bug, that's expected. You need to differentiate between HomeKit and the Home App. HomeKit supports that doorbell, but the Home App didn't provide a UI for it. If you want to make a automatization based on the doorbell you need to use the Eve or Home+ app or similar app. The Home App has removed the support for the doorbell years ago.

morettoni, 2021-12-19:
Thanks for the great clarification!

GloWi001, 2022-01-03:
Would it be possible to improve the doorbell function?

Currently, the switch must be detached to use the input as a doorbell. The switch can now only be activated via HomeKit.

But if you now want to use a chime on the switch, you always have to configure it via HomeKit. This is error-prone and has a high latency.

It would be better to have a "doorbell mode" directly. In this case, the input gives a trigger to HomeKit and the switch is automatically triggered.

- Daniel

timoschilling, 2022-01-04:
@GloWi001 you holding it wrong 🤪
Place the Shelly parallel to the chime and not in series. 

That what you want is already requested in a other issue