mongoose-os-apps/shelly-homekit #850: 2.11.0-alpha1 on Shelly 1 Plus leaves default wifi open

rubentroncon, 2021-11-19:
The 2.11.0-alpha1 firmware on the Shelly 1 Plus leaves the default, unsecured, wifi open. This means that anyone could, for example, open my garage door by just connecting to the passwordless wifi.

timoschilling, 2021-11-19:
@rojer ?

rojer, 2021-11-19:
looks like we need a "Keep AP enabled" setting or something.
i guess what i don't understand is what changed on the stock side - does it now keep AP enabled after connecting to wifi?
can you give me exact steps to reproduce it?

rubentroncon, 2021-11-20:
It isn’t necessarily really an issue in your firmware, it seems Shelly doesn’t disable the Wifi AP function automatically. But in that case a toggle to disable manually, or better yet(from my point of view) an auto detect feature to disable the insecure AP if it is still enabled in the Shelly-homekit firmware.

reproduced with stock firmware:
1. Connect Shelly to power
2. Connect through web interface to 192.168.33.1
3. Go to Networks page and set ssid and password of private network
4. Save
At this point the shelly connects to the network and confirms it was successful and returns an IP.
Wifi AP option remains enabled.

rojer, 2021-11-20:
aha, so Plus leaves AP running after wifi configuration? seems like a bug for sure.
but yes, i agree, no matter what stock does, we currently do not support ap+sta mode and should disable AP if station is configured and connects successfully.

rojer, 2021-11-20:
i was about to submit a quick fix but thought i'd take this opportunity to improve wifi setup experience overall.
so it'll take a bit longer but will be more than just a quick fix for this issue.

nliviu, 2021-11-21:
After OTA updating the shelly to homekit, AP mode remains enabled:
```
mos --port ws://192.168.1.233/rpc config-get wifi.ap
{
  "bandwidth_20mhz": false,
  "channel": 6,
  "dhcp_end": "192.168.33.100",
  "dhcp_start": "192.168.33.2",
  "disable_after": 0,
  "enable": true,
  "gw": "",
  "hidden": false,
  "hostname": "",
  "ip": "192.168.33.1",
  "keep_enabled": false,
  "max_connections": 10,
  "netmask": "255.255.255.0",
  "pass": "",
  "protocol": "BGN",
  "ssid": "ShellyPlus1-A8032ABE9E2C",
  "trigger_on_gpio": -1
}
```
but after saving the wifi settings the device reboots and the AP enable turns to false
```
mos --port ws://192.168.1.233/rpc config-get wifi.ap
{
  "bandwidth_20mhz": false,
  "channel": 6,
  "dhcp_end": "192.168.33.100",
  "dhcp_start": "192.168.33.2",
  "disable_after": 0,
  "enable": false,
  "gw": "",
  "hidden": false,
  "hostname": "",
  "ip": "192.168.33.1",
  "keep_enabled": false,
  "max_connections": 10,
  "netmask": "255.255.255.0",
  "pass": "",
  "protocol": "BGN",
  "ssid": "ShellyPlus1-A8032ABE9E2C",
  "trigger_on_gpio": -1
}
```
After setting the wifi credentials, one can turn off the AP mode in the stock firmware.

rubentroncon, 2021-11-21:
@nliviu is right! Just tested this and when setting the SSID and password through the Shelly-homekit web interface and saving, the AP mode is disabled.

The AP mode in the stock firmware can indeed be disabled manually. Non-plus devices (or at least the Plug S and the Shelly 2.5, those are the only ones I own) disable the AP mode *automatically* after setting the wifi creds, and this is in my opinion the desired action. I created a ticket on the Shelly support site for this.

raaaf, 2021-11-24:
@rubentroncon how can I do that manually? 4 of my Shellys still are active and I'd like to hide 'em ;)

[image]

rojer, 2021-11-24:
for now, please use the following workaround to disable the AP:

`http://A.B.C.D/rpc/Config.Set?config=%7B%22wifi%22%3A%7B%22ap%22%3A%7B%22enable%22%3Afalse%7D%7D%7D&reboot=true`

raaaf, 2021-11-24:
@rojer Wonderful – thanks a lot! 🤘

rojer, 2021-12-17:
fixed in 2.11.0-alpha2, #868

closed 2021-12-17 by rojer