mongoose-os-apps/shelly-homekit #851: HTTPS Support for web interface

rubentroncon, 2021-11-21:
Is it possible to add HTTPS support to the web interface? 
As a first step, using a self-signed certificate would suffice just to prevent the connection to the web interface to be unencrypted. 
A very nice to have would be, in a second stage, to be able to use a personal CA.

timoschilling, 2021-11-21:
@rojer is it right that the esp8266 Shelly’s didn’t have the have the resources to do it?

rojer, 2021-11-21:
correct, we simply cannot afford to have TLS it on the 8266 due to flash size constraints.
in principle, it would be easy to do (just a config change) and it would be possible on the plus devices.
but it's such a niche feature that i don't think it's worth the effort implementing the UI.
i'll leave [this link](https://github.com/mongoose-os-libs/http-server/blob/5b4c77fb71e151625470b09fe52a1b0dbddadc77/mos.yml#L16-L17) here for those wishing to enable it themselves (again - won't work on the gen 1 devices).

closed 2021-11-21 by rojer