mongoose-os-apps/shelly-homekit #855: Wi-Fi password bulk update?

hughfr4nc15, 2021-12-01:
Hello,

First of all thank you for your firmware, it’s priceless.

I have a suggestion/feature request… I’m thinking on changing my Wi-Fi password but have so many devices to update… would it be possible in the future to add an option to update all the Shelly’s Wi-Fi password in every Shelly detected by the script? Similar to what we do with firmwares.

Thanks!

timoschilling, 2021-12-02:
If I remind it right thats already possible. @andyblac ?

andyblac, 2021-12-02:
No this is not possible with my script, you can change between DHCP / STATIC, but yes this should be possible, I would have to find the endpoint where the password is saved. I'll look into it see what can be done.

hughfr4nc15, 2021-12-02:
Thanks a lot for taking the time to look into this.

rojer, 2021-12-02:
heads up: major changes expected for 2.11 in this aream, my wifi config manager change is almost ready for testing, just needs some finishing touches.
it will be possible to have two wifi sta configs and configure AP, wifi setup flow UX will also be improved.

andyblac, 2021-12-02:
cheers @rojer I'll wait a little then. thx

hughfr4nc15, 2022-01-03:
Hello and Happy New Year!

Not pressuring or anything but any idea when the new version will be released?

rojer, 2022-01-03:
2.11 is being tested here: #868 
i think we'll do a couple more test builds.

hughfr4nc15, 2022-01-26:
@andyblac with rojer's new alpha9, would you be able to check the possibility of updating all the Shelly’s Wi-Fi password in every Shelly detected by the script?

Thanks.

andyblac, 2022-01-26:
@hughfr4nc15 I'll take look, just need to find some time.

hughfr4nc15, 2022-01-26:
Hello @andyblac, no rush at all, thank you for your time.

hughfr4nc15, 2022-03-12:
Comment to avoid being closed.

closed 2022-06-24 by github-actions[bot]

hughfr4nc15, 2022-08-22:
Hello @andyblac and @rojer. Not sure if you guys have the bandwidth to make this possible... :) Just wanted to keep it open.