mongoose-os-apps/shelly-homekit #857: Problems with RGBW light strip colors

Kai120517, 2021-12-05:
Hi, I have the Shelly RGBW2 (flashed with HomeKit) connected to a high quality RGBW light strip that I bought, but the problem is that when I try to set a color that is made from only the rgb section, it also turns on the white led. That creates a whole different color that I didn’t want.
Is there a way to fix it? I need it to not send a signal to turn white on.
Thanks in advance

timoschilling, 2021-12-05:
Can you please a picture from the strip, the wiring on the Shelly side and of the web UI please

Kai120517, 2021-12-07:
[image]
[image]
[image]

timoschilling, 2021-12-07:
I tested it with my and it is expected behaviour.

Let me explain: Hue 151, means 100% blue and 52% green, which is mathematically a light green. But the Problem is that a blue LED has a higher intensity then a green one, that's results in a more blue colour.

From my point of view the problem is that your LED strip isn't good (must strips have that problem).

> it also turns on the white led

The my logs are showing that only blue and green are on. So if you directly look on your strip and you really see the white LED on, there is a issue with your Shelly, your strip or the wiring.

closed 2021-12-07 by timoschilling