mongoose-os-apps/shelly-homekit #858: Authentication issue with Homekit Firmware

skynan22, 2021-12-06:
After updating Shellies with current Homekit Firmware 2.10.1 a remote initiatd reboot does not work anymore. Although there is an entry for every Shelly in my flash-shelly.auth.yaml for both name and ip with user & password values included.

shelly1-name.local:
  user: XXX
  password: YYY
shelly1-ip:
  user: XXX
  password: YYY

This works for all Shelly1s with stock firmware (1.11.7) but does not work for the ones with homekit fw (2.10.1):
- **python flash-shelly.py --reboot -y -a**

Error message:
Host: shelly1-name.local is password protected
Invalid user or password found in 'flash-shelly.auth.yaml',please check supplied details are correct.
username: XXX
password: YYY

andyblac, 2021-12-06:
can you try with the latest beta of version 3.0.0 of the script here https://github.com/mongoose-os-apps/shelly-homekit/issues/686

skynan22, 2021-12-06:
> can you try with the latest beta of version 3.0.0 of the script here #686

With beta 6 version of the script, the issue being solved. When do you plan release stable version?

andyblac, 2021-12-06:
Can you provide a log of the devices that don't won't I was not aware that stock security is not working with version 3 of script, it is just wait for roger to give his approval on it.

andyblac, 2021-12-06:
seems to be working here on stock

```
andrewblackburn@iMac tools % ./flash-shelly.py -l shellydimmer-f3cd89.local
Looking for Shelly devices...

Host: shellydimmer-f3cd89.local is password protected
shellydimmer-f3cd89.local is not found in 'flash-shelly.auth.yaml' config file,
please add or use commandline args --user | --password

andrewblackburn@iMac tools % ./flash-shelly.py -l shellydimmer-f3cd89.local --user admin --password zxcvb
Looking for Shelly devices...

Host: http://shellydimmer-f3cd89.local
Device Name: Office Dimmer
Model: ShellyDimmer
Device ID: shellydimmer-F3CD89
Mode: White
SSID: ABFi-IoT-2.4
IP: 192.168.3.53
RSSI: -58
Up Time: 6 days, 23:55:27
Firmware: Stock 1.11.7 ✔

Devices found: 1 Upgradeable: 0
```

and now the user and password has automatically been added to the yaml file, it is automatically used.
```
andrewblackburn@iMac tools % ./flash-shelly.py -l shellydimmer-f3cd89.local
Looking for Shelly devices...

Host: http://shellydimmer-f3cd89.local
Device Name: Office Dimmer
Model: ShellyDimmer
Device ID: shellydimmer-F3CD89
Mode: White
SSID: ABFi-IoT-2.4
IP: 192.168.3.53
RSSI: -57
Up Time: 6 days, 23:55:36
Firmware: Stock 1.11.7 ✔

Devices found: 1 Upgradeable: 0
andrewblackburn@iMac tools %
```

skynan22, 2021-12-06:
> seems to be working here on stock
> 
> ```
> andrewblackburn@iMac tools % ./flash-shelly.py -l shellydimmer-f3cd89.local
> Looking for Shelly devices...
> 
> Host: shellydimmer-f3cd89.local is password protected
> shellydimmer-f3cd89.local is not found in 'flash-shelly.auth.yaml' config file,
> please add or use commandline args --user | --password
> 
> andrewblackburn@iMac tools % ./flash-shelly.py -l shellydimmer-f3cd89.local --user admin --password zxcvb
> Looking for Shelly devices...
> 
> Host: http://shellydimmer-f3cd89.local
> Device Name: Office Dimmer
> Model: ShellyDimmer
> Device ID: shellydimmer-F3CD89
> Mode: White
> SSID: ABFi-IoT-2.4
> IP: 192.168.3.53
> RSSI: -58
> Up Time: 6 days, 23:55:27
> Firmware: Stock 1.11.7 ✔
> 
> Devices found: 1 Upgradeable: 0
> ```
> 
> and now the user and password has automatically been added to the yaml file, it is automatically used.
> 
> ```
> andrewblackburn@iMac tools % ./flash-shelly.py -l shellydimmer-f3cd89.local
> Looking for Shelly devices...
> 
> Host: http://shellydimmer-f3cd89.local
> Device Name: Office Dimmer
> Model: ShellyDimmer
> Device ID: shellydimmer-F3CD89
> Mode: White
> SSID: ABFi-IoT-2.4
> IP: 192.168.3.53
> RSSI: -57
> Up Time: 6 days, 23:55:36
> Firmware: Stock 1.11.7 ✔
> 
> Devices found: 1 Upgradeable: 0
> andrewblackburn@iMac tools %
> ```
> 
> my yaml test file
> 
> ```
> 192.168.3.68:
>   password: '12345'
>   user: admin
> shelly1-FD89C2.local:
>   password: '12345'
>   user: admin
> shellydimmer-f3cd89.local:
>   password: zxcvb
>   user: admin
> ```

Here is the output from bash, when executing reboot cmd with beta-6 python script:

**python flash-shelly-beta.py --reboot -y -a**
Scanning for Shelly devices...

Host: shellyplug-s-A32A81.local is password protected
Invalid user or password found in 'flash-shelly.auth.yaml',please check supplied details are correct.
username: admin
password: ***

Host: http://Licht-Arbeitszimmer.local
Device Name: Licht-Arbeitszimmer
Model: Shelly1L
Firmware: HomeKit 2.10.1 ✔
Rebooting...
waiting for Licht-Arbeitszimmer to reboot
Device has rebooted...

Host: shellyplug-s-80C36A.local is password protected
Invalid user or password found in 'flash-shelly.auth.yaml',please check supplied details are correct.
username: admin
password: ***

Host: http://Rollo-WiGa-Rechts.local
Device Name: Rollo-WiGa-Rechts
Model: Shelly25
Firmware: HomeKit 2.10.1 ✔
Rebooting...
waiting for Rollo-WiGa-Rechts to reboot
Device has rebooted...

Host: http://Licht-Kueche.local
Device Name: Licht-Kueche
Model: Shelly1
Firmware: HomeKit 2.10.1 ✔
Rebooting...
waiting for Licht-Kueche to reboot
Device has rebooted...

Host: http://Licht-Eingang.local
Device Name: Licht-Eingang
Model: Shelly1
Firmware: HomeKit 2.10.1 ✔
Rebooting...
waiting for Licht-Eingang to reboot
Device has rebooted...

Host: http://Rollo-WiGa-Links.local
Device Name: Rollo-WiGa-Links
Model: Shelly25
Firmware: HomeKit 2.10.1 ✔
Rebooting...
waiting for Rollo-WiGa-Links to reboot
Device has rebooted...

Host: http://Rollo-Terasse.local
Device Name: Rollo-Terasse
Model: Shelly25
Firmware: HomeKit 2.10.1 ✔
Rebooting...
waiting for Rollo-Terasse to reboot
Device has rebooted...

Host: shellyplug-s-6E6F96.local is password protected
Invalid user or password found in 'flash-shelly.auth.yaml',please check supplied details are correct.
username: admin
password: ***

Host: http://Licht-Keller.local
Device Name: Licht-Keller
Model: Shelly1
Firmware: HomeKit 2.10.1 ✔
Rebooting...
waiting for Licht-Keller to reboot
Device has rebooted...

Host: shellyplug-s-1D3DA6.local is password protected
Invalid user or password found in 'flash-shelly.auth.yaml',please check supplied details are correct.
username: admin
password: ***

Host: http://Rollo-Esszimmer.local
Device Name: Rollo-Esszimmer
Model: Shelly25
Firmware: HomeKit 2.10.1 ✔
Rebooting...
waiting for Rollo-Esszimmer to reboot
Host: Rollo-Esszimmer.local is password protected
Invalid user or password found in 'flash-shelly.auth.yaml',please check supplied details are correct.
username: admin
password: ***

Device has rebooted...

Host: shellyplug-s-DDAD4A.local is password protected
Invalid user or password found in 'flash-shelly.auth.yaml',please check supplied details are correct.
username: admin
password: ***

Host: http://Rollo-WiGa-Mitte.local
Device Name: Rollo-WiGa-Mitte
Model: Shelly25
Firmware: HomeKit 2.10.1 ✔
Rebooting...
waiting for Rollo-WiGa-Mitte to reboot
Device has rebooted...

Host: http://Licht-Garage.local
Device Name: Licht-Garage
Model: Shelly1
Firmware: HomeKit 2.10.1 ✔
Rebooting...
waiting for Licht-Garage to reboot
Device has rebooted...

Devices found: 10 Upgradeable: 0

skynan22, 2021-12-06:
For every Shelly username & password is the same and I have all combinations (device IP & device name) within the auth file. Shelly Plug S seems to have a problem. Is it possible that the error occurs because of another order of user/password in my auth file? You have 

Device name:
  password: XXX
  user: YYY

In my file it is
Device name:
  user: YYY
  password: XXX

andyblac, 2021-12-06:
the order should not matter, as a test can you remove the data from the yaml file and do manually in command line ?
like

`./flash-shelly.py -l shellydimmer-f3cd89.local --user admin --password zxcvb` also please bare in mine the shelly homelike firmware only supports the user name 'admin' unlike the stock firmware that you can change it to anything you want.

skynan22, 2021-12-07:
This is strange, I moved current auth file to _auth file:
`mv flash-shelly.auth.yaml _flash-shelly.auth.yaml `
After that I used the following command from you with the credentials for that certain Shelly:
`./flash-shelly-beta.py -l shellyplug-s-DDAD4A.local --user admin --password ***`

This results in the following error message
Looking for Shelly devices...
Could not connect to host: shellyplug-s-DDAD4A.local

But the device is reachable from Browser, when opening the page
http://shellyplug-s-ddad4a.local/
a basic auth form appears and after entering the above used credentials, the login is being accepted.

andyblac, 2021-12-11:
as for authentication HK firmware only supports user 'admin' atm, so if you setup authentication in the stock fw, try disable it and re-enable it in HK fw.

so you need to use --user admin --password xxxx in command line.

andyblac, 2021-12-11:
seems to be working fine here with Version 3.0 of script

```
andrewblackburn@iMac tools % ./flash-shelly.py -l 192.168.3.54
Looking for Shelly devices...

Host: 192.168.3.54 is password protected
192.168.3.54 is not found in 'flash-shelly.auth.yaml' config file,
please add or use commandline args --user | --password

andrewblackburn@iMac tools % ./flash-shelly.py -l 192.168.3.54 --user admin --password zxcvb
Looking for Shelly devices...

Host: http://192.168.3.54
Device Name: Office-Window-Strip
Model: ShellyRGBW2
Device ID: shellyrgbw2-AFAF3E
Mode: RGBW
SSID: ABFi-IoT-2.4
IP: 192.168.3.54
RSSI: -57
Up Time: 3 days, 2:45:17
HAP Connections: 0 / 1 / 12
Firmware: HomeKit 2.10.1 ✔

Devices found: 1 Upgradeable: 0
```

https://github.com/mongoose-os-apps/shelly-homekit/issues/686

rojer, 2021-12-17:
seems liek a script issue - if it's an issue at all.

closed 2022-01-24 by github-actions[bot]