mongoose-os-apps/shelly-homekit #859: RGBW2 Brightness issue

Oli1397, 2021-12-06:
following setup:
- Shelly RGBW2 with latest HomeKit release (2.10.1)
- 1m RGBW Lightstrip
- Digital meter measures 0.75A at full brigtness on white color

alternative setup with same device
- Shelly RGBW2 with latest official Firmware
- 1m RGBW Lightstrip
- Digital meter measures 0.89A at full brightness on white color

there is a noticeable difference in the brightness.

timoschilling, 2021-12-06:
Please provide some information about how you measured. And please provide some log output from the HomeKit software when it’s on 100%

Oli1397, 2021-12-08:
First of all here is a drawing of my setup:
[image]

**Szenario 1:**
Mode **RGBW**
Color **#FFFFFF**
Hue **360%**
Saturation **0%**
Brightness **100%**
Homekit value **100%**
Power **0.24A**

**Szenario 2:**
Mode **RGB**
Color **#FFFFFF**
Hue **360%**
Saturation **0%**
Brightness **100%**
Homekit value **100%**
Power **0.68A**

In the attached log file you can see switching from mode RGB to RGBW to RGB.
My first interpretation ist that RGB uses channel 1-3(RGB). RGBW seems to use only channel 4 and turns automatically channel 1-3 off. Not sure if that's the point.
[ShellyRGBW2 Log.txt](https://github.com/mongoose-os-apps/shelly-homekit/files/7675838/ShellyRGBW2.Log.txt)

timoschilling, 2021-12-08:
> My first interpretation ist that RGB uses channel 1-3(RGB). RGBW seems to use only channel 4 and turns automatically channel 1-3 off. Not sure if that's the point.

Yes that's right. Maybe the stock firmware is doing it different. But from my point of view that is wrong.

Oli1397, 2021-12-08:
Yeah could be pov thing. Adding Shelly via Homeassistant or homebridge adds 2 HomeKit devices a RGB Bulb and a Bulb for the W channel. So you can use these strip types as ambient light and daylight or to improve brightness you can activate both.

So the RGBW mode in this HomeKit firmware ist simply a W mode without the RGB channels.
Well understand your point of view, could be difficult to mix the colors right when adding additional white.

But is my use case possible to realize?

timoschilling, 2021-12-08:
> So the RGBW mode in this HomeKit firmware ist simply a W mode without the RGB channels.

That's not the case. Saturation 0 means W is off and Saturation 100 means W is 100% on.

There will come a change to add a RGB+W Mode, which means RGB as it currently is plus the W channel as a 2 bulb. So it will be as what you know from Homeassistant or homebridge.

Can we close this issue or do you have additional comments?

Oli1397, 2021-12-08:
Ah ok, understand that.
Sounds great. Thanks for your help. We can close this one.

closed 2021-12-08 by Oli1397