mongoose-os-apps/shelly-homekit #860: Shelly RGBW2 WiFi issue

lachezartodorov, 2021-12-09:
Hello,
I have issue with the Shelly RGBW2 firmware.
When I upload the mongoose firmware the device disconnects from the wifi and returns to AP mode.
In the WiFi settings I'm selecting Enable and Save, but after couple of secconds the AP is again on and the device never connects to the wifi. But the settings are there and they are correct.
Other strange behaviour is in the "RGB(Light bulb)" section. The "Status" toggle constantly goes from onn to off. And the actual light is flickering a little bit. Not sure if this is related or not, but probably shouldn't be like this.

lachezartodorov, 2021-12-10:
I was using the AC Switch "RGBW LED strip scheme", but without a switch.
I changed it to DC Switch "RGBW LED strip scheme", again without switch and it works.
The only issue now is the buzzing sound when I switch it off (when there is no load), but I guess this is a hardware issue?

timoschilling, 2021-12-10:
In you case AC didn't make sense, it is only for the case that you have a AC (110-230V) Switch. And we are not supporting AV Switch at the moment. The problem with AC is that the input signal is to noisy, which behaves like multiple switch presses in the row, which results in a reset.

So using DC Switch is the right fix!

closed 2021-12-10 by timoschilling