mongoose-os-apps/shelly-homekit #861: New Shelly Pro family

moranchoj, 2021-12-09:
Is there plans to support new Shelly Pro family?

rojer, 2021-12-11:
no immediate plans for Pro but support for Plus devices is coming soon

scuben, 2021-12-13:
> no immediate plans for Pro but support for Plus devices is coming soon

are we talking days or week(s)? No hurry or pressure! Just want to get a feel if I should wait for the next version or install the firmware available in the other issue you already provided.

frankpeters1001, 2021-12-26:
Hey,

Imam curious on an estimated release date too. Just bought 4 Shelly Plus 1 PM and it took quite a while until I realized the plus version is not supported. I am thinking about returning these unless the release time is some time soon. 

Thanks a lot in advance

rojer, 2021-12-26:
you can help test Plus 1 and 1PM support [here](https://github.com/mongoose-os-apps/shelly-homekit/issues/868).

moranchoj, 2021-12-26:
I will test too with shades. I need a relay with 2 dry contacts like 1 Plus
to test shades/curtain functionality.

Does the new beta firmware for shelly 1 plus support shades/curtain
functionality?

Thanks

rojer, 2021-12-27:
no, not yet

LoicDelorme, 2022-03-28:
@rojer any updates regarding this issue?

timoschilling, 2022-03-29:
@LoicDelorme no, because we don’t have any test devices

LoicDelorme, 2022-03-29:
@timoschilling are you waiting for someone from this community to buy some in order to carry out some integration tests?
Or someone to order devices for you?

Sry for my maybe stupid question but I just discovered Shelly devices and this repository :)

Bigg-G, 2022-04-28:
Team, i have just recieved a bunch of Plus 1's and Plus 1 PM's, happy to test, so long as someone can provide step by steo guide how to re-flash to Shelly stock FW (so i can then flash to Homekit FW when it gets to GA :-)
Let me know team.

timoschilling, 2022-04-28:
My problem is that the Pinot of the pro models isn’t documented somewhere. So I need to device to reverse engineer it.

moranchoj, 2022-06-23:
HI Timo

Shelly Pro 2 has already been relased and it is available for selling.

is there plans to develop a firmware version for Shelly Pro 2 with the curtain/shades option?

Shelly Pro 2 has 2 dry contacts and it is very suitable to be used for curtains motors that have 2 control lines not energized with the current of the motor. The curtain motor provides a common line and 2 control lines to activate up and down movement.
 
Do you need a test device to develop the firmware version?

Thanks

Josep Morancho

timoschilling, 2022-06-26:
> Shelly Pro 2 has 2 dry contacts and it is very suitable to be used for curtains motors that have 2 control lines not energized with the current of the motor. The curtain motor provides a common line and 2 control lines to activate up and down movement.

Just for the record we currently didn't support that operation type. Our window cover mode need's non dry contacts and a wiring with power drin through the shelly (like the 2.5, plus pm and pro pm). The motor you describe currently only works in garage door mode.

moranchoj, 2022-06-26:
Does Garage mode mean only 2 positions: fully open and fully close?

Thanks

va escriure:

timoschilling, 2022-06-26:
Yes, only open and closed. And Garage Door is a security device, which Window Cover isn't. That mean you can open a WC on a not authenticated devices (locked iPhone via "Hey Siri" or on a HomePod), but GD can only be opened on authenticated devices (Watch or unlocked iPhone).

moranchoj, 2022-06-26:
This fundtionality is enough. I only need to open the beedroom blinds in
the morning and close them at night via homekit automation shortcuts.

Motor has 3 controls signal lines: open, close and stop, and 1 com line as
shown in the picture below.

I only need to control open and close signal lines with the 2 dry contacts
of Pro 2 device. Blinds stops automstically when they reach the final point.

Would this work with the garage function?

Thanks

timoschilling, 2022-07-07:
@moranchoj yes that would work.

hitindags, 2022-07-10:
Do you need a Shelly Pro 2 device for testing? As long as it's not compatible, mine is practically useless..

rojer, 2022-07-10:
@hitindags please reach out to me at rojer@rojer.me and we can arrange postage.

timowolf, 2022-11-05:
Dear devs, 
I know this a repeated question, but: I just bough a Shelly Pro 1 and looking for the shelly-homekit support. 

Any news here or can I help somehow?

Many thanks for your awesome work here.

hitindags, 2022-11-05:
I ended App using Home Bridge on my NAS Server. Works well!

AirPett, 2023-02-04:
Any news on Pro support in early 2023?

cpm1, 2023-03-09:
If I can help in any way to get Pro 1PM support, please let me know. I'm a dev myself, so while I don't have time to contribute directly to your great project, I might be able to help out otherwise with alpha tests or logging or measurements, etc. Just contact me if needed/wanted.

cpm1, 2023-04-07:
@timoschilling 
> My problem is that the Pinot of the pro models isn’t documented somewhere. So I need to device to reverse engineer it.

I'd happily donate a Pro 1PM (for example, because that is what I would like to use) to you if lack of hardware is the only blocker :) Just get in touch with me if interested.

nextgenoperations, 2023-11-14:
I too would love to see Shelly Pro support, but I also see that Matter is supposed to be coming to these devices by year end (2023) so I guess they should 'natively' work with Homekit when this happens right ?

mpochin, 2024-08-28:
Not sure if Pro models are supported, they are not in the supported devices list,  but in Shelly HomeKit 2.12.1 #1413 it says "Addon support for all supported Plus Models".

Could you please clarify.

Thank you very much

markirb, 2024-08-28:
> Not sure if Pro models are supported, they are not in the supported devices list, but in Shelly HomeKit 2.12.1 #1413 it says "Addon support for all supported Plus Models".
> 
> Could you please clarify.
> 
> Thank you very much

Do not know what you mean. The readme explicitly says that pro are not supported yet. Plus Models are Plus models and Plus add-on is this: https://www.shelly.com/en-de/products/product-overview/shelly-plus-add-on