mongoose-os-apps/shelly-homekit #862: Activation delay

lucaspinelli85, 2021-12-10:
In some cases I need to be able to delay the activation of the relay from when I send the impulse to SW.
Let me explain: after pressing the button, I would like the relay to activate not immediately, but after a settable time.
In practice it is like having an auto ON similar to the existing auto OFF.
Is it possible to implement this function?
I want to avoid having automations as they are slow and sometimes they don't work and I want it to work regardless of the wifi network.
Specifically, I need it for:
- open the motorized door after opening the lock
- switch on series of lights in sequence a few seconds from each other
- activate the alarm after a time that I have gone out
etc. etc.

rojer, 2022-01-05:
i wouldn't call it "auto on", but i get your point.
not difficult to do but i'll leave it to gather some support. this is the first time i get this request, not immediately obvious that it will be useful to people other than just you.

lucaspinelli85, 2022-01-05:
It would be fantastic if you were able to make the change, even if only on a test firmware so in addition to testing the function I can do the automation I need. Then if someone else will also need it, you can share or formalize the change ...

timoschilling, 2023-01-16:
This regest didn't get any 👍 in one year, so I would tend to close it.

closed 2023-01-16 by timoschilling