mongoose-os-apps/shelly-homekit #863: Always on mode for power metering only

fredvd, 2021-12-11:
I'm using 4 shelly plugs in my home to do power metering only and they are connected to a HomeAssistant instance for readout. I have set up the config of the plugs to be initial state on. 

However in the past 6 months it has happened 2 times on different plugs that the Shelly Plug got turned off. One of the times this happened at the same time my wifi network had issues, the other time might have been a power failure. 

Would it make sense to have an extra setup option to override/disable the switching functionality of the shelly's completely,  have the firmware always activate the switch(es) as soon as possible and ignore any incoming switch commands? 

Because I now cannot trust the plug to monitor for example a freezer ftp monitor power usage if the shelly very sporadically does turn the switch off.

I don't know for 100% that something in HomeAssistant did activate the switch, or that the load on the switch causes this, but it would be helpful if the shelly's themselves could be set to "we'll try to be always on, as soon as possible"

timoschilling, 2021-12-11:
I expect you use the wrong repository, this repository has nothing to do with integrating Shellys into HomeAssistant.

closed 2021-12-11 by timoschilling

fredvd, 2021-12-11:
@timoschilling It has nothing to do with HomeAssistant, I just mentionned it because I'm using Homeassistants support to connect to devices with Homekit Protocol. Any home automation system, be it Apple's hardware/software  or other hardware /software can normally control the Shelly switches and I want to prevent/block control on the Shelly itself.

timoschilling, 2021-12-11:
If you talk about home assistant, I assume that you are not using this firmware, because it didn't work together with it. 

Are you using the firmware from this repo?

fredvd, 2021-12-11:
Yes I am using this .  Homeassistant has two drivers: one to expose its managed switces to Homekit (the bridge) but also a Homekit controller that allows you to connect Homekit accesories to HomeKit.

Bridge -> https://www.home-assistant.io/integrations/homekit/
Homekit controller -> https://www.home-assistant.io/integrations/homekit_controller/

When I bought and tested the Shelly plugs last year I found that the stock firmware with HA shelly integration was messy to configure and unstable, wifi connection fell away more than once. The Homekit controller integration with the mongoose firmware worked much better and stable connection.

fredvd, 2022-03-21:
@timoschilling This issue was never re-opened allthough I think I clarified the requested functionality and that it is independent from HomeKit or HomeAssistant. Is there anything other information you need that I can  provide?

timoschilling, 2022-03-21:
Yes you should file a bug on https://www.home-assistant.io/integrations/homekit_controller/ because your intend is to some a problem in that software.

fredvd, 2022-03-21:
No I don’t have an issue with the homrkit part.  The firmware is sometimes turning the switch off when the wifi connection becomes unavailable or during a restart.  My feature request is an option in the firmware control panel to have the switch always on, no matter what is happening, so the device is only available for power metering.

rojer, 2022-03-21:
> Because I now cannot trust the plug to monitor for example a freezer ftp monitor power usage if the shelly very sporadically does turn the switch off.

this is interesting. the way to know is to leave the log tab open and wait for it to happen.

> I don't know for 100% that something in HomeAssistant did activate the switch, or that the load on the switch causes this, but it would be helpful if the shelly's themselves could be set to "we'll try to be always on, as soon as possible"

set "initial mode" to "Last" or "On", this way if indeed the device restarts, it will come back with the switch on.
if that is not enough, we can think of something else.