mongoose-os-apps/shelly-homekit #864: How to use Flashscript on PW-protected shelly homekit

RaBu62, 2021-12-11:
Hi,

is it possible to use flashscript for automated-commands (update, reboot...) on pw-protected Shellys with Homekit FW? I can't find corresponding arguments in your documentation. Tried out the following script for automated reboot of all my shellys:
python3 flash-shelly.py -la --reboot -y
Got the following responses during script execution:

Host: http://ShellyXY.local
Device ShellyXY
Model: Shelly25
Firmware: HomeKit 2.10.1 ✔
Rebooting...
Host: ShellyXY.local is password protected
Please use either command line security (--user | --password) or 'flash-shelly.auth.yaml'
for 'flash-shelly.auth.yaml', create a file called 'flash-shelly.auth.yaml' in tools folder
Example flash-shelly.auth.yaml:
shelly-AF0183.local:
  password: abc123
  user: admin

Shelly-Homekit only supports password activation without user authentication so i tried
the same script as above including the --password  -Argument -> no success.

Do these arguments only work with original-Shelly-firmware?

RaBu62, 2021-12-11:
Sorry, found same problems reading issue #858. Closing and waiting for solution of #858

closed 2021-12-11 by RaBu62

andyblac, 2021-12-11:
as for authentication HK firmware only supports user 'admin' atm, so if you setup authentication in the stock fw, try disable it and re-enable it in HK fw.

so you need to use `--user admin --password xxxx` in command line.