mongoose-os-apps/shelly-homekit #866: I/O Configuration

goeran03, 2021-12-15:
Wouldn't it make sense to have a more generic version? There are several smart plugs based on ESP8266 with slightly different IO-usage. They should also work if there would be the possibility in user inteface may be to configure where relais and switch is connected and if there a temp sensor or not etc.

rojer, 2021-12-15:
sure, it can be easily repurposed for other targets but greater flexibility means greater complexity.
we target shellies specifically, and thus configure pins statically.
the main benefit of the firmware is that it is possible to go back and forth with the stock shelly fw through OTA update, with no physical reflashing. compatibiility with non-shelly devices is simply not a priority for us.

closed 2021-12-15 by rojer

goeran03, 2021-12-15:
ok thanks