mongoose-os-apps/shelly-homekit #868: Testing 2.11.0

rojer, 2021-12-17:
What's new:
 - Plus1 and Plus1PM support (https://github.com/mongoose-os-apps/shelly-homekit/issues/800)
 - New and improved wifi config manager:
   - Supports multiple (2) station configurations
   - Allows configuring static IP
   - Recovers from incorrect STA settings (reverts back to previous config)
   - Improved UX when configuring: AP is kept running, providing responsive UI.
 
Download here: http://rojer.me/files/shelly/2.11.0-alpha2/
Or use script: `tools/flash-shelly.py -a -V 2.11.0-alpha2`

I'm especially interested in wifi manager testing, please report any issues you see.

bmpalmeida, 2021-12-17:
I don’t know if it is intentional in the alpha builds. But the check for updates and revert to stock are missing.

rojer, 2021-12-17:
no, it's a bug - fixed in alpha3

timoschilling, 2021-12-17:
There is a problem with the 2.5 firmware. It is introduced between alpha1 and alpha2. I tested it with 3 different shelly, each with a different installed version. So I think the problem is the new firmware and not the old one.

Observed things:
1. alpha1 can be installed
2. alpha2 & 3 not
3. alpha2 jumps up in size from 992747 to 1005036
4. flash crashes with `ConnectionResetError: [Errno 54] Connection reset by peer`
5. upload via UI stops at 0%

bmpalmeida, 2021-12-17:
> 

I'm also having the same behaviour from the script, i left the script running all night (while [[ 0 -eq 0 ]]; do python3 flash-shelly.py -m keep -y -i 3 -a --variant alpha; sleep 20; clear; done) and none of the 2.5 were updated. But all the 1PM are completely fine.

andyblac, 2021-12-17:
> I'm also having the same behaviour from the script, i left the script running all night (while [[ 0 -eq 0 ]]; do python3 flash-shelly.py -m keep -y -i 3 -a --variant alpha; sleep 20; clear; done) and none of the 2.5 were updated. But all the 1PM are completely fine.

I would like to see the output of that if poss, as it should have timed out.

add `--log-file debug.log` to your command line, post the log file itself.

Thx

bmpalmeida, 2021-12-17:
> 

Looks like the crash is the script trying to update my 3d printer.

Attachment the debug log and also the console output.

[debug.log](https://github.com/mongoose-os-apps/shelly-homekit/files/7734686/debug.log)

Console output:

admin@iMac tools % python3 flash-shelly.py -m keep -y -i 3 -a --variant alpha --log-file debug.log
Scanning for Shelly devices...

Host: http://shellydimmer-D3E51C.local
Device Name: Extrator-Musica
Model: ShellyDimmer
Device ID: shellydimmer-D3E51C
Mode: White
SSID: bma-home
IP: 192.168.50.249
RSSI: -49
Up Time: 7 days, 14:44:55
Firmware: Stock 1.11.7 ✔

Host: http://Bomba-Casa-Maquinas.local
Device Name: Bomba-Casa-Maquinas
Model: Shelly1PM
Device ID: shelly1pm-D220EB
Mode: Switch
SSID: bma-home
IP: 192.168.50.247
Sys Temp: 30˚c
Up Time: 9:42:56
HAP Connections: 0 / 2 / 12
Firmware: HomeKit 2.11.0-alpha3 ✔

Host: http://Musica-Duplo.local
Device Name: Musica-Duplo
Model: Shelly25
Device ID: shellyswitch25-F2F3F6
Mode: Switch
SSID: bma-home
IP: 192.168.50.45
RSSI: -43
Sys Temp: 58˚c
Up Time: 7 days, 14:44:36
HAP Connections: 0 / 2 / 12
Firmware: HomeKit 2.10.1 ➜ 2.11.0-alpha3
Downloading Firmware...
Now Flashing HomeKit 2.11.0-alpha3

Host: http://Luz-Casa-Maquinas.local
Device Name: Luz-Casa-Maquinas
Model: Shelly1PM
Device ID: shelly1pm-FADBF9
Mode: Switch
SSID: bma-home
IP: 192.168.50.24
Sys Temp: 28˚c
Up Time: 9:43:29
HAP Connections: 0 / 2 / 12
Firmware: HomeKit 2.11.0-alpha3 ✔

Host: http://shellydimmer-D3EB09.local
Device Name: Exterior-Parede-Direita
Model: ShellyDimmer
Device ID: shellydimmer-D3EB09
Mode: White
SSID: bma-home
IP: 192.168.50.157
[... 88 lines cut ...]
  File "/usr/local/lib/python3.9/site-packages/requests/api.py", line 117, in post
    return request('post', url, data=data, json=json, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/requests/api.py", line 61, in request
    return session.request(method=method, url=url, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/requests/sessions.py", line 542, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/local/lib/python3.9/site-packages/requests/sessions.py", line 697, in send
    r.content
  File "/usr/local/lib/python3.9/site-packages/requests/models.py", line 836, in content
    self._content = b''.join(self.iter_content(CONTENT_CHUNK_SIZE)) or b''
  File "/usr/local/lib/python3.9/site-packages/requests/models.py", line 761, in generate
    raise ChunkedEncodingError(e)
requests.exceptions.ChunkedEncodingError: ("Connection broken: ConnectionResetError(54, 'Connection reset by peer')", ConnectionResetError(54, 'Connection reset by peer'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/admin/Downloads/shelly-homekit-master/tools/flash-shelly.py", line 1584, in <module>
    sys.exit(main.run_app())
  File "/Users/admin/Downloads/shelly-homekit-master/tools/flash-shelly.py", line 947, in run_app
    traceback.trace_exception(exc_type, exc_value, exc_traceback, file=sys.stdout)
AttributeError: module 'traceback' has no attribute 'trace_exception'

Devices found: 8 Upgradeable: 2 Flashed: 0
Log file created: debug.log

andyblac, 2021-12-17:
please try version 3.0 beta of the script https://github.com/mongoose-os-apps/shelly-homekit/issues/686

bmpalmeida, 2021-12-17:
Now it crashes trying to update one 2.5 (i believe that this is due to what timoschilling was reporting on the firmware problem with 2.5).

[debug.log](https://github.com/mongoose-os-apps/shelly-homekit/files/7734737/debug.log)

admin@iMac Downloads % python3 flash-shelly.py -m keep -y -i 3 -a --variant alpha --log-file debug.log
Scanning for Shelly devices...

Host: http://quarto-direita.local
Device Name: quarto-direita
Model: Shelly1PM
Device ID: shelly1pm-05F7CB
Mode: Switch
SSID: bma-home
IP: 192.168.50.39
Sys Temp: 39˚c
Up Time: 9:50:44
HAP Connections: 0 / 2 / 12
Firmware: HomeKit 2.11.0-alpha3 ✔

Host: http://Bomba-Casa-Maquinas.local
Device Name: Bomba-Casa-Maquinas
Model: Shelly1PM
Device ID: shelly1pm-D220EB
Mode: Switch
SSID: bma-home
IP: 192.168.50.247
Sys Temp: 30˚c
Up Time: 9:57:36
HAP Connections: 0 / 2 / 12
Firmware: HomeKit 2.11.0-alpha3 ✔

Host: http://Luz-Casa-Maquinas.local
Device Name: Luz-Casa-Maquinas
Model: Shelly1PM
Device ID: shelly1pm-FADBF9
Mode: Switch
SSID: bma-home
IP: 192.168.50.24
Sys Temp: 28˚c
Up Time: 9:58:10
HAP Connections: 0 / 2 / 12
Firmware: HomeKit 2.11.0-alpha3 ✔

Host: http://shellyplug-s-51D7C0.local
Device Name: Frigorifico
Model: ShellyPlugS
Device ID: shellyplug-s-51D7C0
SSID: bma-home
IP: 192.168.50.59
RSSI: -55
Up Time: 22 days, 7:36:24
Firmware: Stock 1.11.7 ✔

Host: http://shellyem-C45BBE5FE4C8.local
Device Name: None
Model: ShellyEM
Device ID: shellyem-C45BBE5FE4C8
SSID: bma-home
IP: 192.168.50.253
RSSI: -76
Up Time: 7 days, 14:59:22
Firmware: Stock 1.11.7 ✔

Host: http://Lavandaria.local
Device Name: Lavandaria
Model: Shelly1PM
Device ID: shelly1pm-FAE0FC
Mode: Switch
SSID: bma-home
IP: 192.168.50.134
[... 67 lines cut ...]
    self.probe_device(device)
  File "/Users/admin/Downloads/flash-shelly.py", line 1634, in probe_device
    self.parse_info(device, hk_flash_fw_version)  # main convert run, or update stock to latest firmware if an update or color mode is required.
  File "/Users/admin/Downloads/flash-shelly.py", line 1527, in parse_info
    self.flash_firmware(device)
  File "/Users/admin/Downloads/flash-shelly.py", line 1291, in flash_firmware
    response = device.write_firmware()  # do actual flash firmware.
  File "/Users/admin/Downloads/flash-shelly.py", line 659, in write_firmware
    response = requests.post(url=f'http://{self.wifi_ip}/update', auth=HTTPDigestAuth(self.username, self.password), files=files)
  File "/usr/local/lib/python3.9/site-packages/requests/api.py", line 117, in post
    return request('post', url, data=data, json=json, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/requests/api.py", line 61, in request
    return session.request(method=method, url=url, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/requests/sessions.py", line 542, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/local/lib/python3.9/site-packages/requests/sessions.py", line 697, in send
    r.content
  File "/usr/local/lib/python3.9/site-packages/requests/models.py", line 836, in content
    self._content = b''.join(self.iter_content(CONTENT_CHUNK_SIZE)) or b''
  File "/usr/local/lib/python3.9/site-packages/requests/models.py", line 761, in generate
    raise ChunkedEncodingError(e)
requests.exceptions.ChunkedEncodingError: ("Connection broken: ConnectionResetError(54, 'Connection reset by peer')", ConnectionResetError(54, 'Connection reset by peer'))

Devices found: 8 Upgradeable: 1 Flashed: 0
Log file created: debug.log

andyblac, 2021-12-17:
`ConnectionResetError: [Errno 54] Connection reset by peer` same error as @timoschilling

rojer, 2021-12-19:
hm, sorry, i don't understand - alpha3 cannot update?
seems to be working for me, uploading firmware to an alpha3 device worked just fine.

andyblac, 2021-12-19:
@rojer it seem to going from 2.10.1 direct to alpha3, (I have not tried myself, getting rooms re-plastered here, everything upside down atm.)

timoschilling, 2021-12-19:
@rojer im not able to flash a 2.10.1/2.10.0/2.9.x to 2.11.0-alpha3. The upload via script, via curl and via UI didn’t work.

lucaspinelli85, 2021-12-19:
Can't update[image]

rojer, 2021-12-23:
ah, you're right. 2.5 firmware size has crossed the magic 716KB. 2.5 is the biggest of them and i did not happen to have one on hand for testing the alpha.
i will look at the things to trim, or this may in fact be the time when we need to shrink the FS to make more room for code.

rojer, 2021-12-23:
i dropped rpc-service-ota and this freed up some space, so 2.5 should fit again. pushed alpha4, please give it a go.

miguelpruivo, 2021-12-24:
Hi! I have one question regarding 1 plus integration: does it uses Wi-Fi, Bluetooth or both to connect to a HomeKit hub (iPad, HomePod etc)? Because since it has a Bluetooth connection it might be handy to use it as a fallback only.

Also regarding this, is it possible to completely disable Bluetooth module and use only Wi-Fi in the settings? This should save some power while also preventing collision interference when a lot of IoT devices are preset.

Thank you!

Multisaft7, 2021-12-24:
Did I do something wrong when I try to do the firmware upgrade from my shelly 1 plus?

`http://192.168.77.92/ota?url=http://rojer.me/files/shelly/2.11.0-alpha4/shelly-homekit-ShellyPlus1.zip`

It says "Not Found"?!

rojer, 2021-12-24:
> Hi! I have one question regarding 1 plus integration: does it uses Wi-Fi, Bluetooth or both to connect to a HomeKit hub (iPad, HomePod etc)? Because since it has a Bluetooth connection it might be handy to use it as a fallback only.

so, situation with bluetooth is kinda unfortunate. i tried it, and it works but it doesn;t work seamlessly with wifi. once a bluetooth accessory, always a bluetooth accessory. accrding to apple, having accessory available on both BLE and IP transports is [not supported](https://github.com/apple/HomeKitADK/issues/112). so when wifi becomes available, hub still insists on using BT, which in practice is slower and less responsive. additionally, bridge mode is not available over bluetooth and we use "bridge" mode to expose multiple accessories. so, in short, at the moment BLE is not supported. it's not much of a problem since when setting up shelly-hk, it's already connected to wifi.

> firmware upgrade from my shelly 1 plus?

plus series cannot be upgraded this way, see instructions [here](https://github.com/mongoose-os-apps/shelly-homekit/issues/800#issuecomment-950398372).

miguelpruivo, 2021-12-24:
@rojer so basically the 1 plus with HomeKit has Bluetooth completely disable? It’s not an issue for me, actually. I’d rather have Bluetooth off and all on Wi-Fi to prevent network collisions than having a lot of Shelly’s using both Wi-Fi and Bluetooth.

rojer, 2021-12-24:
@miguelpruivo correct, currently bluetooth is not enabled in shelly-hk frmware.

lucaspinelli85, 2021-12-25:
Hi all!
I managed to upgrade to alfa4, i have these questions:
1- how is the homekit qrcode generated now having two wifi? based on 1? or on the active one?
2- how does it connect to one wifi rather than the other? try the first if it fails, go to the second? or try both and connect to the best?
3- what happens if I put the same wifi in both?
4- by default the ap mode will be disabled, it would be convenient to auto-enable when the device fails to connect to the two wifi.
Thanks!

rojer, 2021-12-25:
off> Hi all! I managed to upgrade to alfa4, i have these questions:
> 1- how is the homekit qrcode generated now having two wifi? based on 1? or on the active one?

based on wifi 1

> 2- how does it connect to one wifi rather than the other? try the first if it fails, go to the second? or try both and connect to the best?

try all usable APs (RSSI > -75) for wifi 1 first, from best to worst, then wifi 2. 

> 3- what happens if I put the same wifi in both?

nothing much. APs will be deduplicated.

> 4- by default the ap mode will be disabled, it would be convenient to auto-enable when the device fails to connect to the two wifi.

this will be the case during initial setup. however, not after connection was established.
it might be convenient but since APs are open by default, it's kind of a security problem if device throws itself wide open on possibly transient connection failure. i think wifi reset via on/off or physical button is easy enough that this is not necessary. 

> Thanks!

it-pilot23, 2021-12-26:
Hi,
Manual calibration of roller shutter from 2.11.0-alpha1 is not supported in 2.11.0-alpha4.
When opening, the idle position never reaches 100. It stops at 99. Increasing the movement and limiting time unfortunately does not fix the problem. When the roller shutter is opened manually and later the HomeKit automation wants to open the roller shutter again at idle pos 99, the Home app always says "will open".

Thank you.

bmpalmeida, 2021-12-26:
Hello all,

i've installed alpha4 on a shelly 1PM before setup it on HomeKit, and every time i click setup i receive a request timeout (the interesting bit is that it was discoverable in the home app but i didn't knew the code).

This specific device is with RSSI -86 which could not help, but with 2.10.1 i manage to setup it perfectly.

timoschilling, 2021-12-26:
@bmpalmeida I noticed something similar, try do make a reset and retry then.

rojer, 2021-12-27:
timeout when setting up homekit? interesting, i'll take a look. are you connected to AP or STA address at the time?

bmpalmeida, 2021-12-27:
> timeout when setting up homekit? interesting, i'll take a look. are you connected to AP or STA address at the time?

STA Address, but with a low RSSI (-86).

Another strange thing that was happen, i see the AP broadcasted even with the device connected to STA, but never manage to connect directly. Now its fine, but i didn't understand what happen for it to get "normal".

rojer, 2021-12-27:
> i see the AP broadcasted even with the device connected to STA

it should turn itself off after a minute of inactivity on the AP.
so, if you started in AP mode, configured STA, it connected, AP will remain up as long as you have the page open. when you close it, you should see AP disabled after 1 minute.

> but never manage to connect directly

what do you mean by that?

bmpalmeida, 2021-12-28:
> 

I didn't know that the AP would continue open.

After i already manage to open the page via my network, i could see the AP, connect to it, but the page never opens.

Multisaft7, 2021-12-28:
Works now for me like a charm with my Shelly Plus 1 as a Light :-)

lucaspinelli85, 2022-01-03:
I have a problem with alpha4.
In my Shelly2.5 roller shutter mode an obstacle continues to occur while running.
Both outward and return.
I have tried calibration numerous times.
Can you help me understand what the problem is?
I share the log ..

32945749 shelly_debug.cpp:231    No log file, sending new entries
34504107 shelly_main.cpp:526     Up 34.49, HAP 0/5/12 ns 5, RAM: 24344/41872; st 54; 4.1: c:1 mp:8.43 mt_ms:20589 cp:80.14 tp:80.14 md:0 lemd:0 lhmd:0
42504191 shelly_main.cpp:526     Up 42.49, HAP 0/5/12 ns 5, RAM: 24344/41872; st 53; 4.1: c:1 mp:8.43 mt_ms:20589 cp:80.14 tp:80.14 md:0 lemd:0 lhmd:0
50504010 shelly_main.cpp:526     Up 50.49, HAP 0/5/12 ns 5, RAM: 24344/41872; st 53; 4.1: c:1 mp:8.43 mt_ms:20589 cp:80.14 tp:80.14 md:0 lemd:0 lhmd:0
58504139 shelly_main.cpp:526     Up 58.49, HAP 0/5/12 ns 5, RAM: 24344/41872; st 53; 4.1: c:1 mp:8.43 mt_ms:20589 cp:80.14 tp:80.14 md:0 lemd:0 lhmd:0
60993061 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:64176
62001091 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:64176
62992865 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:64176
64016788 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:64176
64718383 mg_rpc.c:310            Shelly.SetState via WS_in 192.168.0.3:64176
64725610 shelly_hap_window_c:347 WC 1: Tgt pos 80.14 -> 100.00 (RPC)
64764674 shelly_hap_window_c:327 WC 1: State: idle -> move (0 -> 20)
64819447 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:64176
64863909 shelly_output.cpp:63    Output 2: off -> on (move)
64870647 shelly_hap_window_c:327 WC 1: State: move -> rampup (20 -> 22)
64964831 shelly_hap_window_c:549 P = 9.60 -> 8.43
64971328 shelly_hap_window_c:327 WC 1: State: rampup -> moving (22 -> 23)
64992356 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:64176
65066556 shelly_hap_window_c:336 WC 1: Cur pos 80.14 -> 80.56, P = 1.52
65072225 shelly_hap_window_c:599 Moving to 100, p 1.52
65866546 shelly_hap_window_c:336 WC 1: Cur pos 83.97 -> 84.45, P = 8.33
65871732 shelly_hap_window_c:599 Moving to 100, p 8.33
65993155 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:64176
66504326 shelly_main.cpp:526     Up 66.49, HAP 0/5/12 ns 5, RAM: 24296/41872; st 54; 4.1: c:1 mp:8.43 mt_ms:20589 cp:87.37 tp:100.00 md:1 lemd:0 lhmd:0
66667101 shelly_hap_window_c:336 WC 1: Cur pos 87.85 -> 88.34, P = 8.29
66672264 shelly_hap_window_c:599 Moving to 100, p 8.29
66994865 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:64176
67466789 shelly_hap_window_c:336 WC 1: Cur pos 91.73 -> 92.22, P = 8.57
67471941 shelly_hap_window_c:599 Moving to 100, p 8.57
67994737 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:64176
68266918 shelly_hap_window_c:336 WC 1: Cur pos 95.62 -> 96.11, P = 10.31
68272195 shelly_hap_window_c:599 Moving to 100, p 10.31
68994444 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:64176
69067129 shelly_hap_window_c:336 WC 1: Cur pos 99.51 -> 100.00, P = 8.73
69072354 shelly_hap_window_c:599 Moving to 100, p 8.73
69865297 shelly_hap_window_c:599 Moving to 100, p 1.03
70005783 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:64176
70665425 shelly_hap_window_c:599 Moving to 100, p 1.14
71136198 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:64176
71465275 shelly_hap_window_c:599 Moving to 100, p 1.29
72080559 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:64176
72264939 shelly_hap_window_c:599 Moving to 100, p 1.03
73065392 shelly_hap_window_c:599 Moving to 100, p 1.18
73565509 shelly_output.cpp:63    Output 2: on -> off (moving)
73572276 shelly_hap_window_c:327 WC 1: State: moving -> stop (23 -> 24)
73823682 mgos_sys_config.c:323   Saved to conf9.json
73831760 shelly_hap_window_c:327 WC 1: State: stop -> stopping (24 -> 25)
73851843 shelly_hap_window_c:327 WC 1: State: stopping -> idle (25 -> 0)
74165649 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:64176
74504435 shelly_main.cpp:526     Up 74.49, HAP 0/5/12 ns 5, RAM: 24344/41872; st 54; 4.1: c:1 mp:8.43 mt_ms:20589 cp:100.00 tp:100.00 md:0 lemd:0 lhmd:0
75168458 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:64176
75484299 mg_rpc.c:310            Shelly.SetState via WS_in 192.168.0.3:64176
75491470 shelly_hap_window_c:347 WC 1: Tgt pos 100.00 -> 0.00 (RPC)
75565767 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:64176
75590366 shelly_hap_window_c:327 WC 1: State: idle -> move (0 -> 20)
75664106 shelly_output.cpp:63    Output 1: off -> on (move)
75670730 shelly_hap_window_c:327 WC 1: State: move -> rampup (20 -> 22)
75764577 shelly_hap_window_c:549 P = 10.44 -> 8.43
75771093 shelly_hap_window_c:327 WC 1: State: rampup -> moving (22 -> 23)
76065044 shelly_hap_window_c:599 Moving to 0, p 5.90
76162494 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:64176
76466633 shelly_hap_window_c:336 WC 1: Cur pos 97.15 -> 96.66, P = 9.25
76865265 shelly_hap_window_c:599 Moving to 0, p 10.24
[... 322 lines cut ...]
175621139 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:64176
176833063 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:64176
178504640 shelly_main.cpp:526     Up 178.49, HAP 0/3/12 ns 3, RAM: 24332/41872; st 55; 4.1: c:1 mp:8.43 mt_ms:20589 cp:100.00 tp:100.00 md:0 lemd:0 lhmd:0
179739547 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:64176
[... cut ...]
266504262 shelly_main.cpp:526     Up 266.49, HAP 0/2/12 ns 2, RAM: 25696/41872; st 54; 4.1: c:1 mp:8.43 mt_ms:20589 cp:100.00 tp:100.00 md:0 lemd:0 lhmd:0
274504514 shelly_main.cpp:526     Up 274.49, HAP 0/2/12 ns 2, RAM: 25696/41872; st 54; 4.1: c:1 mp:8.43 mt_ms:20589 cp:100.00 tp:100.00 md:0 lemd:0 lhmd:0
282504343 shelly_main.cpp:526     Up 282.49, HAP 0/2/12 ns 2, RAM: 25696/41872; st 54; 4.1: c:1 mp:8.43 mt_ms:20589 cp:100.00 tp:100.00 md:0 lemd:0 lhmd:0
290504335 shelly_main.cpp:526     Up 290.49, HAP 0/2/12 ns 2, RAM: 25696/41872; st 54; 4.1: c:1 mp:8.43 mt_ms:20589 cp:100.00 tp:100.00

rojer, 2022-01-03:
@lucaspinelli85 "obstruction" condition is raised if we detect high power consumption during movement.
in this case, during calibration we determined that normal power consumption is 8.43 watts but at some point it spikes to over 24:
```
89666539 shelly_hap_window_c:586 Obstruction: p = 23.74 t = 13887
...
134466523 shelly_hap_window_c:586 Obstruction: p = 24.35 t = 7687
```
this is over 2.5 times the normal, so movement is stopped. the real question is why power spikes so high compared to normal.
it's also strange that calibrated power value is just 8 watts, i see values > 10 consistently during movement in both directions.
can you run calibration and post the log?

lucaspinelli85, 2022-01-03:
Thanks for your answer.
My avg power is 8w, so I set the threshold to 5w otherwise the calibration was not successful (default is 10w).
I have no idea why it goes above 20w, but there is no obstacle!
Is there a "curl" in which I can raise the maximum threshold?
Sorry for the off-topic, I thought it was a firmware problem ...

rojer, 2022-01-03:
@lucaspinelli85 makes sense. but based on th elogs you posted it should calibrate to something higher, more like 10. in which case obstruction detection would not kick in. can you run calibration and post the logs? i'd like to understand why it comes up with 8 as average power consumption.

the obstruction detection threshold is fixed at 2.5 for the moment, there is no config value for it. we can always add one, but before we go for that, i'm thinking - if this is indeed just a short term spike, we may be able to get over it by being a bit more patient. currently if at any point we measure p > mp * 2.5, we panic and declare obstruction. but instead, maybe we should keep a moving average over a few seconds, to smooth out short spikes. let me try it.

p.s. i [lowered the default idle power threshold to 5](https://github.com/mongoose-os-apps/shelly-homekit/commit/3fd69047870cfdd66ef7f2a73c2e006c95c9b0d2) so you won't need to adjust it manually in newer version.

lucaspinelli85, 2022-01-03:
This il log of calibration but
he did not go all the way, he did not reach the limit switch...

8528510749 shelly_debug.cpp:231    No log file, sending new entries
8530504409 shelly_main.cpp:526     Up 8530.49, HAP 0/3/12 ns 3, RAM: 25316/41872; st 54; 4.1: c:1 mp:8.43 mt_ms:20589 cp:100.00 tp:100.00 md:0 lemd:0 lhmd:0
8538504353 shelly_main.cpp:526     Up 8538.49, HAP 0/3/12 ns 3, RAM: 25316/41872; st 54; 4.1: c:1 mp:8.43 mt_ms:20589 cp:100.00 tp:100.00 md:0 lemd:0 lhmd:0
8546504439 shelly_main.cpp:526     Up 8546.49, HAP 0/3/12 ns 3, RAM: 25316/41872; st 53; 4.1: c:1 mp:8.43 mt_ms:20589 cp:100.00 tp:100.00 md:0 lemd:0 lhmd:0
8550103004 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65080
8551101541 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65080
8552127027 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65080
8553153222 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65080
8554161279 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65080
8554504522 shelly_main.cpp:526     Up 8554.49, HAP 0/3/12 ns 3, RAM: 25316/41872; st 54; 4.1: c:1 mp:8.43 mt_ms:20589 cp:100.00 tp:100.00 md:0 lemd:0 lhmd:0
8555189567 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65080
8556211373 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65080
8557221273 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65080
8557913664 mg_rpc.c:310            Shelly.SetState via WS_in 192.168.0.3:65080
8557933007 shelly_hap_window_c:327 WC 1: State: idle -> precal0 (0 -> 10)
8557959776 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65080
8558031432 shelly_hap_window_c:449 Begin calibration
8558189738 mgos_sys_config.c:323   Saved to conf9.json
8558196787 shelly_output.cpp:63    Output 2: off -> on (precal0)
8558202815 shelly_hap_window_c:327 WC 1: State: precal0 -> cal0 (10 -> 11)
8558216318 shelly_hap_window_c:465 WC 1: P0 = 1.238
8558253344 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65080
8558933006 shelly_hap_window_c:465 WC 1: P0 = 1.189
8559261218 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65080
8559732700 shelly_hap_window_c:465 WC 1: P0 = 1.067
8560292437 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65080
8560532293 shelly_hap_window_c:465 WC 1: P0 = 1.116
8561307150 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65080
8561332465 shelly_hap_window_c:465 WC 1: P0 = 1.030
8562132515 shelly_hap_window_c:465 WC 1: P0 = 1.335
8562334717 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65080
8562504496 shelly_main.cpp:526     Up 8562.49, HAP 0/3/12 ns 3, RAM: 25316/41872; st 54; 4.1: c:0 mp:8.43 mt_ms:20589 cp:100.00 tp:100.00 md:0 lemd:0 lhmd:0
8562932536 shelly_hap_window_c:465 WC 1: P0 = 1.268
8563359903 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65080
8563732568 shelly_hap_window_c:465 WC 1: P0 = 1.213
8564360615 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65080
8564532443 shelly_hap_window_c:465 WC 1: P0 = 1.165
8565332222 shelly_hap_window_c:465 WC 1: P0 = 1.079
8565379995 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65080
8566132441 shelly_hap_window_c:465 WC 1: P0 = 1.159
8566332844 shelly_output.cpp:63    Output 2: on -> off (cal0)
8566339268 shelly_hap_window_c:327 WC 1: State: cal0 -> postcal0 (11 -> 12)
8566401692 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65080
8566433468 shelly_hap_window_c:327 WC 1: State: postcal0 -> precal1 (12 -> 13)
8566532319 shelly_output.cpp:63    Output 1: off -> on (precal1)
8566538663 shelly_hap_window_c:327 WC 1: State: precal1 -> cal1 (13 -> 14)
8566633034 shelly_hap_window_c:496 WC 1: P1 = 13.128
8567429526 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65080
8567453014 shelly_hap_window_c:496 WC 1: P1 = 11.335
8568232678 shelly_hap_window_c:496 WC 1: P1 = 10.976
8569033433 shelly_hap_window_c:496 WC 1: P1 = 11.189
8569460477 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65080
8569832674 shelly_hap_window_c:496 WC 1: P1 = 10.988
8570504907 shelly_main.cpp:526     Up 8570.49, HAP 0/3/12 ns 3, RAM: 23736/41872; st 54; 4.1: c:0 mp:8.43 mt_ms:20589 cp:100.00 tp:100.00 md:0 lemd:0 lhmd:0
8570632723 shelly_hap_window_c:496 WC 1: P1 = 10.677
8571433059 shelly_hap_window_c:496 WC 1: P1 = 10.518
8571500556 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65080
8572232588 shelly_hap_window_c:496 WC 1: P1 = 10.848
8572525187 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65080
8572549867 shelly_output.cpp:63    Output 1: on -> off (cal1)
8572558362 shelly_hap_window_c:500 WC 1: calibration done, move_time 6005, move_power 10.876
8572564557 shelly_hap_window_c:327 WC 1: State: cal1 -> postcal1 (14 -> 15)
8572789844 mgos_sys_config.c:323   Saved to conf9.json
8572798485 shelly_hap_window_c:347 WC 1: Tgt pos 100.00 -> 50.00 (postcal1)
8572804804 shelly_hap_window_c:327 WC 1: State: postcal1 -> idle (15 -> 0)
8572818560 shelly_hap_window_c:327 WC 1: State: idle -> move (0 -> 20)
8572831476 shelly_output.cpp:63    Output 2: off -> on (move)
[... 17 lines cut ...]
8577628175 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65080
8578504460 shelly_main.cpp:526     Up 8578.49, HAP 0/3/12 ns 3, RAM: 25316/41872; st 54; 4.1: c:1 mp:10.88 mt_ms:6005 cp:49.73 tp:49.73 md:0 lemd:0 lhmd:0
[... cut ...]
8682504110 shelly_main.cpp:526     Up 8682.49, HAP 0/3/12 ns 3, RAM: 25316/41872; st 53; 4.1: c:1 mp:10.88 mt_ms:6005 cp:49.73 tp:49.73 md:0 lemd:0 lhmd:0
8690504170 shelly_main.cpp:526     Up 8690.49, HAP 0/3/12 ns 3, RAM: 25316/41872; st 53; 4.1: c:1 mp:10.88 mt_ms:6005 cp:49.73 tp:49.73 md:0 lemd:0 lhmd:0
8698504561 shelly_main.cpp:526     Up 8698.49, HAP 0/3/12 ns 3, RAM: 25316/41872; st 53; 4.1: c:1 mp:10.88 mt_ms:6005 cp:49.73 tp:49.73 md:0 lemd:0 lhmd:0
8706504495 shelly_main.cpp:526     Up 8706.49, HAP 0/3/12 ns 3, RAM: 25316/41872; st 53; 4.1: c:1 mp:10.88 mt_ms:6005 cp:49.73 tp:49.73 md:0 lemd:0 lhmd:0

lucaspinelli85, 2022-01-03:
I repeated the calibration starting from the middle of the stroke instead of from the limit switch.
Now it has gone up to 10W, it should be enough, but very close to the limit...
Surely being able to evade the peak power, safeguarding the safety of the obstacle, or being able to edit the maximum obstacle intervention threshold would be an improvement. Thanks for the help!
Please also take into consideration my other issues regarding roller shutter mode with low power and / or no absorption motors.
Happy New Year!!!

rojer, 2022-01-03:
@lucaspinelli85 looking at the calibration log, cal0 phase looks weird:
```
8558196787 shelly_output.cpp:63 Output 2: off -> on (precal0)
8558202815 shelly_hap_window_c:327 WC 1: State: precal0 -> cal0 (10 -> 11)
8558216318 shelly_hap_window_c:465 WC 1: P0 = 1.238
8558253344 mg_rpc.c:310 Shelly.GetInfoExt via WS_in 192.168.0.3:65080
8558933006 shelly_hap_window_c:465 WC 1: P0 = 1.189
8559261218 mg_rpc.c:310 Shelly.GetInfoExt via WS_in 192.168.0.3:65080
8559732700 shelly_hap_window_c:465 WC 1: P0 = 1.067
8560292437 mg_rpc.c:310 Shelly.GetInfoExt via WS_in 192.168.0.3:65080
8560532293 shelly_hap_window_c:465 WC 1: P0 = 1.116
8561307150 mg_rpc.c:310 Shelly.GetInfoExt via WS_in 192.168.0.3:65080
8561332465 shelly_hap_window_c:465 WC 1: P0 = 1.030
8562132515 shelly_hap_window_c:465 WC 1: P0 = 1.335
8562334717 mg_rpc.c:310 Shelly.GetInfoExt via WS_in 192.168.0.3:65080
8562504496 shelly_main.cpp:526 Up 8562.49, HAP 0/3/12 ns 3, RAM: 25316/41872; st 54; 4.1: c:0 mp:8.43 mt_ms:20589 cp:100.00 tp:100.00 md:0 lemd:0 lhmd:0
8562932536 shelly_hap_window_c:465 WC 1: P0 = 1.268
8563359903 mg_rpc.c:310 Shelly.GetInfoExt via WS_in 192.168.0.3:65080
8563732568 shelly_hap_window_c:465 WC 1: P0 = 1.213
8564360615 mg_rpc.c:310 Shelly.GetInfoExt via WS_in 192.168.0.3:65080
8564532443 shelly_hap_window_c:465 WC 1: P0 = 1.165
8565332222 shelly_hap_window_c:465 WC 1: P0 = 1.079
8565379995 mg_rpc.c:310 Shelly.GetInfoExt via WS_in 192.168.0.3:65080
8566132441 shelly_hap_window_c:465 WC 1: P0 = 1.159
8566332844 shelly_output.cpp:63 Output 2: on -> off (cal0)
8566339268 shelly_hap_window_c:327 WC 1: State: cal0 -> postcal0 (11 -> 12)
```
there's barely any power consumed, just 1 watt. this is under even your lower threshold of 5, and stays that way, so in fact it ends early after allowed ramp-up time (5 seconds) expires. looking at your previous log, your motor is consuming about 10W moving in both directions, so what was it doing during these 5 seconds? no idea...

now, i have made a [change](https://github.com/mongoose-os-apps/shelly-homekit/commit/658bb5dc01c8d2c3a254a2afdb1c30953022e9da) to obstruction detection to require that overpower condition persists for 3 seconds, and also moved some hard-coded constants into config. here's a build with it: http://rojer.me/files/shelly/misc/shelly25-868.zip can you give it a try?

lucaspinelli85, 2022-01-03:
Yes i try!

lucaspinelli85, 2022-01-03:
this is the log, do you see something strange? now avg power is 9w

51947684 shelly_debug.cpp:231    No log file, sending new entries
57055877 shelly_main.cpp:525     Up 57.04, HAP 0/3/12 ns 3, RAM: 24832/41880; st 55; 4.1: c:1 mp:10.20 ip:1.00 mt_ms:20889 cp:58.41 tp:58.41 md:0 lemd:0 lhmd:0
65056110 shelly_main.cpp:525     Up 65.04, HAP 0/3/12 ns 3, RAM: 24832/41880; st 55; 4.1: c:1 mp:10.20 ip:1.00 mt_ms:20889 cp:58.41 tp:58.41 md:0 lemd:0 lhmd:0
73055984 shelly_main.cpp:525     Up 73.04, HAP 0/3/12 ns 3, RAM: 24832/41880; st 55; 4.1: c:1 mp:10.20 ip:1.00 mt_ms:20889 cp:58.41 tp:58.41 md:0 lemd:0 lhmd:0
76631521 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65271
77651939 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65271
78769694 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65271
79739042 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65271
80720043 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65271
81056078 shelly_main.cpp:525     Up 81.04, HAP 0/3/12 ns 3, RAM: 24832/41880; st 55; 4.1: c:1 mp:10.20 ip:1.00 mt_ms:20889 cp:58.41 tp:58.41 md:0 lemd:0 lhmd:0
81176802 mg_rpc.c:310            Shelly.SetState via WS_in 192.168.0.3:65271
81202230 shelly_hap_window_c:327 WC 1: State: idle -> precal0 (0 -> 10)
81214476 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65271
81300469 shelly_hap_window_c:449 Begin calibration
81463748 mgos_sys_config.c:323   Saved to conf9.json
81470718 shelly_output.cpp:63    Output 2: off -> on (precal0)
81476722 shelly_hap_window_c:327 WC 1: State: precal0 -> cal0 (10 -> 11)
81488274 shelly_hap_window_c:465 WC 1: P0 = 2.817
81737702 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65271
82201201 shelly_hap_window_c:465 WC 1: P0 = 6.457
82751602 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65271
83001267 shelly_hap_window_c:465 WC 1: P0 = 7.098
83786443 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65271
83809898 shelly_hap_window_c:465 WC 1: P0 = 7.055
84601517 shelly_hap_window_c:465 WC 1: P0 = 7.707
84791370 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65271
85401193 shelly_hap_window_c:465 WC 1: P0 = 8.970
85843659 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65271
86201276 shelly_hap_window_c:465 WC 1: P0 = 8.591
86831346 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65271
87001253 shelly_hap_window_c:465 WC 1: P0 = 8.409
87801441 shelly_hap_window_c:465 WC 1: P0 = 8.957
87854066 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65271
88601491 shelly_hap_window_c:465 WC 1: P0 = 8.201
88886166 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65271
89055978 shelly_main.cpp:525     Up 89.04, HAP 0/3/12 ns 3, RAM: 24832/41880; st 55; 4.1: c:0 mp:10.20 ip:1.00 mt_ms:20889 cp:58.41 tp:58.41 md:0 lemd:0 lhmd:0
89401244 shelly_hap_window_c:465 WC 1: P0 = 8.409
89891669 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65271
90204813 shelly_hap_window_c:465 WC 1: P0 = 8.317
90928302 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65271
91001190 shelly_hap_window_c:465 WC 1: P0 = 1.177
91801179 shelly_hap_window_c:465 WC 1: P0 = 1.360
91931491 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65271
92601447 shelly_hap_window_c:465 WC 1: P0 = 1.165
92959395 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65271
93401550 shelly_hap_window_c:465 WC 1: P0 = 1.244
93985507 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65271
94201302 shelly_hap_window_c:465 WC 1: P0 = 1.110
94998113 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65271
95022774 shelly_hap_window_c:465 WC 1: P0 = 1.421
95701684 shelly_output.cpp:63    Output 2: on -> off (cal0)
95708187 shelly_hap_window_c:327 WC 1: State: cal0 -> postcal0 (11 -> 12)
95802273 shelly_hap_window_c:327 WC 1: State: postcal0 -> precal1 (12 -> 13)
95900891 shelly_output.cpp:63    Output 1: off -> on (precal1)
95907247 shelly_hap_window_c:327 WC 1: State: precal1 -> cal1 (13 -> 14)
96001766 shelly_hap_window_c:496 WC 1: P1 = 10.024
96013525 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65271
96801589 shelly_hap_window_c:496 WC 1: P1 = 8.890
97031147 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65271
97065859 shelly_main.cpp:525     Up 97.05, HAP 0/3/12 ns 3, RAM: 23040/41880; st 55; 4.1: c:0 mp:10.20 ip:1.00 mt_ms:20889 cp:58.41 tp:58.41 md:0 lemd:0 lhmd:0
97601438 shelly_hap_window_c:496 WC 1: P1 = 8.530
98060332 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65271
98401702 shelly_hap_window_c:496 WC 1: P1 = 9.512
99087054 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65271
99201506 shelly_hap_window_c:496 WC 1: P1 = 9.707
100001372 shelly_hap_window_c:496 WC 1: P1 = 9.409
100091372 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65271
100801404 shelly_hap_window_c:496 WC 1: P1 = 9.738
[... 61 lines cut ...]
122539883 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65271
122603437 shelly_hap_window_c:336 WC 1: Cur pos 26.52 -> 27.00, P = 8.50
123403423 shelly_hap_window_c:336 WC 1: Cur pos 30.39 -> 30.87, P = 9.17
123553898 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65271
124203440 shelly_hap_window_c:336 WC 1: Cur pos 34.25 -> 34.74, P = 8.57
[... cut ...]
129056190 shelly_main.cpp:525     Up 129.04, HAP 0/3/12 ns 3, RAM: 25048/41880; st 56; 4.1: c:1 mp:9.76 ip:1.00 mt_ms:20689 cp:49.72 tp:49.72 md:0 lemd:0 lhmd:0
129680535 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65271
130697709 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65271
131719013 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65271
132739187 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.0.3:65271
137056063 shelly_main.cpp:525     Up 137.04, HAP 0/3/12 ns 3, RAM: 25048/41880; st 56; 4.1: c:1 mp:9.76 ip:1.00 mt_ms:20689 cp:49.72 tp:49.72 md:0 lemd:0 lhmd:0

rojer, 2022-01-03:
@lucaspinelli85 everything seems normal. but i also don't see the power spike to 24W during movement, so no real test happened. try a few times, see if you can reproduce the false obstruction detection with the new code.

lucaspinelli85, 2022-01-03:
ok, I'll try to see if I can replicate the peak or if it still happens, or if with this firmware that you changed me it doesn't give any more problems and I'll let you know! Thanks again!

rojer, 2022-01-05:
pushed alpha5

lucaspinelli85, 2022-01-05:
> ok, I'll try to see if I can replicate the peak or if it still happens, or if with this firmware that you changed me it doesn't give any more problems and I'll let you know! Thanks again!

With firmware "868" I have no longer had problems with obstacles.
Is the change also present in alpha5?

rojer, 2022-01-05:
yes, it's been merged to master and is present in alpha5 onwards.

bmpalmeida, 2022-01-05:
Just a small detail, using the script on the 2.11.1 it no longer reports the RSSI of the devices.

timoschilling, 2022-01-05:
@rojer have investigated this?

> i've installed alpha4 on a shelly 1PM before setup it on HomeKit, and every time i click setup i receive a request timeout

It's still an issue. Reproducible via:
1. Take a Stock devise (reverted to stock works too)
2. flash to 2.10.1 (maybe optional)
3. flash to 2.11.0-alpha5
4. open UI
5. click Setup in HomeKit Settings section
6. result: Request timeout

Doing the same on the same Shells at the same place but skipping step 3 and using 2.10.1 works.
If I reset the device (via System section), bringing back to Wifi and clicking Setup again and it works.

rojer, 2022-01-06:
@timoschilling thanks! timeout was too short, fixed in alpha6

rojer, 2022-01-06:
@bmpalmeida yes, script will need to be adjusted for new format of wifi info and config.

timoschilling, 2022-01-06:
@rojer you missed to push the code for alpha6, repo still has only alpha5.

rojer, 2022-01-06:
@timoschilling you're right, forgot to push the master branch. done.

rojer, 2022-01-06:
pushed alpha7 with fixes for #884

sebashb, 2022-01-07:
Just installed alpha7 on a Shelly 1 Plus that is controlling a garage door. Working flawlessly! Good work! I just want to mention that I was not able to upload the firmware with the Python Script in the master, it was not able to find the Shelly 1 Plus, it only found a Shelly 3EM that I also have. Nonetheless, with the curl commands from [here](https://github.com/mongoose-os-apps/shelly-homekit/issues/800#issuecomment-950398372) it worked perfectly.

andyblac, 2022-01-07:
> Just installed alpha7 on a Shelly 1 Plus that is controlling a garage door. Working flawlessly! Good work! I just want to mention that I was not able to upload the firmware with the Python Script in [...]

the script is not compatible with GEN2 devices, the stock API is vastly different, and thus, I'm not sure it worth the effort in rewriting the script for the 3rd time.

sebashb, 2022-01-07:
Oh I was misled by the phrase "Or use script: `tools/flash-shelly.py -a -V 2.11.0-alpha7`" in the first post of this issue. For me curl is perfect and pretty easy to upload also.

rojer, 2022-01-07:
i think the script should support plus devices. shouldn't need a full rewrite. Andy, if you don't have the time i can take a look.

andyblac, 2022-01-07:
> i think the script should support plus devices. shouldn't need a full rewrite. Andy, if you don't have the time i can take a look.

I took a look at the GEN2 API and it really awful, and messy, loads of missing info, ie the scanner just returns "GEN2" thats it nothing else. if you do want to take Ia a look use version 3 of script that is way more unto to date.

rojer, 2022-01-09:
pushed alpha8 with temperature add-on support (thanks @markirb !)

partytrik, 2022-01-10:
@rojer Upgrading 30 devices to Alpha 7, 29 went fine, but 1 of them doesn't load the web interface, when trying to open it on static IP. It seems to have connected to the router, the router as well as a LAN scanner show it as active device under its static IP. That's a behaviour I have not once experienced with any of the 20 or so updates since the earliest versions

rojer, 2022-01-10:
@partytrik
1. does it respond to pings (`ping A.B.C.D`)?
2. does it respond over HAP?
3. can you fetch the page content with curl (`curl http://A.B.C.D/`)?
4. can you load other pages, like `/debug/info`?

partytrik, 2022-01-10:
@rojer 

1. No
2. Yes !!
3. if you meant via Terminal, the result was: "(7) failed to connect to A.B.C.D port 80: Operation timed out
4. No

[144 more replies]

closed 2022-06-19 by rojer