mongoose-os-apps/shelly-homekit #869: Curl requests not possible if password is set

n11f, 2021-12-17:
I finally managed to control my shelly 1 via curl using the following:

`curl http://my.ip/rpc/Shelly.SetState -d '{"id": 1, "type": 0, "state": {"state": true}}'`

However this will only work as long as there is no password set for security settings. If i protect my Shelly via password, this will not work anymore.

Do you have any idea on how to solve this? 

Thanks in advance

n11f, 2021-12-22:
anybody?

timoschilling, 2021-12-22:
have you tried this?
`curl http://user:password@ip/...`

n11f, 2021-12-22:
> have you tried this? `curl http://user:password@ip/...`

yes that came to my mind as well, but how should this be possible? There is only a password that can be set and no user at all.

andyblac, 2021-12-22:
> > have you tried this? `curl http://user:password@ip/...`
> 
> yes that came to my mind as well, but how should this be possible? There is only a password that can be set and no user at all.

user is `admin` in this firmware

n11f, 2021-12-23:
> user is `admin` in this firmware

that does not work either

timoschilling, 2021-12-23:
try `curl --digest -u admin:1234 http://...`

n11f, 2021-12-23:
> try `curl --digest -u admin:1234 http://...`

thank you guys, that finally solved it! Maybe you should take this to the docs as well

closed 2021-12-23 by n11f