mongoose-os-apps/shelly-homekit #870: Before update, always update stock firmware?

mensa84, 2021-12-18:
Hello,

the documentation says, that one should update to latest stock FW, before converting to Shelly-HomeKit.

So will I also have to periodically check latest stock firmware and update and then switch back to Shelly-HomeKit? Or will I have to do that only the very first time?

Is something of the stock firmware still running, although converted to Shelly-HomeKit?

sterl1i, 2021-12-18:
Thank you for asking. I was wondering that too! Do I have to completely setup my Shelly with the switch in order to do this or can I just power the Shelly?

mensa84, 2021-12-18:
> Thank you for asking. I was wondering that too! Do I have to completely setup my Shelly with the switch in order to do this or can I just power the Shelly?

Just power your shelly is enough. My question is different.

andyblac, 2021-12-18:
Just the very first time before you convert to HomeKit is all that is needed.

mensa84, 2021-12-18:
> Just the very first time before you convert to HomeKit is all that is needed.

Thanks, but isn't there a feature which will update the stock FW automatically before converting to HK?

rojer, 2021-12-18:
at the moment updating stock before flashing HK is not strictly necessary but it may become required in the future.
a feature of the script that would update stock before flashing hk would be nice.

closed 2021-12-18 by rojer