mongoose-os-apps/shelly-homekit #871: Inverted Input for Shelly Plug S Please

ScottLeMagicien, 2021-12-19:
Hi!

I love having an inverted input (and output!) for my Shelly1's, etc., and would like top have this option for my Shelly Plug S's as well.

The reason is that I like to be able to do a short power-off cycle to reset the devices attached to them.  Since this sometimes interrupts my WiFi network, it's great that I can use the Auto Off Delay to make sure that they power up again!  (Because the output is reversed.)  With the Shelly1's I can also invert the input, which means the Home app shows them as being ON normally.  Sadly with the Shelly Plug S there is no way to invert the input, so the Home app shows them as OFF - even when they are actually ON.

Thank you!!!

rojer, 2021-12-19:
"inverted input" is about the input pin of the shelly. as such, plug s has no inputs.

timoschilling, 2021-12-19:
@rojer yes the plug s didn’t have the software input component. I think he means the toggle button on the device

lucaspinelli85, 2021-12-19:
practically you need the auto on, like my issue #862

ScottLeMagicien, 2021-12-20:
Hi, yes, the input in this case would be the switch on the device and not an external switch as with a Shelly1.

If this isn't possible, an Auto On function would also work.  

Or a function that would simply invert the ON/Off state as it is displayed in the Home app.  (As is is now, using the Invert the Output function shows it in the Home app as being powered OFF when it is actually ON.)

rojer, 2022-01-05:
i think we can add "Invert HomeKit State" setting to all switches, easy enough to do and in combination with invert output and auto-off should give you what you want.

PR #1224 (2023-02-14): option to invert hap state
closes #871
closed 2023-02-14 by timoschilling