mongoose-os-apps/shelly-homekit #872: Shelly 1L Mongoose FW zip not loading

sterl1i, 2021-12-20:
This is my first Shelly FW upgrade so be patient. I have my Shelly 1L attached to my switch and wall. The Shelly app found it and I updated the original FW as your instructions stated. I set a static IP and also did so within my router.

When I attempted to go to the Shelly 1L zip website, the message said the server could not be found. (http://a.b.c.d/ota?url=http://rojer.me/files/shelly/shelly-homekit-Shelly1L.zip) I was following the video but it wouldn’t work that way.

I thought maybe I needed my laptop instead of iPad. That, too, said the server could not be found. I can’t figure out how to get to the download for my 1L. Any suggestions?

This is a one-pole, one-gang, no dim, not currently using LED light for outside sconce. Just want to set sunrise/sunset schedule. I have a bypass ready to install.

I will repeat this setup for another outside light in front but the switch will reside in a 2-gang box-still only one-pole. I have a bypass ready to install.

timoschilling, 2021-12-20:
Sounds like you didn’t replaced a.b.c.d with the IP of your Shelly in http://a.b.c.d/ota?url=http://rojer.me/files/shelly/shelly-homekit-Shelly1L.zip

sterl1i, 2021-12-20:
I didn’t read anywhere that I was suppose to. Thank you.

sterl1i, 2021-12-20:
Can anyone tell me if Shelly 1L works with a touch light switch? Someone said it might not but that was just one person.

timoschilling, 2021-12-20:
Depends on the touch switch. But I think here is not the right place for that question.

closed 2021-12-20 by timoschilling