mongoose-os-apps/shelly-homekit #875: Question: homebridge vs Shelly HK?

miguelpruivo, 2021-12-25:
Hi! First of all, merry Christmas to everyone!

Now, I’m planning on setting up a new house ceiling lights and I have a plenty of questions. 

The house will use led drivers everywhere (I’m not sure if they’re gonna to be dimmable or not beforehand, but one can assume that those can be changed later to any bulb).

I’m divided between Shelly 1 Plus to control all lights or Shelly Dimmer 2. The Plus has dry contacts which in one hand seem to be a good feature, however, the dimmer 2 has the dimming effect, although some claim to experience peak variation issues sometimes with it.

One thing that might help is related with Shelly HK firmware. It’s not yet supported for dimmer 2, but… I’m also using homebridge.

Despite great job and effort of doing this firmware I would like to sincerely know if using the Shelly-HK firmware has a real advantage vs. using the original Shelly fw + homebridge. What’s the pros and cons between the two usages? Because if using Shelly HK has any good advantage compared to homebridge (other than being independent of it, because server will always be online) I might start discarding dimming features and go with plus 1.

Also, a quick side note: plus 1 says <1.3 w while dimmer 2 says <1w consumption. Does that mean that I’ll have less consumption with dimmer (it’s important because I’ll have all the house setup with it).

Thank you!!!

timoschilling, 2021-12-25:
The power consumption of the Shelly is only the peak/max value. I have not tested the plus models yet, but the power consumption of the old models is way les then 1w, I mean it’s about 0.3W in average. I expect around the same, maybe +0.1W for the plus models. Plus had WiFi and Bluetooth, so it can consume more, but we only use WiFi. 

Plus 1 support will come in the next release. Dimmer 2 maybe in the future. 

Shelly HomeKit vs HomeBridge: this should not be a bashing, just my experience: the HomeKit on the device is much more reliable then via bridge. I need to restart my bridge one a week, Shelly’s with HK I need to restart maybe 1-2 devices per year. The Shelly’s with HK are much faster then via bridge. If you control multiple at once, all the HK going off at the same time, the bridged are only one by one after time. 

My setup is to have a bridge for Dimmer / Door Window / Motion and one device firmware for all supported devices. 

One other option could be Shelly Bulbs + i3 or i4 for you. We will bring support for all bulbs with 2.11 or 2.12.

miguelpruivo, 2021-12-25:
@timoschilling thank you for the inputs.

I’m the buyer of the house but not the owner of the project/construction, so I’ll ask them to buy the Shelly devices and install when mounting the switches/outlets. I won’t be able to be there to check whether everything is properly setup or not (but I assume they’ll test it, and they have experience with it because they were the ones suggesting me to use Shelly 2.5 in the rollers).

With all of this is mind, and since I can’t know if they’re going to use dimmable bulbs or not (I’m afraid they won’t), and based on your Shelly reliability experience, would you rather ask them to install the dimmer 2 or Shelly 1 plus devices in the light switches?

I’m tempted on the dimmers because of dimming support, but I also prioritize a lot reliability and peace of mind.

timoschilling, 2021-12-25:
Try or not didn’t matter for your case from my point of view. 

Dimmer or not is something you need to decide.

closed 2021-12-25 by timoschilling

miguelpruivo, 2021-12-25:
@timoschilling thank you once again. Do you have any idea when dimmers will be supported with Shelly HK?

timoschilling, 2021-12-26:
@miguelpruivo no