mongoose-os-apps/shelly-homekit #876: Shelly 1L & 2.5 Crashing

jasonyates, 2021-12-28:
I've had several Shelly relays installed for many months now with no issues. All are flashed with the Homekit firmware.

In the last 24 hours, 2 of them repeatedly reboot at seemingly random intervals. Watching the log reveals nothing obvious. In inputs turn off and there is an audible click of the relay closing / opening.

1283815188 shelly_main.cpp:504     Up 1283.80, HAP 0/3/12 ns 3, RAM: 24088/42200; st 58; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:1 in_st:1 inm:2 ininv:0
1283982199 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.113:50387
1284987010 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.113:50387
1285988194 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.113:50387
1286983658 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.113:50387
1287983203 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.113:50387
1288987232 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.113:50387
1289986087 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.113:50387
1290983859 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.113:50387
1291613156 shelly_main.cpp:504     Up 1291.60, HAP 0/3/12 ns 3, RAM: 23396/42200; st 58; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:1 in_st:1 inm:2 ininv:0
1299612809 shelly_main.cpp:504     Up 1299.60, HAP 0/3/12 ns 3, RAM: 24932/42200; st 58; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:1 in_st:1 inm:2 ininv:0
1300979885 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.113:50387
1301989300 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.113:50387
1302981102 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.113:50387
1303982589 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.113:50387
1304983331 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.113:50387
1305985556 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.113:50387
1306986987 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.113:50387
1307612924 shelly_main.cpp:504     Up 1307.60, HAP 0/3/12 ns 3, RAM: 23176/42200; st 58; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:1 in_st:1 inm:2 ininv:0
1307980848 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.113:50387
1308982441 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.113:50387
1309980161 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.113:50387
1315612827 shelly_main.cpp:504     Up 1315.60, HAP 0/3/12 ns 3, RAM: 24932/42200; st 58; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:1 in_st:1 inm:2 ininv:0
1323612725 shelly_main.cpp:504     Up 1323.60, HAP 0/3/12 ns 3, RAM: 24932/42200; st 58; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:1 in_st:1 inm:2 ininv:0
1331612622 shelly_main.cpp:504     Up 1331.60, HAP 0/3/12 ns 3, RAM: 24932/42200; st 58; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:1 in_st:1 inm:2 ininv:0
1339612588 shelly_main.cpp:504     Up 1339.60, HAP 0/3/12 ns 3, RAM: 24932/42200; st 58; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:1 in_st:1 inm:2 ininv:0
1347612970 shelly_main.cpp:504     Up 1347.60, HAP 0/3/12 ns 3, RAM: 24660/42200; st 58; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:1 in_st:1 inm:2 ininv:0
1355612933 shelly_main.cpp:504     Up 1355.60, HAP 0/3/12 ns 3, RAM: 24932/42200; st 59; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:1 in_st:1 inm:2 ininv:0
1363612939 shelly_main.cpp:504     Up 1363.60, HAP 0/3/12 ns 3, RAM: 24932/42200; st 59; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:1 in_st:1 inm:2 ininv:0
1371612677 shelly_main.cpp:504     Up 1371.60, HAP 0/3/12 ns 3, RAM: 24932/42200; st 59; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:1 in_st:1 inm:2 ininv:0
1379612580 shelly_main.cpp:504     Up 1379.60, HAP 0/3/12 ns 3, RAM: 24932/42200; st 59; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:1 in_st:1 inm:2 ininv:0
1387612904 shelly_main.cpp:504     Up 1387.60, HAP 0/3/12 ns 3, RAM: 24932/42200; st 59; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:1 in_st:1 inm:2 ininv:0
1395612724 shelly_main.cpp:504     Up 1395.60, HAP 0/3/12 ns 3, RAM: 24932/42200; st 59; 0.1: st:1 in_st:1 inm:2 ininv:0; 0.2: st:1 in_st:1 inm:2 ininv:0

jasonyates, 2021-12-28:
Some further testing on my side. It appears to reboot when it roams between AP's or coincidentally it connects to a different AP when it reboots. Additionally, it appears more unstable when both outputs are enabled. Output 1 is 37W and Output 2 is 47W, both are LED lights.

The shelly devices are connected to a specific 2.4Ghz only SSID. The shelly itself is downstairs and pretty close to the downstairs AP, when first powered up it connects to the downstairs AP but eventually I see it reboot, and connect to the upstairs AP, 2 floors away and sit with a poor RSSI.

rojer, 2021-12-28:
can you check `/debug/core` on the crashing device? if there's any output, please send it to rojer@rojer.me

ya-kay, 2022-01-10:
I have the same issue for one of four devices, it crashes, when the input turns off. After the crash the wifi is disabled and the device is in the ap mode. After connecting via it´s own network and enabling wifi again, it's back online until I turn off the light switch.

rojer, 2022-01-10:
ok, let's get to the bottom of it.
first, let's try rolling forward - please try the 2.11 pre-relase version (#868). there's one change that i think might improve stability.
second, can you remember which version worked for you before this started happening (if any)?

flying-stefan, 2022-01-25:
Hello, just updated one of my Shellys from 2.9.1 to latest 2.10.1. 
Immediately after the update, I could reach the Shelly via its IP address. But it was no longer reachable by the HomeKit app. 
I tried the reset (switching 5 times after power interruption). 
Now I nether can reach the shelly via its IP address nor HomeKit. It’s even hard to turn on lights by the hard switch. So it’s somehow messed up because I cannot reach the Shelly for downgrading it again. Any connection is impossible. 
Any ideas? Thanks a lot! 
Stefan

flying-stefan, 2022-01-25:
> Hello, just updated one of my Shellys from 2.9.1 to latest 2.10.1.
> Immediately after the update, I could reach the Shelly via its IP address. But it was no longer reachable by the HomeKit app.
> I [...]

Don't ask me how, but after trying to reset again and again, I managed to see the surface of the shelly again. 
I was able to load 2.9.1 again. 
Everything works fine just as before. Now I'm afraid of upgrading again. Any suggestions? 
Kind regards, Stefan

rojer, 2022-01-26:
@flying-stefan
this is recurring theme but i can't quite put my finger on it. none of my devices have any issues.
i'd very much appreciate if someone can work with me on this, i have a number of things to try but we need to establish a reliable test/repro loop between us.
@flying-stefan what model is your shelly? is it physically accessible? unless it's Shelly1, it has a button, you have [failsafe mode](https://github.com/mongoose-os-apps/shelly-homekit/wiki/Recovery#from-crash-looping-or-otherwise-misbehaving-firmware).
if device is even barely conscious, it will allow you to reflash and recover (note that it wipes all the settings too).
if you have some time, please put 2.10.1 on it again and see if the issue is still there.
if yes, try latest 2.11 alpha (https://github.com/mongoose-os-apps/shelly-homekit/issues/868) and see it it's any better.

closed 2022-03-05 by github-actions[bot]

sloncek, 2022-08-24:
hm, there was no solution for this :( but I am still seeing the issue when trying to turn on mu lights... LED lights with a bypass. I have physical access and I have tried to revert to 2.9.x, where I have the same issue.