mongoose-os-apps/shelly-homekit #879: Wrong shutter direction with shelly2.5 in single input mode

lucaspinelli85, 2022-01-03:
I have configured my shutters with single button (input mode: single) which does the function open, stop, close, stop.
It often happens that the roller shutter does not go in the correct direction when interacting both from the button and from the app, this happens:
- I press in the app and the shutter goes up
- I press in the app again to stop it
- now I press the button, I expect the shutter to lower, because the last action was to go up ... **and instead it gets up again, forcing me to press again to stop it and press again to lower it !!!**
What I understood is that the shelly does the opposite action to the previous one, but with respect to how I commanded it. Example: the roller shutter is lowered via a button if the action before the button was to go up, regardless of what I have done in the meantime with the app ...
In conclusion: I would like the roller shutter, regardless of how it was controlled, to always and in any case perform the opposite action to the previous one.
I hope I have explained myself well and that the translation is understood!
Thank you all

rojer, 2022-01-03:
yes, it's clear. you are right, we keep last action for external input and HAP separate. what you say sounds reasonable, and i don't remember why we track them separately... i'll make the change.

lucaspinelli85, 2022-01-03:
thank you so much

closed 2022-01-04 by rojer

rojer, 2022-01-05:
released in alpha5, please test

lucaspinelli85, 2022-01-05:
> released in alpha5, please test

Test it! Work perfectly now!
Thanks!

lucaspinelli85, 2022-01-05:
Thank you @rojer, great job, you are the best!