mongoose-os-apps/shelly-homekit #880: Dump characteristics and additional infos using http?

markus-fischbacher, 2022-01-04:
Hi there,

is it possible to dump supported characteristics and additional infos for HomeKit loaded Shelly devices?
Something like http://192.168.1.10:9000/accessories
Which headers are necessary to be set and with which values?

timoschilling, 2022-01-04:
The URL you are posted is from your private network, we can't read that.

As far as I know it's not possible with our firmware to access the HomeKit / HAP interface without peering authentication.

Maybe this is helps you https://github.com/mongoose-os-apps/shelly-homekit/wiki/Development#API

rojer, 2022-01-05:
yes, not possible to directly enumerate HAP characteristics but we have the JSON RPC API.

closed 2022-01-05 by rojer