mongoose-os-apps/shelly-homekit #883: Shelly 2.5 PIR Mode

Marcusg562, 2022-01-06:
Hi all,

perhaps is this mode already implemented but I would like to know if it is possible to add a PIR Mode for the Shelly 2.5

Should looks like that:
```
CASE
IF CASE SW01 = ON && L01 = OFF 
THEN USE SW01 FOR 01
IF CASE SW01=OFF && L01=ON
THEN USE L01 FOR 01
IF CASE SW01 = ON && DELAYED START L01=ON
THEN IGNORE SW01 && USE L01 FOR 01
IF CASE DELAYED START SW01 = ON && L01=ON
THEN IGNORE SW01 && USE L01 FOR 01
IF CASE SW01 = OFF && L01 = OFF
THEN 01 = OFF
```

I would like to always overwrite the Input of SW01 with HomeKit and prevent the Shelly from getting too much output on 01 and to prevent the lights turning on when the PIR turns off.

Why:
HomeKit Camera used as PIR and a HW PIR attached to Shelly in case the Camera PIR will be faulty.

Perhaps this idea makes sense. If it is already implemented pls tell me which switch setting I should use.

Thanks in advance!
Marcus

timoschilling, 2022-01-06:
That logic is to specific to implement it, in a project like this.

What you can do is, wire your PIR to the 2.5 input and configure the input as motion sensor. Then you can implement your logic as HomeKit automatisation.

closed 2022-01-06 by timoschilling

Marcusg562, 2022-01-06:
I don‘t think that this is too specific compared to the other ones. But I also didn‘t got your suggestion. If I wire the PIR as Input the Shelly will only be usable in case the PIR recognized someone.

timoschilling, 2022-01-06:
> I don‘t think that this is too specific compared to the other ones.

At the moment we don't have any logic for a motion sensor.

> If I wire the PIR as Input, the Shelly will only be usable in case the PIR recognized someone.

Please read the following wiki sections:
- "When "Detached" mode is enabled you should see a new section called "Input 1", you will see this new section for each switch you enable "Detached" mode for. To use "Detached" mode we recommend using a Momentary type physical switch."
- https://github.com/mongoose-os-apps/shelly-homekit/wiki/Input-Switch-Settings#motion-sensor--occupancy-sensor--contact-sensor--doorbell

Then you can control the output of the shelly via HomeKit and the input is represented as motion sensor in HomeKit. Then you can make a automatisation to control the output with the logic you need.

> the PIR will not be functional in case that my WIFI is down

Yes, but that's how HomeKit works.

rojer, 2022-01-06:
i agree, this is too narrow and specific to be included in the firmware by default. you are welcome to implement it and build a custom firmware for yourself.