mongoose-os-apps/shelly-homekit #884: RGBW2 problem with 2.11.0-alpha6

timoschilling, 2022-01-06:
I noticed three problems with the RGBW2, I opened a separate issue to not mess up #868 

Situation: color=green or blue, brightness=1%, saturation=100%
Observations:
1. Web UI shows 1%, LED looks like 1%, but HomeKit shows 100%. If I change the value the LED switched to the new value and HomeKit shows the new value. After some sec HomeKit switches back to 100% and LED keep at new value.
2. After some minutes the LED starts flickering.
3. After some more time the shelly becomes unavailable via Web UI and HomeKit, LEDs are still on.

All three things happened on three different Shelly's after updating from alpha4 to alpha6, and gone since I reverted. So I assume that's a software issue and not a hardware issue.

Maybe it has to do with this change https://github.com/mongoose-os-apps/shelly-homekit/commit/fdeac316a6baf2c2010fefff3c63c8c23d60d986

rojer, 2022-01-06:
sorry about that, probably some miscasting of the value. i don't actually have RGBW2 on hand to test with but will review the change.

rojer, 2022-01-06:
ooh, i see. std::bind copies reference arguments, so my idea to use references for the helpers was not great. mea culpa, fix incoming.

rojer, 2022-01-06:
can you build firmware from master and try now?
i'm also curious about the flickering and dying part, that would not be explained by this bug. can you monitor device logs for unusual activity?

timoschilling, 2022-01-06:
1. HomeKit Control works as expected again 👍 
2. It is really scary, the flickering starts ~10 sec after reboot or power connected. And it can be stopped by open the Web UI, closing the Web UI or changing the brightness via HomeKit. See the attached movies.
3. not happened again, will continue to observing

```
5974751 shelly_debug.cpp:231    No log file, sending new entries
5980823 mgos_mongoose.c:66      New heap free LWM: 23312
5991806 HAPPlatformTCPStrea:121 0x3fff5804 10.0.0.178:55514 Incoming HAP connection, ns 2/1/12
6162820 mgos_mongoose.c:66      New heap free LWM: 22168
6174354 HAPPlatformTCPStrea:121 0x3fff5944 10.0.0.188:51783 Incoming HAP connection, ns 3/1/12
6187918 mgos_mongoose.c:66      New heap free LWM: 21936
6318847 HAPPlatformTCPStrea:276 0x3fff5a54 10.0.0.110:49208 Accepted HAP connection, ns 3/2/12 ts 0x3fff5b64
7473307 mgos_mongoose.c:66      New heap free LWM: 21544
7496836 mgos_mongoose.c:66      New heap free LWM: 21352
7773578 mgos_mongoose.c:66      New heap free LWM: 21040
7828628 HAPPlatformTCPStrea:276 0x3fff5adc 10.0.0.134:53786 Accepted HAP connection, ns 2/3/12 ts 0x3fff57ac
8993366 mgos_mongoose.c:66      New heap free LWM: 20776
9128630 mgos_sys_config.c:323   Saved to conf9.json
9140831 shelly_main.cpp:534     Up 9.13, HAP 2/3/12 ns 3, RAM: 22664/42160; st 0; 11.1: sta: on, b: 1, h: 120, sa: 100
9146260 mgos_mongoose.c:66      New heap free LWM: 20128
9350678 HAPPlatformTCPStrea:276 0x3fff5804 10.0.0.178:55514 Accepted HAP connection, ns 1/4/12 ts 0x3fff50d4
10516932 esp_main.c:137          SDK: LmacRxBlk:1
10758726 esp_main.c:137          SDK: pm open,type:1 0
10860501 HAPPlatformTCPStrea:276 0x3fff5944 10.0.0.188:51783 Accepted HAP connection, ns 0/5/12 ts 0x3fff4de4
12618052 HAPPlatformTCPStrea:121 0x3fff640c 10.0.0.109:50381 Incoming HAP connection, ns 0/5/12
12627591 HAPPlatformTCPStrea:276 0x3fff640c 10.0.0.109:50381 Accepted HAP connection, ns 0/6/12 ts 0x3fff6494
13817969 HAPPlatformTCPStrea:121 0x3fff5d04 10.0.0.139:49835 Incoming HAP connection, ns 0/6/12
13833567 HAPPlatformTCPStrea:105 0x3fff5a54 10.0.0.110:49208 HAP connection closed, ns 1/6/12 i 2
13841220 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff5b64 nc 0x0 1/5/12
14138242 HAPPlatformTCPStrea:276 0x3fff5d04 10.0.0.139:49835 Accepted HAP connection, ns 0/6/12 ts 0x3fff63b4
15349921 HAPPlatformTCPStrea:105 0x3fff5d04 10.0.0.139:49835 HAP connection closed, ns 0/6/12 i 2
15357579 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff63b4 nc 0x0 0/5/12
15385234 HAPPlatformTCPStrea:121 0x3fff59cc 10.0.0.185:52437 Incoming HAP connection, ns 0/5/12
15651468 HAPPlatformTCPStrea:276 0x3fff59cc 10.0.0.185:52437 Accepted HAP connection, ns 0/6/12 ts 0x3fff4e0c
17336522 HAPPlatformTCPStrea:105 0x3fff5adc 10.0.0.134:53786 HAP connection closed, ns 0/6/12 i 2
17344185 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff57ac nc 0x0 0/5/12
17545389 shelly_main.cpp:534     Up 17.53, HAP 0/5/12 ns 5, RAM: 24508/42160; st 0; 11.1: sta: on, b: 1, h: 120, sa: 100
18031187 HAPPlatformTCPStrea:121 0x3fff632c 10.0.0.139:49836 Incoming HAP connection, ns 0/5/12
18040706 HAPPlatformTCPStrea:276 0x3fff632c 10.0.0.139:49836 Accepted HAP connection, ns 0/6/12 ts 0x3fff4eec
25545245 shelly_main.cpp:534     Up 25.53, HAP 0/6/12 ns 6, RAM: 25160/42160; st 0; 11.1: sta: on, b: 1, h: 120, sa: 100
33545261 shelly_main.cpp:534     Up 33.53, HAP 0/6/12 ns 6, RAM: 25160/42160; st 0; 11.1: sta: on, b: 1, h: 120, sa: 100
41545380 shelly_main.cpp:534     Up 41.53, HAP 0/6/12 ns 6, RAM: 25160/42160; st 0; 11.1: sta: on, b: 1, h: 120, sa: 100
49545320 shelly_main.cpp:534     Up 49.53, HAP 0/6/12 ns 6, RAM: 25160/42160; st 0; 11.1: sta: on, b: 1, h: 120, sa: 100
57545239 shelly_main.cpp:534     Up 57.53, HAP 0/6/12 ns 6, RAM: 25160/42160; st 0; 11.1: sta: on, b: 1, h: 120, sa: 100
65545205 shelly_main.cpp:534     Up 65.53, HAP 0/6/12 ns 6, RAM: 25160/42160; st 0; 11.1: sta: on, b: 1, h: 120, sa: 100
73545525 shelly_main.cpp:534     Up 73.54, HAP 0/6/12 ns 6, RAM: 25160/42160; st 0; 11.1: sta: on, b: 1, h: 120, sa: 100
81545292 shelly_main.cpp:534     Up 81.53, HAP 0/6/12 ns 6, RAM: 25160/42160; st 0; 11.1: sta: on, b: 1, h: 120, sa: 100
89545271 shelly_main.cpp:534     Up 89.53, HAP 0/6/12 ns 6, RAM: 25160/42160; st 0; 11.1: sta: on, b: 1, h: 120, sa: 100
97545273 shelly_main.cpp:534     Up 97.53, HAP 0/6/12 ns 6, RAM: 25160/42160; st 0; 11.1: sta: on, b: 1, h: 120, sa: 100
105545399 shelly_main.cpp:534     Up 105.53, HAP 0/6/12 ns 6, RAM: 25160/42160; st 0; 11.1: sta: on, b: 1, h: 120, sa: 100
113545275 shelly_main.cpp:534     Up 113.53, HAP 0/6/12 ns 6, RAM: 25160/42160; st 0; 11.1: sta: on, b: 1, h: 120, sa: 100
121545355 shelly_main.cpp:534     Up 121.53, HAP 0/6/12 ns 6, RAM: 25160/42160; st 0; 11.1: sta: on, b: 1, h: 120, sa: 100
129545309 shelly_main.cpp:534     Up 129.53, HAP 0/6/12 ns 6, RAM: 25160/42160; st 0; 11.1: sta: on, b: 1, h: 120, sa: 100
137545456 shelly_main.cpp:534     Up 137.53, HAP 0/6/12 ns 6, RAM: 25160/42160; st 0; 11.1: sta: on, b: 1, h: 120, sa: 100
140869024 mgos_http_server.c:180  0x3fff5f7c HTTP connection from 10.0.0.192:50803
140888614 mg_rpc.c:310            Shelly.GetInfo via WS_in 10.0.0.192:50803
140907371 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 10.0.0.192:50803
140926243 mgos_mongoose.c:66      New heap free LWM: 19208
141817312 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 10.0.0.192:50803
[... cut ...]
175183297 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff50d4 nc 0x3fff5804 0/2/12
175192713 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff4de4 nc 0x3fff5944 0/1/12
175200639 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff6494 nc 0x3fff640c 0/0/12
175213998 HAPPlatformTCPStrea:105 0x3fff445c 0.0.0.0:9000 HAP connection closed, ns 0/0/12 i 1
175221682 shelly_main.cpp:357     HAP server state: 0
175226614 shelly_main.cpp:339     === Destroying accessories
```

Need a moment to add the movies

timoschilling, 2022-01-06:
Sorry movies are too big for direct upload to GitHub, so let me know if the iCloud link is a problem then I will search for a other solution.
https://share.icloud.com/photos/044lLRt_ja29R5TFNhf7kx1RA

rojer, 2022-01-06:
ok, the only remotely relevant change i can think of is the wifi sleep mode - https://github.com/mongoose-os-apps/shelly-homekit/commit/dc0e7b454cac2f70fd31cd4103601e45acf39e23 maybe it changes something in how interrupts / pwm work?
can you set it back to 0 and see if that fixes it?

timoschilling, 2022-01-06:
Yes setting `wifi.sta_ps_mode=0` solves the flickering problem!

Maybe you it's enough to set it for the rgbw2 to `0`.

rojer, 2022-01-06:
interesting, i wonder how exactly putting wifi to sleep affects timer interrupt processing.
would need to investigate, meanwhile i've reverted the commit.

rojer, 2022-01-06:
pushed alpha7

closed 2022-01-06 by rojer

timoschilling, 2022-01-10:
@rojer little appendix: I have noticed a little flickering on the LED spots behind a 1L, it starts after flashing the alpha6, first I have assumed a problem with the spots, but since flashing alpha7 they are no longer flickering. Maybe it's a coincidence, maybe it's relevant. I just want to leave it as a note.

jasonsiwi81, 2022-11-04:
hi，i did update homekit firmware by script but i get same issue. flickering and wifi changed to AP mode. any suggest to flash to alpha 7？ thanks