mongoose-os-apps/shelly-homekit #886: “Open/Close the roller shutter” not working

hertor, 2022-01-07:
Hi, when I ask Siri to open/close the roller shutter, Siri ask me for a percentage to work. I thought that saying directly “open” or “close”, Siri would assume 100% or 0% respectively. Is it a bug? Thank you,

rojer, 2022-01-07:
firmware has no control over that.

timoschilling, 2022-01-07:
It’s right was Rojer says. 

That could be a bug in Siri, normal open and close should work as command. 

Are you using Siri in english or in a different language? Does your language maybe has different words for close? Maybe try up and down or high and low.

hertor, 2022-01-08:
Thanks all for your answers. My idiom is Spanish. I tested with “up” and “down” and the roller shutter up/down only 1% and stops. Weird.
Best regards!

timoschilling, 2022-01-08:
Open the log and then say Siri to open, maybe then you can see something

hertor, 2022-01-08:
> Open the log and then say Siri to open, maybe then you can see something

Ok, I will check that. Thank you.

Other issue I saw: I have two physical buttons for the roller shutter. The input mode is configured as “separate toggle”. The test is the following: Up the roller shutter with the physical button or HomeKit. When it reaches the top, it stops and then after 2 or 3 seconds a “click” sound can be heard (I suppose is the relay). Until it sounds, you can not down the roller shutter with the physical button. I tested it with the original Shelly firmware and happens the same but very quickly: the “click” sound happens immediately when the roller shutter reaches the top and so, yo can down it with the physical button without any wait. 
According to this, it seems a (very) little SW issue, but I wanted to share this with you. 

Thank you all! Any suggestion/clarification is very appreciated.

timoschilling, 2022-01-08:
Try to recalibrate

hertor, 2022-01-08:
Ok, but the “click” sound is something normal, isn’t?

hertor, 2022-01-09:
> Open the log and then say Siri to open, maybe then you can see something

Hi, this is the log when I ask Siri to “down the roller shutter”. As you can see, it only moves a 1%.

And please, could you confirm me that Shelly produces a soft/quiet sound (a “click”) when the roller shutter reaches the top an stops?

Thank you in advance!

[image]

timoschilling, 2022-01-09:
The click can come from the Shelly when the relay state changes. 

If I read the log right the cover is already nearly down at 93% and then moves to 94.5%. 

Maybe you wired the Shelly outputs wrong. 
you could try to say Siri, open the WC when it’s open, so the result could be that a close is performed. 

@rojer can you have a look here, I don’t own a WC so I can’t give a deeper here.

hertor, 2022-01-10:
> The click can come from the Shelly when the relay state changes.
>
> If I read the log right the cover is already nearly down at 93% and then moves to 94.5%.
>
> Maybe you wired the Shelly outputs wrong. [...]

Thanks @timoschilling. The direction is the right one, maybe I said Up an not Down. The point is that only it is moved a 1% when I say “open/close”.

Best regards!

Xaruqui, 2022-01-11:
Hola Hertor. This has an easy fix, create a shortcut in iOS that rolls it up completely and name it 'open the roller shutter'  so every time you say it, it will launch the shortcut and it'll open at 100%.

Regards

hertor, 2022-01-11:
> Hola Hertor. This has an easy fix, create a shortcut in iOS that rolls it up completely and name it 'open the roller shutter' so every time you say it, it will launch the shortcut and it'll open at 100%.
> 
> Regards

Thank you, although I already did that workaround and works fine as you says. The point is we shouldn’t create a shortcut to achieve it.
Regards!