mongoose-os-apps/shelly-homekit #887: Power Recliner Sofa

andyblac, 2022-01-08:
I have a new power recliner sofa due in feb/march, I was just wondering if anyone have tried fitting some Shelly 2.5's to them one for each side for up / down, was thinking of using roller shade mode, and calibrate as the sofa should have auto stop at each end point, then only thing would be normally you press and hold to do movement, and release when you want to stop moving, but I assume for roller you press and release to start movement, and press again to stop.

Would it be possible to add an option for press and hold for movement, and on release stop, and double press would act as normal ie go to end point?

would like people thoughts on this.

bmpalmeida, 2022-01-08:
I have exactly the same situation.

In my case i have an additional situation, is the the motors are DC, for what i understand there is no possibility to calibrate DC motors, is that correct?

lucaspinelli85, 2022-01-09:
I have DC motor in sofà with Shelly2.5 and i calibrate it with apposite scheme

lucaspinelli85, 2022-01-09:
Using 3PDT relay [image]
Or DPDT relay and 2 power[image]
Or SPDT + DPDT[image]
Is it use fil for you?

andyblac, 2022-01-09:
can we not power the shelly 2.5 simply via DC ?, also your diagram does not use the manual switch from the sofa itself.

lucaspinelli85, 2022-01-09:
I don't think the DC powered shelly works in roller shutter mode, because it measures the power for calibration. I don't know, but in my case I had to power it in AC because my motors are in 12V so I had no choice.
for the manual switch simply connect the single button to SW1 (single mode) or the double button to SW1 and SW2.

andyblac, 2022-01-09:
> I don't think the DC powered shelly works in roller shutter mode, because it measures the power for calibration. I don't know, but in my case I had to power it in AC because my motors are in 12V so [...]

ok, but what about the press and hold manual button?

lucaspinelli85, 2022-01-09:
> can we not power the shelly 2.5 simply via DC ?, also your diagram does not use the manual switch from the sofa itself.

the concept of the scheme is always the same, but I don't know if the shelly can do the calibration powered in DC. I have never tried and I don't know if it works, otherwise you should use timed manual calibration![image]

lucaspinelli85, 2022-01-09:
> ok, but what about the press and hold manual button?

Currently I press the manual button to start the movement, and then re-start to stop it in the desired position.
I did not understand your need well.
But if you want it to only move when you press then just set to toogle mode and use a physical button!

andyblac, 2022-01-17:
@lucaspinelli85 thanks I'll give it a try once my new sofa arrives.

closed 2022-01-17 by andyblac