mongoose-os-apps/shelly-homekit #888: Feature request: Adding Apples Homekey to Shelly1 door lock

dweiden, 2022-01-08:
Hey there, 
as I’m using a Shelly 1 as door opener and as the CES passed by, where Home key was a thing, I was wondering if there is any chance to implement the Home key function and use any NFC Sticker as a trigger. That would be a great addition, if this is possible and worth the work. 

Thank you and best regards, 
Denis

rojer, 2022-01-08:
i'm not sure what is required of an accessory so apple recognizes it as a "compatible lock".
we'll need some technical details on how to actually implement it.

dweiden, 2022-01-09:
Thank you for your answer. Was already thinking about that. However it would be great if this could work in the future. I appreciate your work!

timoschilling, 2022-01-10:
tl;dr: I expect that it will never work!

First I will give a short understanding about how HomeKey works. @dweiden suggested we could use a NFC sticker as trigger. That will not work. HomeKit didn't work like a NFC tag triggers a process and the iPhone unlocks the lock via HomeKit. HomeKit works more like a Apple Pay transaction, the payment terminal is the lock and on the end the door opens instead of a money transfer. 

I see the following problems:
1. The ESP8266 is to slow for that because you need a fest encryption. As far as I know the communication is time sensitive to prevent signal tunnelling. So the ESP will not be able to work.
2. I expect you need certificates signed by Apple, because the iPhone creates the certificate for the NFC communication.
3. At the moment there is no public documentation about the process / protocol available and I expect that will not change.

@rojer from my point of view we should close this issue as not possible from now.

rojer, 2022-01-10:
sounds good. closing as infeasible until proven otherwise.

closed 2022-01-10 by rojer