mongoose-os-apps/shelly-homekit #889: Physical switch not responding when flashed, works fine with stock firmware

lanceuppercut47, 2022-01-09:
With the stock firmware I’m able to control the lights with the Shelly app but importantly, the physical light switch.

I’ve flashed my 1L and I’m able to control the lights within Apple Home, but my physical switch no longer works.

Tested rolling back to stock, physical switch works, back to this and it stops working again.

timoschilling, 2022-01-09:
Have you tried the different input modes?

lanceuppercut47, 2022-01-09:
I’ve tried the default toggle and also edge, both with no luck.

timoschilling, 2022-01-09:
Have you tried Momentary? What kind of switch do you have?

lanceuppercut47, 2022-01-09:
I tried all the options in the end and edge shared or whatever it was called, worked even though just “edge” didn’t.

closed 2022-01-09 by lanceuppercut47