mongoose-os-apps/shelly-homekit #898: can't update firmware to homekit shelly 1l

AmirHen, 2022-01-15:
Hi,
i need help please..

trying yo upgrade to HK 
with this link: http://A.B.C.D/ota?url=http://rojer.me/files/shelly/shelly-homekit-Shelly1L.zip
but i see that the new verion and ths old version are the same....
"old_version":"20211109-125044/v1.11.7-g682a0db"

i read on https://github.com/mongoose-os-apps/shelly-homekit/issues/323 
to try and load beta 2.7.0 with this link:
http://a.b.c.d/ota?url=http://rojer.me/files/shelly/2.7.0-beta2/shelly-homekit-Shelly1L.zip
but its the same...
nothing happend and i can not switch to HomeKit

andyblac, 2022-01-17:
why 2.7.0 that really old the latest beta is `2.11.0-alpha8`, the latest current stable firmware is `2.10.1`

AmirHen, 2022-01-17:
thanks..
i tried now with 2.11.0.-alpha8 and its the same..
"{"status":"updating","has_update":false,"new_version":"20211109-125044/v1.11.7-g682a0db","old_version":"20211109-125044/v1.11.7-g682a0db"}"

but nothing happend..
yes i replace a.b.c.d to the currnet IP.. if not i couldn't fo inside shelly web interface right?

AmirHen, 2022-01-17:
after i set Static IP its work!

closed 2022-01-17 by AmirHen