mongoose-os-apps/shelly-homekit #899: WiFi 6 (AX)

RichJonk, 2022-01-17:
The firmware on my Shelly Plug S has issues with WiFi 6 (AX), the official firmware works very well.

I am using ASUS Zenwifi AX.

andyblac, 2022-01-17:
make sure you split 2.4ghz and 5Ghz, do not combine them, I have WiFi6 here and have no issues with this firmware.

timoschilling, 2022-01-17:
I see it different then Andy, but I agree with him that is not an firmware/wifi6 issue. 

@rojer what do you think?

RichJonk, 2022-01-18:
Yeah, I tried a lot but WiFi 6 works but only when Tri Band Smart Connect is turned of.

When it is turned on the devices are having lots of problems. The device with the Shelly firmware works with that setting on so the Shelly Plug S can work with Tri Band Smart Connect.

I can provide you with logs, not sure what logfiles you need.

[image]

trendchiller, 2022-02-04:
AX works here too - without any problems - Handsteuerung is activated…

closed 2023-01-21 by github-actions[bot]