mongoose-os-apps/shelly-homekit #902: Replacement switch app setting

LiranETN, 2022-01-18:
Hi all,
Need assistance with configuring the app. 
I am using shelly 2.5 with 2 Replacement switch that are wired correctly but some configuration makes it endless blinking what is the correct config for this scenario?
Thanks,

timoschilling, 2022-01-18:
Sorry I don't understand your question. What app are you talking about? What is exactly blinking? You should explain more what your scenario is.

LiranETN, 2022-01-18:
Sure. 
Shelly 2.5 wire connection:
[image]

This is how Shelly is configured via web connection to the app:
[image]

In this scenario, the light is blinking. I think that something needs to be configured correctly so that Shelly know that this is a replacement switch and not a regular switch.

I hope this more clear.

timoschilling, 2022-01-18:
not sure how your switches are working, have you tried edge or momentary?

LiranETN, 2022-01-18:
Each switch can turn on/off the current light state. 
I am not sure what the correct config should be so this is why I post here to see if someone knows.

LiranETN, 2022-01-18:
No. Should I set both switch the same (both edge / both momentary)

timoschilling, 2022-01-18:
only that one that is wired

Adi-Kha, 2022-01-18:
This should work in your case !

[image]

LiranETN, 2022-01-18:
OK Thanks. will check and update

LiranETN, 2022-01-18:
Thanks all works

closed 2022-01-18 by timoschilling