mongoose-os-apps/shelly-homekit #904: BUG: Unresponsive 1L

stevelw, 2022-01-21:
I have a 1L fitted to a light switch. All working well until I turned off the mains to do some unrelated work fitting a new 1L elsewhere. 

Now the old 1L is dead. Physical connections still good. Was set to detached so can't test relay response. Doesn't connect to WiFi and reset procedure appears to do nothing.

stevelw, 2022-01-21:
Should add: does not feel hot so not a heat protection issue.

timoschilling, 2022-01-21:
So you can't see the shelly1l-xxxx wifi?

stevelw, 2022-01-21:
No. ☹️

stevelw, 2022-01-21:
Am I understanding correctly that the reset procedure is identical to the stock — power off then on, toggle 10x (on then off at switch)?

timoschilling, 2022-01-21:
There are different ways to reset, I think if you have the 1l already out off the wall, the it would be the easiest to push the small button on the back for 10-20 sec.

> should I then hear the relay switch rapidly like a stock reset?

no

stevelw, 2022-01-21:
I don't think the 1L has a physical reset — I'll double check…

stevelw, 2022-01-21:
Having left it alone for an hour it's now working as normal again… and thankfully non of my reset attempts did anything. 

Is there a way to get logs to find out what happened?

timoschilling, 2022-01-21:
Filesystem logging is disabled, otherwise it will kill the flash over time. If you have the problem more often, you can enable it.

closed 2022-01-21 by timoschilling

stevelw, 2022-01-21:
I wanna take this opportunity to thank all the devs. This firmware was super easy to use and does exactly what I need. 👍