mongoose-os-apps/shelly-homekit #906: Shelly 1L shows locked ??

sterl1i, 2022-01-21:
I’m new here. I have 2 Shelly 1L. One shows it’s locked and the other isn’t. I don’t know how I locked it. I know how to unlock it but I don’t know what to enter (user name and pw). Since I didn’t lock it, I don’t know what was used as values. Has anyone seen this or have any suggestions for unlocking permanently?

The locked one doesn’t have cloud access and I can’t change it while locked. The unlocked one does have cloud access.

[image]

timoschilling, 2022-01-21:
From what is that screenshot? Are you sure that you using our firmware or do you use the original firmware?

timoschilling, 2022-01-21:
I expect that you are using the original firmware, because we don’t have something like cloud access. If that’s the case you are wrong here.

sterl1i, 2022-01-21:
No, I am using Mongoose. That was a screen capture from Shelly Home.

Shelly Home is a cross platform app created by the software developer Dirk Gausmann. It is developed in his free time and aims to bring a better user experience to Shelly users

Shelly Home is available on Android, iOS, MacOS and Windows. You can add, manage, update and control your Shelly devices aswell as highly customize how and what information you want to see.
The app also does work without a user account and can directly access the Shelly devices located in the same network. Optionally the user can connect to its Shelly Cloud account, to be able to control the devices from anywhere.

Lesley
314-962-1499
***@***.***

Animals are such agreeable friends—they ask no 
questions, they pass no criticisms.

closed 2022-01-21 by timoschilling