mongoose-os-apps/shelly-homekit #909: Roller shutter

jorgerray, 2022-01-23:
Hi! 
My roller is closed when Home says open. What can I do?
2.10.1
Thanks

timoschilling, 2022-01-23:
What happens when you close it in the Home App?

romg13, 2022-01-24:
hello, 

I have exactly the same problem since the 2.10.1 update 
I found a solution to open the roller 
you just need to open a browser like safari, enter the ip address of your roller and clic to calibrate, you can also use the open or close button in this webpage 

I think there is a bug in this update, I use the shelly 2.5 since 2 years and I never had problems, my problem began with this 2.10.1 update 

when you press the "open" button on Homekit, nothing happens and the button remains with the "open" status while the shutter is closed.
This is the second time this has happened to me since the update.

timoschilling, 2022-01-24:
sounds like a double of #844

rojer, 2022-01-24:
may or may not be the same issue, let's triage.
@romg13 how often does this happen? and when it is incorrect, can you open the debug log tab in the web ui, leave it open for ~10 seconds (so at least one status update is logged), then try to control from home app. post the log here, i want to see it.

romg13, 2022-01-25:
hello, this happened 2 times (within a week) since the update to 2.10.1 and never before for 2 years. I have to wait for it to happen again to send you the log.
when the bug occurs, the roller shutter is in the closed position, but Homekit sends me an "open" information, when I click on the button, it remains on "open" while the roller shutter remains closed.

as soon as it happens again i will send you the log

jorgerray, 2022-01-25:
Just the same

closed 2022-03-04 by github-actions[bot]