mongoose-os-apps/shelly-homekit #913: Shelly 1 becomes unresponsive and reboot every couple of hours or so

Rotothor, 2022-01-26:
Hi, I have a mongoose shelly one hooked up to a light fitting rebooting every few hours. 
Wifi signal is rock solid it is pretty close to the mesh AP,  and I had no recent wifi outage. Can anyone help me root cause ? I'm attaching a core_dump 

Thanks ! 
[core_dump.txt](https://github.com/mongoose-os-apps/shelly-homekit/files/7940735/core_dump.txt)

timoschilling, 2022-01-26:
If you have a core dump mail them to rojer@rojer.me with a link to this issue.

rojer, 2022-01-26:
hm, it's a real head-scratcher. the stack trace looks completely normal - device recently turned on an output (last log message is "shelly_output.cpp:63    Output 1: off -> on (HAP)") and is busy encrypting a response to the controller.
and the crash happens in a middle of an innocent crypto operation, at instructions that do not even involve memory and are just bit operations on registers:

   0x4028f112 <+70>:    slli    a2, a2, 24
   0x4028f115 <+73>:    or      a2, a2, a4

program counter is 0x4028f113, which is suspicious - it falls between instructions. it's not captured in the dump but i suspect the exception is invalid instruction.
so, i actually blame hardware, or more likely voltage spike or some other disturbance associated with turning on the load. it screws up the CPU state and things go awry.

mtzaquia, 2022-01-27:
Hello there!

First of all, thank you for the awesome piece of firmware here. I have been using it and, all-in-all, it helped me getting into the world of DIY automation!

To add to the issue, I have been facing something similar, but on a longer period (days instead of hours). The device unfortunately does not reboot on its own, either (that would actually "auto-fix" the issue, I believe).

Every 10 days or so, **Shelly1** will go unresponsive (requests to it will time out). After a power cycle, it goes back to normal. Could it be some sort of memory leak creeping up over a period of time and eventually maxing out memory?

For awareness, I activate the switch mostly with the RPC API:
http:/<server>/rpc/Shelly.SetState?id=1&type=0&state=%7b%22state%22%3atrue%7d

closed 2022-03-06 by github-actions[bot]