mongoose-os-apps/shelly-homekit #914: WiFi Signal Strength Figure

andyblac, 2022-01-26:
does anyone have issue with the WiFI signal strength in the WEB UI, being wrong by a lot ?

Router UI
[image]

Device WEB UI
[image]

timoschilling, 2022-01-26:
What does it shows for your Mac and what shows your Mac (option click on the wifi icon in the menu bar)?
I would expect that the results are always different, because they are showing the signal strange and that is depending more on the own antenna then on the sensing antenna.

andyblac, 2022-01-26:
> What does it shows for your Mac and what shows your Mac (option click on the wifi icon in the menu bar)?
> I would expect that the results are always different, because they are showing the signal strange and that is depending more on the own antenna then on the sensing antenna.

both the same 
[image]
[image]

markirb, 2022-01-26:
RSSI is by no means a defined and thus comparable measure - see e.g. Wikipedia for reference - could it be the case that you are trying to compare apples and oranges here?

rojer, 2022-01-26:
unfortunately, we have zero visibility into how this number is computed, it's handed to us by the closed Espressif SDK.
as long as it seems sane, i.e. no obviously garbage values are displayed, then it means the value is making it alright from the SDK, but that's about all we know.

andyblac, 2022-01-26:
@rojer can make available in the UI the option to control the RSSI value that searches for a new AP then, with an option to disable it, as sometime I see it trying to find a better AP, when there is no need as the value on my router Is bellow -45