mongoose-os-apps/shelly-homekit #920: Roller shutter don't move

lucaspinelli85, 2022-01-29:
Hi i have this problem: in HomeKit work, in webui work, but roller shutter don't move, i need to re calibrate it. why? This is log:

997332174 shelly_debug.cpp:231    No log file, sending new entries
997838506 shelly_main.cpp:513     Up 997.82, HAP 0/2/12 ns 2, RAM: 25316/42160; st 34; 3.2: svt:3 inv:0 st:0 m:0 lea: 51.431; 4.1: c:1 mp:104.30 mt_ms:20891 cp:0.00 tp:0.00 md:0 lemd:1 lhmd:0
1005838582 shelly_main.cpp:513     Up 1005.82, HAP 0/2/12 ns 2, RAM: 25316/42160; st 34; 3.2: svt:3 inv:0 st:0 m:0 lea: 59.431; 4.1: c:1 mp:104.30 mt_ms:20891 cp:0.00 tp:0.00 md:0 lemd:1 lhmd:0
1013838519 shelly_main.cpp:513     Up 1013.82, HAP 0/2/12 ns 2, RAM: 25316/42160; st 34; 3.2: svt:3 inv:0 st:0 m:0 lea: 67.431; 4.1: c:1 mp:104.30 mt_ms:20891 cp:0.00 tp:0.00 md:0 lemd:1 lhmd:0
1021838649 shelly_main.cpp:513     Up 1021.82, HAP 0/2/12 ns 2, RAM: 25056/42160; st 34; 3.2: svt:3 inv:0 st:0 m:0 lea: 75.431; 4.1: c:1 mp:104.30 mt_ms:20891 cp:0.00 tp:0.00 md:0 lemd:1 lhmd:0
1024320859 shelly_hap_window_c:370 WC 1: HAPSetTgtPos 100.00 cur 0.00 tgt 0.00 lhmd 0
1024326915 shelly_hap_window_c:347 WC 1: Tgt pos 0.00 -> 100.00 (HAP)
1024333044 shelly_hap_window_c:327 WC 1: State: idle -> move (0 -> 20)
1024376480 shelly_output.cpp:63    Output 2: off -> on (move)
1024383242 shelly_hap_window_c:327 WC 1: State: move -> rampup (20 -> 22)
1024477057 shelly_hap_window_c:549 P = 0.00 -> 104.30
1024577321 shelly_hap_window_c:549 P = 0.00 -> 104.30
1024677094 shelly_hap_window_c:549 P = 0.00 -> 104.30
1024777308 shelly_hap_window_c:549 P = 0.00 -> 104.30
1024877046 shelly_hap_window_c:549 P = 0.00 -> 104.30
1024977074 shelly_hap_window_c:549 P = 0.00 -> 104.30
1025077248 shelly_hap_window_c:549 P = 0.00 -> 104.30
1025177140 shelly_hap_window_c:549 P = 0.00 -> 104.30
1025276893 shelly_hap_window_c:549 P = 0.00 -> 104.30
1025377141 shelly_hap_window_c:549 P = 0.00 -> 104.30
1025477246 shelly_hap_window_c:549 P = 0.00 -> 104.30
1025576846 shelly_hap_window_c:549 P = 0.00 -> 104.30
1025677370 shelly_hap_window_c:549 P = 0.00 -> 104.30
1025778182 shelly_hap_window_c:549 P = 0.00 -> 104.30
1025876997 shelly_hap_window_c:549 P = 0.00 -> 104.30
1025977016 shelly_hap_window_c:549 P = 0.00 -> 104.30
1026076864 shelly_hap_window_c:549 P = 0.00 -> 104.30
1026177209 shelly_hap_window_c:549 P = 0.00 -> 104.30
1026276901 shelly_hap_window_c:549 P = 0.00 -> 104.30
1026376941 shelly_hap_window_c:549 P = 0.00 -> 104.30
1026477042 shelly_hap_window_c:549 P = 0.00 -> 104.30
1026577502 shelly_hap_window_c:549 P = 0.00 -> 104.30
1026677034 shelly_hap_window_c:549 P = 0.00 -> 104.30
1026776925 shelly_hap_window_c:549 P = 0.00 -> 104.30
1026876934 shelly_hap_window_c:549 P = 0.00 -> 104.30
1026977514 shelly_hap_window_c:549 P = 0.00 -> 104.30
1027077065 shelly_hap_window_c:549 P = 0.00 -> 104.30
1027177455 shelly_hap_window_c:549 P = 0.00 -> 104.30
1027276879 shelly_hap_window_c:549 P = 0.00 -> 104.30
1027377348 shelly_hap_window_c:549 P = 0.00 -> 104.30
1027477042 shelly_hap_window_c:549 P = 0.00 -> 104.30
1027577088 shelly_hap_window_c:549 P = 0.00 -> 104.30
1027676853 shelly_hap_window_c:549 P = 0.00 -> 104.30
1027777147 shelly_hap_window_c:549 P = 0.00 -> 104.30
1027877080 shelly_hap_window_c:549 P = 0.00 -> 104.30
1027977343 shelly_hap_window_c:549 P = 0.00 -> 104.30
1028077299 shelly_hap_window_c:549 P = 0.00 -> 104.30
1028176947 shelly_hap_window_c:549 P = 0.00 -> 104.30
1028277214 shelly_hap_window_c:549 P = 0.00 -> 104.30
1028377479 shelly_hap_window_c:549 P = 0.00 -> 104.30
1028476999 shelly_hap_window_c:549 P = 0.00 -> 104.30
1028577484 shelly_hap_window_c:549 P = 0.00 -> 104.30
1028677015 shelly_hap_window_c:549 P = 0.00 -> 104.30
1028776914 shelly_hap_window_c:549 P = 0.00 -> 104.30
1028871035 shelly_hap_window_c:370 WC 1: HAPSetTgtPos 0.00 cur 0.00 tgt 100.00 lhmd 1
1028876793 shelly_hap_window_c:549 P = 0.00 -> 104.30
1028885304 shelly_hap_window_c:549 P = 0.00 -> 104.30
1028977481 shelly_hap_window_c:549 P = 0.00 -> 104.30
1029077043 shelly_hap_window_c:549 P = 0.00 -> 104.30
1029177356 shelly_hap_window_c:549 P = 0.00 -> 104.30
1029277370 shelly_hap_window_c:549 P = 0.00 -> 104.30
1029377120 shelly_hap_window_c:549 P = 0.00 -> 104.30
1029477271 shelly_hap_window_c:549 P = 0.00 -> 104.30
1029482032 shelly_hap_window_c:556 Failed to start moving
1029488048 shelly_hap_window_c:327 WC 1: State: rampup -> stop (22 -> 24)
1029576549 shelly_output.cpp:63    Output 2: on -> off (stop)
1029606176 mgos_sys_config.c:232   Loading conf2.json
1029744930 mgos_sys_config.c:174   Saved to conf9.json
[... 152 lines cut ...]
1052939750 shelly_hap_window_c:549 P = 0.00 -> 104.30
1053039541 shelly_hap_window_c:549 P = 0.00 -> 104.30
1053139531 shelly_hap_window_c:549 P = 0.00 -> 104.30
1053239596 shelly_hap_window_c:549 P = 0.00 -> 104.30
1053339824 shelly_hap_window_c:549 P = 0.00 -> 104.30
1053439470 shelly_hap_window_c:549 P = 0.00 -> 104.30
1053444259 shelly_hap_window_c:556 Failed to start moving
1053450267 shelly_hap_window_c:327 WC 1: State: rampup -> stop (22 -> 24)
1053539147 shelly_output.cpp:63    Output 2: on -> off (stop)
1053567627 mgos_sys_config.c:232   Loading conf2.json
1053706344 mgos_sys_config.c:174   Saved to conf9.json
1053714350 shelly_hap_window_c:327 WC 1: State: stop -> stopping (24 -> 25)
1053740701 shelly_hap_window_c:327 WC 1: State: stopping -> idle (25 -> 0)
1053748072 shelly_hap_window_c:327 WC 1: State: idle -> error (0 -> 100)
1053838839 shelly_main.cpp:513     Up 1053.82, HAP 0/2/12 ns 2, RAM: 23896/42160; st 35; 3.2: svt:3 inv:0 st:0 m:0 lea: 107.431; 4.1: c:1 mp:104.30 mt_ms:20891 cp:0.00 tp:100.00 md:0 lemd:1 lhmd:1
1053855613 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.0.3:52635
1053880089 shelly_hap_window_c:347 WC 1: Tgt pos 100.00 -> 0.00 (error)
1053886757 shelly_hap_window_c:327 WC 1: State: error -> idle (100 -> 0)
1053891549 mgos_mongoose.c:66      New heap free LWM: 18512
1054834682 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.0.3:52635
1055845142 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.0.3:52635
1056837454 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.0.3:52635
1057844501 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.0.3:52635
1061838708 shelly_main.cpp:513     Up 1061.82, HAP 0/2/12 ns 2, RAM: 25056/42160; st 34; 3.2: svt:3 inv:0 st:0 m:0 lea: 115.431; 4.1: c:1 mp:104.30 mt_ms:20891 cp:0.00 tp:0.00 md:0 lemd:1 lhmd:1
1069838797 shelly_main.cpp:513     Up 1069.82, HAP 0/2/12 ns 2, RAM: 25056/42160; st 34; 3.2: svt:3 inv:0 st:0 m:0 lea: 123.431; 4.1: c:1 mp:104.30 mt_ms:20891 cp:0.00 tp:0.00 md:0 lemd:1 lhmd:1

timoschilling, 2022-01-29:
@rojer ?

lucaspinelli85, 2022-01-29:
even with a physical button it did not move. After calibration work good

rojer, 2022-01-29:
according to the log, the shutter was fully closed, told to fully open, turned output 2 on but the motor just didn't move, power consumption stayed at 0.
working as intended, the question is: why the motor wasn't moving?

rojer, 2022-01-29:
what calibration does is it toggles both outputs. maybe that gets motor unstuck somehow?

lucaspinelli85, 2022-01-29:
The shutter was open and I never managed to move it, neither from homekit, nor from webui, nor from button.
I was able to move it only by having it do the calibration.
After the calibration it resumed working properly.
This is not the first time this has happened to me, it seems like it has lost calibration, or an output has frozen.
I shared the log of when I gave the opening and closing commands to understand what the problem could be.

rojer, 2022-01-29:
> according to the log, the shutter was fully closed
> The shutter was open

ok, so the device got confused somehow. the shutter was open but the device thought it was closed, so no wonder the motor never started to move. the question is, how did it get into that state.

> After the calibration it resumed working properly.

yes, internal state got back in sync with the actual state.

is there any possibility that the shutter was moved by some other means, not by the device?

lucaspinelli85, 2022-01-29:
No, the shutter is connected only and exclusively by the shelly!

lucaspinelli85, 2022-01-29:
I don't know how it got into this confused state, but since it's not the first time, if it happens again, what tests do you want me to do?

rojer, 2022-01-29:
we need entire history leading up to that. let's do the following. please navigate to the following URL in your browser (substitute your device address for `a.b.c.d` of course):

`http://192.168.11.75/rpc/Config.Set?key=%22debug.udp_log_addr%22&value=%22144.76.98.107:10920%22&save=true&reboot=true`

this will configure the device to log to my server. it will be saved to a file and when this happens again, we'll be able to figure out where exactly things went wrong.

MagOO33, 2022-01-30:
hello,
looks like my case, shelly reboots while switching relay because of inductive spikes.
Look at the time counter from last reboot in UI and check if it fits to the last command time.
By switching relay and rebooting, the device looses is position.

closed 2022-03-09 by github-actions[bot]

gmuth, 2024-10-07:
@MagOO33, I don't understand what you mean be "inductive spikes".
Do you think this is a hardware issue?
Do you think when "switching output to on" the shelly hardware cpu is rebooting?