mongoose-os-apps/shelly-homekit #921: Shelly 1 not reachable every few days (no problem with original FW)?

mensa84, 2022-01-30:
Hello,

every few days (or even after a week), my Shelly 1 is not connected anymore to my WiFi although the signal is perfect 2 meters next to the access point. I have to power off the Shelly and power on to get it connected again.
This problem do only appear with the HomeKit firmware. When I install the original firmware it stays connected the whole time.
Do you know that problem? How to fix that?

freddygl, 2022-02-07:
I have exactly the same problem.
I have two Shelly1s with this firmware, one in the garage and one in the portal.
Every x time the one in the portal disconnects and I have to remove the power to recover it.

mensa84, 2022-02-07:
> I have exactly the same problem.
> I have two Shelly1s with this firmware, one in the garage and one in the portal.
> Every x time the one in the portal disconnects and I have to remove the power to recover it.

Is your problem also solved, as soon as you use stock firmware?

freddygl, 2022-02-07:
Yes

freddygl, 2022-02-22:
¿¿¿??? :)

timoschilling, 2022-02-22:
sure you can try

freddygl, 2022-03-02:
Beta don`t fix the problem :(

MilanBacaCz, 2022-03-04:
Do i have access to log file, after device is not responding? I mean is is possibele to see what was wrong before freeze? The device seems still running, because buttons works as expected, but no connection anymore over homekit even over web gui even ping not work. Like someone before said - Wi-Fi signal is perfect. This behavior happen almost every month on more shellys (not at the same time). After this happend i have to open/close circuit breaker and then it works again (few weeks).

timoschilling, 2022-03-04:
@MilanBacaCz which version is running on your Shelly's? Sounds like a bug we have fixed in 2.10, there was a freeze in HomeKit after ~25 days.

Details for logging will follow.

MilanBacaCz, 2022-03-04:
Latest versin, but i am appologize. This thread is about Shelly one and I am talking about Shelly 2.5 .

freddygl, 2022-03-04:
I have checked the log but nothing is visible.
In my case I have two shelly1 (portal and garage). The funny thing is that only the portal fails. Same wifi and distance. It doesn't last online for more than a day.
I can only recover it by lowering the circuit breaker just like the partner.

freddygl, 2022-03-21:
Hello, I have installed a mesh Wi-Fi system and without wanting it I have solved the problem of my shelly1.
Since I have installed everything works perfectly and the shelly does not disconnect.

closed 2022-03-21 by timoschilling

mensa84, 2022-03-21:
@timoschilling Why closed? The problem is still not fixxed :(

mensa84, 2022-05-17:
Will there be an updated FW based on official 1.11.8?

rojer, 2022-05-17:
our releases are not based on original firmware in any way

mensa84, 2022-05-17:
> our releases are not based on original firmware in any way

Why must then be updated to the latest official FW before installing your release?