mongoose-os-apps/shelly-homekit #922: Device not responding in homekit

Gomas17, 2022-01-31:
Whenever I access the home application to control some shelly, it just stays in "Updating" mode and doesn't load, only after a while the shelly state appears.
When I access the IP directly through the browser (w-fi or ethernet) I can see the status and control the shelly without any delay, I think it's not wifi connection problems

timoschilling, 2022-01-31:
How many and what kind of bridges do you have in your setup?

Gomas17, 2022-01-31:
i have a HomePod Mini in the same space as the shellys

timoschilling, 2022-01-31:
No sorry I mean, do you have a Philips Hue, Ikea Tratfry, Homebridge?

MilanBacaCz, 2022-01-31:
is Philips hue an issue?

S pozdravem,
Milan Bača

timoschilling, 2022-02-03:
Having bridges can slow down and timeout the update process. It often happens with many bridged devices on one bridge or with DIY bridges like Homebridge.

Gomas17, 2022-02-05:
> Having bridges can slow down and timeout the update process. It often happens with many bridged devices on one bridge or with DIY bridges like Homebridge.

I only have Philips Hue Bridge

mleznik, 2022-02-05:
Hi, 
Similar problem on my side, Shelly 1 hangs on "Updating" in Homekit every 1-2 days, works fine after a restart via the web interface. 

2 Homepods and an ATV.
Bosch Smarthome Bridge & Hue Bridge.

Any idea how to fix that?

Thanks!

closed 2022-03-15 by github-actions[bot]